espressif/esp-matter #371: Compilation stuck on  Completed 'bootloader'.

abhijits3, 2023-05-05:
I am trying to compile the light example but it gets stuck on Completed 'bootloader'. All other examples have the same issue.
I am using esp-idfv4.4.2 on  mac

leep.c.obj
[75/97] Building C object esp-idf/esp_rom/CMakeFiles/__idf_esp_rom.dir/patches/esp_rom_uart.c.obj
[76/97] Building C object esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/src/esp_err_to_name.c.obj
[77/97] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj
[78/97] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/bootloader_start.c.obj
[79/97] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_buffers.c.obj
[80/97] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log.c.obj
[81/97] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_clk.c.obj
[82/97] Linking C static library esp-idf/log/liblog.a
[83/97] Linking C static library esp-idf/esp_rom/libesp_rom.a
[84/97] Linking C static library esp-idf/esp_common/libesp_common.a
[85/97] Linking C static library esp-idf/xtensa/libxtensa.a
[86/97] Linking C static library esp-idf/esp_hw_support/libesp_hw_support.a
[87/97] Linking C static library esp-idf/esp_system/libesp_system.a
[88/97] Linking C static library esp-idf/efuse/libefuse.a
[89/97] Linking C static library esp-idf/bootloader_support/libbootloader_support.a
[90/97] Linking C static library esp-idf/spi_flash/libspi_flash.a
[91/97] Linking C static library esp-idf/micro-ecc/libmicro-ecc.a
[92/97] Linking C static library esp-idf/soc/libsoc.a
[93/97] Linking C static library esp-idf/hal/libhal.a
[94/97] Linking C static library esp-idf/main/libmain.a
[95/97] Linking C executable bootloader.elf
[96/97] Generating binary image from built executable
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /Users/abhijit/esp/esp-matter/examples/light/build/bootloader/bootloader.bin
[97/97] cd /Users/abhijit/esp/esp-matter/examples/light/build/bootloader/esp-idf/esptool_py && /Users/abhijit/.espressif/python_env/idf4.4_py3.8_env/bin/python /Users/abhijit/esp/esp-idf/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x1000 /Users/abhijit/esp/esp-matter/examples/light/build/bootloader/bootloader.bin
Bootloader binary size 0x6330 bytes. 0x4cd0 bytes (44%) free.
[1144/1298] Completed 'bootloader'

PSONALl, 2023-05-05:
@abhijits3 Can you please retry with the updated IDF commit?
For ESP32, ESP32-C3, and ESP32-S3, ESP-IDF [v5.0.1 release](https://github.com/espressif/esp-idf/releases/tag/v5.0.1) is required. and for ESP32-C2, ESP32-H2(preview) and ESP32-C6(preview) and Zigbee Bridge example, ESP-IDF branch at [commit bb9200ace](https://github.com/espressif/esp-idf/tree/bb9200ace) in branch release/v5.1 should be used.

abhijits3, 2023-05-05:
Tried with [commit bb9200ace](https://github.com/espressif/esp-idf/tree/bb9200ace). Now it stuck at libespressif__mdns.a

-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /Users/abhijit/esp/esp-idf/components/bootloader /Users/abhijit/esp/esp-idf/components/bootloader_support /Users/abhijit/esp/esp-idf/components/efuse /Users/abhijit/esp/esp-idf/components/esp_app_format /Users/abhijit/esp/esp-idf/components/esp_common /Users/abhijit/esp/esp-idf/components/esp_hw_support /Users/abhijit/esp/esp-idf/components/esp_rom /Users/abhijit/esp/esp-idf/components/esp_system /Users/abhijit/esp/esp-idf/components/esptool_py /Users/abhijit/esp/esp-idf/components/freertos /Users/abhijit/esp/esp-idf/components/hal /Users/abhijit/esp/esp-idf/components/log /Users/abhijit/esp/esp-idf/components/bootloader/subproject/main /Users/abhijit/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /Users/abhijit/esp/esp-idf/components/newlib /Users/abhijit/esp/esp-idf/components/partition_table /Users/abhijit/esp/esp-idf/components/soc /Users/abhijit/esp/esp-idf/components/spi_flash /Users/abhijit/esp/esp-idf/components/xtensa
-- Configuring done (13.7s)
-- Generating done (0.4s)
-- Build files have been written to: /Users/abhijit/esp/esp-matter/examples/light/build/bootloader
[102/103] Generating binary image from built executableesptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /Users/abhijit/esp/esp-matter/examples/light/build/bootloader/bootloader.bin
[103/103] cd /Users/abhijit/esp/esp-matter/exampl...er/examples/light/build/bootloader/bootloader.binBootloader binary size 0x6790 bytes. 0x4870 bytes (41%) free.
[626/1151] Performing configure step for 'chip_gn'Done. Made 297 targets from 230 files in 674ms
[975/1151] Building C object esp-idf/espressif__l...sif__led_strip.dir/src/led_strip_rmt_ws2812.c.objIn file included from /Users/abhijit/esp/esp-matter/examples/light/managed_components/espressif__led_strip/src/led_strip_rmt_ws2812.c:20:
/Users/abhijit/esp/esp-idf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~
[996/1151] Linking C static library esp-idf/espressif__mdns/libespressif__mdns.a

abhijits3, 2023-05-05:
Thanks..thats worked

shubhamdp, 2023-05-05:
@abhijits3 Matter is a big project and compilation takes more time than usual, be patient and wait till the compilations gets successful.
There are two variants of build system being used here, gn and cmake, cmake triggers the gn for compiling files in connectedhomeip repository so when you see something being stuck it is actually compiling files in connectdhomeip repository.

shubhamdp, 2023-05-05:
Please close the issue if example compiles successfully.

shubhamdp, 2023-05-05:
> v5.0.1 release keeps on giving the message Please use idf.py only in an ESP-IDF shell environment.

This means you are not using idf python environment and using different python environment for building the matter application.

Please follow [development setup guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup) to not run into such issues.

Also, you can check the active python environments by below command
```
which -a python3
```
If the very first one shall be similar to `/Users/shubhampatil/.espressif/python_env/idf5.0_py3.11_env/bin/python3`.

abhijits3, 2023-05-05:
yes. I followed this guide

which -a python3 output is:
/Users/abhijit/.espressif/python_env/idf5.0_py3.8_env/bin/python3
/usr/bin/python3

shubhamdp, 2023-05-05:
Looking at python3 interpreter path, you have installed the idf tools perfectly.

I could suspect one thing though that build directory still have the stale stuff from idf v4.4.x

Can you please clean up the build and sdkconfig and try again in fresh terminal?
```
/// open new terminal
cd esp-idf
source export.sh
cd  esp-matter
source export.sh
cd examples/light
rm -rf sdkconfig build
idf.py build
```

abhijits3, 2023-05-06:
Sorry that didnt work. Just running 
/// open new terminal
cd esp-idf
source export.sh

gave the message:
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

**Please use idf.py only in an ESP-IDF shell environment.**

abhijits3, 2023-05-06:
I followed the complete steps again in a different folder and the issue seems resolved. Not sure what the problem with the original installation was. Closing for now. Will reopen if the issue persists

closed 2023-05-06 by abhijits3

shubhamdp, 2023-05-08:
Glad to hear that you have a working setup!