espressif/esp-matter #372: matter onboardingcodes none ouput information is incorrect.

yurenlee, 2023-05-05:
I have flashed the binary for the factory data partition, but the output onboarding payload information is incorrect.

```bash
> matter onboardingcodes none
QRCode:            I (32688) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
MT:Y.K9061901KA0648G00
QRCodeUrl:         I (32688) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K9061901KA0648G00
ManualPairingCode: I (32708) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
34970112332
Done
````

PSONALl, 2023-05-05:
Can you specify which vendor Id, product ID and discriminator you are using for generating factory partition?
Can you check if you have these config options are enabled if you are using values different than default values: 
CONFIG_FACTORY_PARTITION_DAC_PROVIDER=y
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y

yurenlee, 2023-05-05:
I'm sure that I have specified the VID, PID, and discriminator parameters. I can commission the device using the PNG file generated by the mfg_tool.

This is my sdkconfig.

```bash
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
# CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER is not set
CONFIG_ENABLE_ESP32_DEVICE_INFO_PROVIDER=y
CONFIG_EXAMPLE_DAC_PROVIDER=y
# CONFIG_FACTORY_PARTITION_DAC_PROVIDER is not set
# CONFIG_SEC_CERT_DAC_PROVIDER is not set
```

shubhamdp, 2023-05-05:
@yurenlee While creating a factory partition, it does not include the pincode, this is done to respect the security considerations in Matter Specifications.

For generation of onboarding codes, we required pincode along with other details. So, whatever is generated from the console command is incorrect, and not expected to use that.

Please continue using the qrcode text/image generated using mfg_tool.py

yurenlee, 2023-05-05:
Thanks.I understand. However, in the generic_switch example's README.md, it is noted that the 20202020_3841.bin file is flashed from the mfg_binaries directory. Nevertheless, no QR code is provided.

PSONALl, 2023-05-05:
You can use this QR code for above configurations: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT:634J0CEK01KA0648G00

If you want to generate the QRCode with your own custom values, you can do it using chip-tool command for example in this case this command can be used:
```
./out/debug/chip-tool payload generate-qrcode --existing-payload MT:Y.K9042C00KA0648G00 --discriminator 3841 --setup-pin-code 20202021 --vendor-id 65522 --product-id 32769 --commissioning-mode 0 --rendezvous 2

```
and then QRCode can be generated here: https://project-chip.github.io/connectedhomeip/qrcode.html

yurenlee, 2023-05-05:
Thank you very much, now I understand. However, the generic_switch example is confusing because there is no QR code provided and the matter onboadingcodes none command generates an incorrect QR code. This incorrect QR code causes the device to not be discovered in Apple Home and to fail PASE in Google Home.

yurenlee, 2023-05-05:
> You can use this QR code for above configurations: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT:634J0CEK01KA0648G00
>
> If you want to generate the QRCode with your own custom [...]
@PSONALl 
Thank you very much, now I understand. However, the generic_switch example is confusing because there is no QR code provided and the matter onboadingcodes none command generates an incorrect QR code. This incorrect QR code causes the device to not be discovered in Apple Home and to fail PASE in Google Home.

yurenlee, 2023-05-05:
> @yurenlee While creating a factory partition, it does not include the pincode, this is done to respect the security considerations in Matter Specifications.
>
> For generation of onboarding codes, we [...]

@shubhamdp 
Thanks.I understand. However, in the generic_switch example's README.md, it is noted that the 20202020_3841.bin file is flashed from the mfg_binaries directory. Nevertheless, no QR code is provided.

shubhamdp, 2023-05-05:
@yurenlee Thanks for pointing it out, will add the onboarding codes and qrcode to that example.

closed 2023-05-10 by espressif-bot

anuarrozman, 2023-09-25:
Hi @shubhamdp , I would like to ask if is it possible to retrieve the QR payload (MT:) from the device itself?