espressif/esp-matter #374: ESP Matter WiFi device pairing failed

Focalcrest-Madi, 2023-05-05:
ESP Matter SDK version: release/v1.0
ESP IDF SDK version: v4.4.3
[matter-device.log](https://github.com/espressif/esp-matter/files/11405051/matter-device.log)
[matter-chip-tool.log](https://github.com/espressif/esp-matter/files/11405053/matter-chip-tool.log)

Focalcrest-Madi, 2023-05-06:
@shubhamdp  Yes, my computer system is MacOS Ventura 13.2.1

dhrishi, 2023-05-07:
@Focalcrest-Madi Please see this for a solution: https://github.com/project-chip/connectedhomeip/issues/26049#issuecomment-1503506828

If this doesn't work, then please raise an issue in the upstream connectedhomeip repository as it seems to be MacOS specific issue.

closed 2023-05-07 by dhrishi

solstice-gao, 2023-12-07:
在 macOS 上使用chip-tool，或在 iOS 上使用 CHIP Tool
使用开源 Matter Darwin 芯片工具和 iOS 芯片工具配对 Matter 配件需要在 MacOS 或 iOS/iPadOS 上安装 [蓝牙 Central Matter 客户端开发人员模式配置文件](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig) 。

下载蓝牙 Central Matter 客户端开发人员模式配置文件并将其安装在 iOS/iPadOS 16.1 beta 3 和 MacOS 13.1 beta 3 或更高版本的系统上。

对于macOS，可以通过“设置”->“隐私和安全”->“配置文件”安装配置文件
对于iOS/iPadOS，如有必要，请通过电子邮件发送配置文件或使用 AirDrop 将配置文件传输到iOS/iPadOS 16 beta 3设备。
重新启动您的系统

对于iOS/iPadOS，启用开发者模式。请参阅 [此开发者页面](https://developer.apple.com/documentation/xcode/enabling-developer-mode-on-a-device)
编译适用于 macOS 的芯片工具或适用于 iOS 的 CHIP Tool