espressif/esp-matter #375: Reporting of an attribute change results in Core 1 Panic'ed

CoolerIV, 2023-05-05:
Hi,

I'm trying to report back an attribute value change, but esp32s3 devboard reboots.

I've used and example app of a light switch. This works, I'm able to commission the device to Home Kit and everything is working as expected.

ESP-IDF: 5.0.1
MATTER: main

My goal: let's say light status gets updated: for the sake of an example, say that when you turn it ON via home kit, GPIO input pin trigger turns it OFF. I need to report this change back to ecosystem so in apple Home Kit light status will be represented correctly.

What I do:

        esp_matter_attr_val_t attr_val = esp_matter_bool(true);
        esp_matter::attribute::report(light_endpoint_id1, OnOff::Id, OnOff::Attributes::OnOff::Id, &attr_val);

light_endpoint_id1 <= is valid and working endpoint id

the result I get is this:

I (46533) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 1 *********Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40381e3d  PS      : 0x00060e33  A0      : 0x80380a9d  A1      : 0x3fcb72f0  
0x40381e3d: xTaskRemoveFromEventList at /esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3798

A2      : 0x00000000  A3      : 0x0000003d  A4      : 0x3fc993c0  A5      : 0x00000001  
A6      : 0x3c0e6530  A7      : 0x0000abab  A8      : 0x80381e33  A9      : 0x3fcb72c0  
A10     : 0x00000001  A11     : 0x3fc9dec4  A12     : 0x00000004  A13     : 0x00060e23  
A14     : 0x3fcf049c  A15     : 0x0000abab  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffa

Advice or insights are appreciated, 
Thank you

dhrishi, 2023-05-06:
@CoolerIV See this: https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L67-#L70
Here, we toggle the state of light on a button press. Please check if this helps.

CoolerIV, 2023-05-07:
Thank you, it was exactly what I was looking for, and it works like a charm!

closed 2023-05-07 by CoolerIV