espressif/esp-matter #376: Problem with using own PAA

slayerjojo, 2023-05-06:
I use the following command to use my own paa,there is a error in commissioning step 'AttestationVerification'.
`Failed in verifying 'Attestation Information' command received from the device: err 601. Look at AttestationVerificationResult enum to understand the errors`
- generate PAA file：
`chip-cert gen-att-cert --type a --subject-cn "TEST PAA 01" --valid-from "2020-10-15 14:23:43" --lifetime 7305 --out-key Chip-PAA-Key.pem --out Chip-PAA-Cert.pem`
- generate CD file:
`chip-cert gen-cd --key Chip-PAA-Key.pem --cert Chip-PAA-Cert.pem --out Sunricher-CD-1224-1000.der --format-version 1 --vendor-id 1224 --product-id 1000 --device-type-id 0101 --certificate-id CSA00000SWC00000-01 --security-level 0 --security-info 0 --version-number 1 --certification-type 1`
- use mfg_tool to generate factory bin:
`~/.espressif/esp-matter/tools/mfg_tool/mfg_tool.py -n 1 -cn MTWF9101SAC --paa -c Chip-PAA-Cert.pem -k Chip-PAA-Key.pem               -cd Sunricher-CD-1224-1000.der -v 0x1224 --vendor-name Sunricher -p 0x1000               --product-name "Smart WiFi Phase Cut Dimmer" --hw-ver 1 --hw-ver-str v1.0`
- flash factory bin
`parttool.py --port /dev/ttyUSB0 write_partition --partition-name=nvs --input ~/Sunricher/matter/cert/out/1224_1000/cdc9a829-131e-44b3-b40c-a67916161c7a/cdc9a829-131e-44b3-b40c-a67916161c7a-partition.bin`
- convert paa format to der
`chip-cert convert-key Chip-PAA-Key.pem Chip-PAA-Key.der --x509-der
chip-cert convert-cert Chip-PAA-Cert.pem Chip-PAA-Cert.der --x509-der`
- pair
`chip-tool pairing ble-wifi 0x7283 TP-LINK_754E xxxxxxx 67382209 1862 --paa-trust-store-path ./`

slayerjojo, 2023-05-08:
I found that the reason for the error was that I used my own PAA certificate in generating the CD file, so I should have used the test certificate

shubhamdp, 2023-05-08:
Please see [here](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/attestation_verifier/DeviceAttestationVerifier.h#L69), 601 is `kCertificationDeclarationNoCertificateFound`, so you may have missed some enabling some config option.

Also, there is some problem with how you have passed the vendor id and product id in above commands, in `chip-cert` command you have used decimal values `1224` and `1000` whereas in `mfg_gen` command you have used hex representationm `0x1224` and `0x1000` and 0x1224 is not equal to 1224, so even if you generate CD with test CD certs, it would have failed some other place.

Ideally to use the attestation certs and cd from factory following options shall be enabled.
```
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_FACTORY_PARTITION_DAC_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
```

dhrishi, 2023-05-16:
@slayerjojo Please close the issue if resolved

dhrishi, 2023-05-24:
Closing. Please feel free to re-open in case of any issues

closed 2023-05-24 by dhrishi