espressif/esp-matter #377: OTA failed , help T T

moculll, 2023-05-09:
```
I (5002130) chip[SWU]: Stopping the Periodic Query timer
I (5002140) chip[SWU]: Starting the watchdog timer, timeout: 21600 seconds
I (5002150) chip[DIS]: Resolving 3A2BB8C447E77D6A:0000000000000002 ...
E (5002160) chip[SWU]: Failed to connect to node 0x0000000000000002: 2d
I (5002160) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x002A's Attribute 0x0002 is 2 **********
I (5002180) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x002A's Attribute 0x0002 is 1 **********
I (5002190) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x002A's Attribute 0x0003 is null **********
I (5002190) chip[SWU]: Unknown idle state reason so set the periodic timer for a next attempt
I (5002210) chip[SWU]: Stopping the watchdog timer
I (5002210) chip[SWU]: Starting the periodic query timer, timeout: 86400 seconds
```
above is the code , it says "Failed to connect to node", my worktool is :
chip-tool on raspiberry 4b
nordic52dk <-> raspiberry4b  as ot-br
chip-ota-provider-app on ubuntu 18.04 vmware
esp32 matter light

I want to know that my chip-ota-provider-app is running on vmware,which means I don't have bluetooth, does it matter?
Be thankful if someone can help me with this problem

moculll, 2023-05-09:
by the way,
esp32 is set to node 1
ota-provider-app is set to node 2

jadhavrohit924, 2023-05-09:
@moculll Please provide us full logs in files ( do not paste ). Also check is ota-requestor and ota-provider are on same network and try again.

moculll, 2023-05-09:
> @moculll Please provide us full logs in files ( do not paste ). Also check is ota-requestor and ota-provider are on same network and try again.

On my way, please wait a minute

moculll, 2023-05-09:
[MobaXterm_1_20230509_134255.txt](https://github.com/espressif/esp-matter/files/11427741/MobaXterm_1_20230509_134255.txt)
@jadhavrohit924

moculll, 2023-05-09:
[log_esp.txt](https://github.com/espressif/esp-matter/files/11427754/log_esp.txt)
@jadhavrohit924 this is the log of esp's, and above is chip-tool's

moculll, 2023-05-09:
> @moculll Have you checked if ota-provider is online/reachable and on same network?
```
wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fd2f:9851:8d71:da24:0:ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        inet6 fd2f:9851:8d71:da24:0:ff:fe00:6400  prefixlen 64  scopeid 0x0<global>
        inet6 fd0c:e97a:ddf4:1:dc87:990c:e438:3ec2  prefixlen 64  scopeid 0x0<global>
        inet6 fd2f:9851:8d71:da24:0:ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fd2f:9851:8d71:da24:a68d:28e8:821e:5934  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::8840:d49:f923:96ad  prefixlen 64  scopeid 0x20<link>
        inet6 fd2f:9851:8d71:da24:0:ff:fe00:fc11  prefixlen 64  scopeid 0x0<global>
        inet6 fd2f:9851:8d71:da24:0:ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 337  bytes 67402 (67.4 KB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 4446  bytes 523981 (523.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
here is my ot-br's address, I find that my vmware can ping to "fd0c:e97a:ddf4:1:dc87:990c:e438:3ec2"(line 3), and others can't

wqx6, 2023-05-09:
Look like the OTA requestor cannot resolve the address of the OTA provider. Can you use wireshark to capture the mdns packets so that we can see whether the provider replies responses for the mdns query? BTW, ot-br is not necessary for the OTA test. You can disable it.

moculll, 2023-05-09:
> Looks like the OTA requestor cannot resolve the address of the OTA provider. Can you use wireshark to capture the mdns packets so that we can see whether the provider replies responses for the mdns query? BTW, ot-br is not necessary for the OTA test. You can disable it.

I want to use chip-tool to change the announce-ota-provider func, but I don't know how to use apple hub instead of ot-br to connect to my esp and controlling by chip-tool

wqx6, 2023-05-10:
Are you using ESP32 now? It is a Wi-Fi device so chip-tool can control it without ot-br. And please make sure that both the ESP32 and OTA-provider are commissioned to the same fabric. If you have done these, could you send us the wireshark capture so that we can know why the ESP32 cannot resolve the address of the OTA-provider?

moculll, 2023-05-11:
> Are you using ESP32 now? It is a Wi-Fi device so chip-tool can control it without ot-br. And please make sure that both the ESP32 and OTA-provider are commissioned to the same fabric. If you have [...]

I'm using esp32-h2, it is a matter-over-thread device, I've tested connectedhomeip/examples folder's requester and provider samples, they worked well, and that means the problem maybe is that esp32h2 cannot link with node 2(provider), I will see if I can get the packets from provider and esp32h2

Jerry-ESP, 2023-05-19:
> I'm using esp32-h2, it is a matter-over-thread device, I've tested connectedhomeip/examples folder's requester and provider samples, they worked well, and that means the problem maybe is that [...]

Can you share us all the log include commissioning process? I want to your whole progress about testing esp32h2 OTA.

Jerry-ESP, 2023-05-19:
> > Looks like the OTA requestor cannot resolve the address of the OTA provider. Can you use wireshark to capture the mdns packets so that we can see whether the provider replies responses for the mdns query? BTW, ot-br is not necessary for the OTA test. You can disable it.
> 
> I want to use chip-tool to change the announce-ota-provider func, but I don't know how to use apple hub instead of ot-br to connect to my esp and controlling it by chip-tool

If you want to use Apple hub instead od OT-BR, I think you should conmissioning the h2 device first through HomeKit, after successfully,  open conmissing window, then conmissioning the device use chip-tool with command: "./chip-tool pairing code 2 04603319906"

moculll, 2023-05-22:
> If you want to use Apple hub instead od OT-BR, I think you should conmissioning the h2 device first through HomeKit, after successfully, open conmissing window, then conmissioning the device use [...]

Thank you for the guidence! I will give it a try, and get some logs

dhrishi, 2023-06-02:
@moculll Is there an update you can share? Please close the issue if the OTA is working fine

moculll, 2023-06-06:
> @moculll Is there an update you can share? Please close the issue if the OTA is working fine
Nothing changed T T

dhrishi, 2023-06-24:
@moculll Are you still seeing issues? If yes, can you please provide the relevant logs.

dhrishi, 2023-08-12:
Closing this now. Please re-open if you still see the issue, along with all the relevant logs

closed 2023-08-12 by dhrishi