espressif/esp-matter #378: How to create thermostat endpoint and add additional features (cluster)?

eme-wesleywong, 2023-05-09:
ESP Matter version: Branch release/v1.0
I would like to ask what is the proper steps to create thermostat endpoint and add additional cluster.
In esp-matter/examples, there is no thermostat example. I can only use light example to try to create thermostat endpoint.
i would like to know if following steps are correct or I am missing something.
Can I just use light example, delete all the light related code and then add following code to create thermostat related project?
```
app_driver_handle_t thermostat_handle = app_driver_thermostat_init();
node::config_t node_config;
node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

// Create endpoint
thermostat::config_t thermostat_config;
thermostat_config.thermostat.local_temperature = 2500;
thermostat_config.thermostat.system_mode = 0;
endpoint_t *thermostat_endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, thermostat_handle);

// Add additional feature
cluster_t *thermostat_cluster = cluster::get(thermostat_endpoint, Thermostat::Id);
cluster::thermostat::feature::heating::config_t heating_config;
heating_config.abs_max_heat_setpoint_limit = 3000;
heating_config.abs_min_heat_setpoint_limit = 1500;
heating_config.max_heat_setpoint_limit = 2800;
heating_config.min_heat_setpoint_limit = 2000;
heating_config.occupied_heating_setpoint = 2300;
heating_config.pi_heating_demand = 0;
cluster::thermostat::feature::heating::add(thermostat_cluster, &heating_config);
```

jadhavrohit924, 2023-05-09:
@eme-wesleywong The code you have added is correct. Above code will create thermostat endpoint with heating feature.

jadhavrohit924, 2023-05-09:
Are you facing any issues with this code? If yes, please provide us more details.

dhrishi, 2023-05-15:
@eme-wesleywong Please provide details if you are facing any problems or else close the issue

dhrishi, 2023-05-26:
Closing this now

closed 2023-05-26 by dhrishi

aggaddam, 2023-06-01:
@jadhavrohit924 I took the copy of the Light example and made changes for the thermostat device. After that I am unable to commission the device with chip tool

`chip-tool interactive start`

`pairing ble-wifi 0x7283 <ssid> <passphrase> 20202021 3840`

Here is the sample code I modified for the thermostat.. Do I need to add any configuration settings to support thermostat device type in SDK config ?

```
extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t thermostat_handle = app_driver_thermostat_init();

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    if (node)
    {
        // Add enpoint to the node
        thermostat::config_t thermostat_config;
        thermostat_config.thermostat.local_temperature = 2500;
        thermostat_config.thermostat.system_mode       = 0;    // OFF state
        endpoint_t *thermostat_endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);

        if (thermostat_endpoint)
        {
            thermostat_endpoint_id = endpoint::get_id(thermostat_endpoint);
            ESP_LOGI(TAG, "Thermostat endpoint_id = %d", thermostat_endpoint_id);

            // Add additional cooling cluster
            cluster_t *thermostat_cluster = cluster::get(thermostat_endpoint, Thermostat::Id);
            esp_matter::cluster::thermostat::feature::cooling::config_t cooling_config;
            //cooling_config.config();
            cooling_config.abs_min_cool_setpoint_limit = 1600;
            cooling_config.abs_max_cool_setpoint_limit = 3200;
            cooling_config.pi_cooling_demand           = 50;  // 50%
            cooling_config.occupied_cooling_setpoint   = 2600;
            cooling_config.min_cool_setpoint_limit     = 1600;
            cooling_config.max_cool_setpoint_limit     = 3200;

            ESP_LOGE(TAG, "Thermostat cluster id = %d", esp_matter::cluster::thermostat::feature::cooling::get_id());
            esp_matter::cluster::thermostat::feature::cooling::add(thermostat_cluster, &cooling_config);

            // Matter start
            err = esp_matter::start(app_event_cb);
            if (err != ESP_OK)
            {
                ESP_LOGE(TAG, "Matter start failed: %d", err);
            }

        }
        else
        {
            ESP_LOGE(TAG, "Matter endpoint creation failed");
        }
    }
    else
    {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#endif
}
```

jadhavrohit924, 2023-06-01:
@aggaddam 
> Do I need to add any configuration settings to support thermostat device type in SDK config ?

No.

Ideally, commissioning should work. Have you changed any other files? Please attach logs from device and commissioner.

aggaddam, 2023-06-01:
@jadhavrohit924 Here is the log file

```
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1718
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (30) boot: ESP-IDF v5.1-dev-4784-g6cfc6f53be 2nd stage bootloader
I (30) boot: compile time May 31 2023 14:35:48
I (31) boot: chip revision: v0.3
I (35) boot.esp32c3: SPI Speed      : 80MHz
I (40) boot.esp32c3: SPI Mode       : DIO
I (44) boot.esp32c3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (73) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (80) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (88) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (95) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (103) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (110) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (118) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (125) boot: End of partition table
I (130) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=38de8h (232936) map
I (175) esp_image: segment 1: paddr=00058e10 vaddr=3fc94a00 size=03494h ( 13460) load
I (178) esp_image: segment 2: paddr=0005c2ac vaddr=40380000 size=03d6ch ( 15724) load
I (184) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=f6dfch (1011196) map
I (351) esp_image: segment 4: paddr=00156e24 vaddr=40383d6c size=10ae8h ( 68328) load
I (370) boot: Loaded app from partition at offset 0x20000
I (370) boot: Disabling RNG early entropy source...
I (382) cpu_start: Unicore app
I (382) cpu_start: Pro cpu up.
I (391) cpu_start: Pro cpu start user code
I (391) cpu_start: cpu freq: 160000000 Hz
I (391) cpu_start: Application information:
I (394) cpu_start: Project name:     light
I (399) cpu_start: App version:      v1.0
I (403) cpu_start: Compile time:     May 31 2023 16:39:51
I (410) cpu_start: ELF file SHA256:  76529db0b127621b...
I (415) cpu_start: ESP-IDF:          v5.1-dev-4784-g6cfc6f53be
I (422) cpu_start: Min chip rev:     v0.3
I (427) cpu_start: Max chip rev:     v0.99 
I (432) cpu_start: Chip rev:         v0.3
I (436) heap_init: Initializing. RAM available for dynamic allocation:
I (444) heap_init: At 3FCA8310 len 00034400 (209 KiB): DRAM
I (450) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (456) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (463) spi_flash: detected chip: generic
I (467) spi_flash: flash io: dio
W (472) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (483) sleep: Configure to isolate all GPIO pins in sleep state
I (489) sleep: Enable automatic switching of GPIO sleep configuration
I (496) coexist: coex firmware version: ebddf30
I (501) coexist: coexist rom version 9387209
I (506) app_start: Starting scheduler on CPU0
I (511) main_task: Started on CPU0
I (511) main_task: Calling app_main()
E (521) app_driver: Initing Thermostat Device
I (521) led_driver_ws2812: Initializing light driver
I (531) app_main: Thermostat endpoint_id = 1
E (531) app_main: Thermostat cluster id = 2
I (541) pp: pp rom version: 9387209
I (541) net80211: net80211 rom version: 9387209
I (551) wifi:wifi driver task: 3fcb3374, prio:23, stack:6656, core=0
I (551) wifi:wifi firmware version: a0ee385
[... 131 lines cut ...]
I (30981) NimBLE: GAP procedure initiated: advertise; 
I (30981) NimBLE: disc_mode=2
I (30991) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (31001) NimBLE: 

I (31011) chip[DL]: CHIPoBLE advertising started
I (33331) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (33331) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (33341) chip[DL]: Device already advertising, stop active advertisement and restart
I (33341) NimBLE: GAP procedure initiated: stop advertising.

I (33351) NimBLE: GAP procedure initiated: advertise; 
I (33361) NimBLE: disc_mode=2
I (33371) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (33381) NimBLE: 

I (60911) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (60911) chip[DL]: Device already advertising, stop active advertisement and restart
I (60921) NimBLE: GAP procedure initiated: stop advertising.

I (60921) NimBLE: GAP procedure initiated: advertise; 
I (60931) NimBLE: disc_mode=2
I (60931) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (60941) NimBLE: 
```

jadhavrohit924, 2023-06-01:
@aggaddam Please attach ( not paste ) both log files device and chip-tool.

anuarrozman, 2023-12-21:
Hi @jadhavrohit924 , Ive also encountered the same problem. 

here is my log files of the device. Im using Alexa app to commission.
[idf_py_stdout_output_55337.txt](https://github.com/espressif/esp-matter/files/13737655/idf_py_stdout_output_55337.txt)