espressif/esp-matter #380: chip-tool did not find the matter wifi device

Focalcrest-Madi, 2023-05-11:
I run the chip-tool pairing command on the Ubuntu virtual machine, but the device cannot be paired. What is the reason for this? The following is my chip-tool log information

chip-tool operating environment: MacOS running Ubuntu 20.04 virtual machine

[matter-pair.txt](https://github.com/espressif/esp-matter/files/11450345/matter-pair.txt)

jadhavrohit924, 2023-05-11:
@Focalcrest-Madi 
Please attach complete logs and device logs too.

Focalcrest-Madi, 2023-05-12:
@jadhavrohit924 
The attachment is the complete log and device log.
[chip-tool.log](https://github.com/espressif/esp-matter/files/11457737/chip-tool.log)
[device.log](https://github.com/espressif/esp-matter/files/11457753/device.log)

dhrishi, 2023-05-15:
@Focalcrest-Madi This does not look an issue on the ESP32 side. The chip-tool logs show that it found the device with the matching discriminator and tried to connect over BLE
```
[1683853359.576417][194361:194362] CHIP:BLE: Device discriminator match. Attempting to connect.
```
But, nothing happened after that. The device did not even get the BLE connection request. Please check if BLE communication is working fine on the host machine running the chip-tool

jonsmirl, 2023-05-19:
Ubuntu can not share the bluetooth adapter with chip-tool. So if you are using bluetooth for other purposes you have to install a second adapter (USB adapters are $2) and then use --ble-adapter=1 on the chip-tool command.

dhrishi, 2023-05-25:
Closing this as it isn't an issue on the ESP32 side.

closed 2023-05-25 by dhrishi