espressif/esp-matter #382: 'chip::app::Clusters::WindowCovering::WindowCoveringFeature' has not been declared

surjeetk, 2023-05-12:
Facing new issue on latest pull - esp_matter_feature.cpp:384:38: error: 'chip::app::Clusters::WindowCovering::WindowCoveringFeature' has not been declared
return (uint32_t)WindowCovering::WindowCoveringFeature::kLift;

Please check.

ivishaltejwani, 2023-05-12:
did the git pull today and facing the same issue since then , was working before

dhrishi, 2023-05-12:
@surjeetk Can you upload (and not paste), the entire log? Please include all the setup related (install/export) instructions you execute too. I was able to compile the `light` example on the latest main branch following the documentation. Also, CI is successful before pushing the changes to Github

ivishaltejwani, 2023-05-12:
https://pastebin.com/DzWBGbL0

@dhrishi  , heres the log same issue as @surjeetk , was working till yesterday , pulled today and giving this error no changes to the example

surjeetk, 2023-05-13:
This got resolved by updating the submodules. Just by git pull it didn't pull the latest changes in submodule connectedhomeip/connectedhomeip.

used below commands:
cd esp-matter
git submodule update --init --depth 1 --progress
cd ./connectedhomeip/connectedhomeip
./scripts/checkout_submodules.py --platform esp32 linux --shallow

closed 2023-05-13 by surjeetk