espressif/esp-matter #383: Test Harness TC-ACE-1.5

stefano055415, 2023-05-12:
When I try to run the TC-ACE-1.5 test the test crashes

Executing Test Step: Commission from TH2

[image]

fails to do the second commissioning even though the window is open

Do I have to set something special?

Thank you

wqx6, 2023-05-12:
Hi, @stefano055415 Could you provide the log of the TH instead of the error picture? BTW, are you using minimal mdns or platform mdns on ESP32?

stefano055415, 2023-05-12:
@wqx6 

I think I understand the problem.

I have to run: ./chip-all-clusters-app
on raspberry

while the test starts to use it as the second TH

wqx6, 2023-05-16:
The WebUI will start another TH automatically. You don't need another raspberry. Could you please provide the log which can be downloaded from the WebUI? I can't know the reason without the log.

esterdv, 2023-05-16:
@wqx6 I also encountered the same issue when running the test. Please see attached logs.
[TC-ACE_2023_5_16_15_5_6.log](https://github.com/espressif/esp-matter/files/11485067/TC-ACE_2023_5_16_15_5_6.log)

stefano055415, 2023-05-16:
@wqx6 @esterdv  If I don't start a second TH this is the result

[TC-ACE-1_5_run_2023_5_16_9_46_47.log](https://github.com/espressif/esp-matter/files/11485352/TC-ACE-1_5_run_2023_5_16_9_46_47.log)

wqx6, 2023-05-17:
@stefano055415 In your log, the first TH pairs the DUT with the passcode `19628801` and opens the basic commissioning window in Test Step 2.
```
CHIPTOOL   | 2023-05-16 07:47:10.085820 | [1684223228.978864][23:30] CHIP:DMG: 			CommandPathIB =
CHIPTOOL   | 2023-05-16 07:47:10.087627 | [1684223228.978880][23:30] CHIP:DMG: 			{
CHIPTOOL   | 2023-05-16 07:47:10.089287 | [1684223228.978895][23:30] CHIP:DMG: 				EndpointId = 0x0,
CHIPTOOL   | 2023-05-16 07:47:10.101486 | [1684223228.978910][23:30] CHIP:DMG: 				ClusterId = 0x3c,
CHIPTOOL   | 2023-05-16 07:47:10.102975 | [1684223228.978924][23:30] CHIP:DMG: 				CommandId = 0x1,
CHIPTOOL   | 2023-05-16 07:47:10.104173 | [1684223228.978937][23:30] CHIP:DMG: 			},
```
Then the TH2 tries to pair the DUT with this QR-Code `MT:-24J0AFN00KA0648G00`
```
./chip-tool payload parse-setup-payload MT:-24J0AFN00KA0648G00                                                                                          [1684289362.931186][3435807:3435807] CHIP:SPL: Parsing base38Representation: MT:-24J0AFN00KA0648G00
[1684289362.931526][3435807:3435807] CHIP:SPL: Version:             0
[1684289362.931530][3435807:3435807] CHIP:SPL: VendorID:            65521
[1684289362.931532][3435807:3435807] CHIP:SPL: ProductID:           32769
[1684289362.931535][3435807:3435807] CHIP:SPL: Custom flow:         0    (STANDARD)
[1684289362.931538][3435807:3435807] CHIP:SPL: Discovery Bitmask:   0x04 (On IP network)
[1684289362.931540][3435807:3435807] CHIP:SPL: Long discriminator:  3840   (0xf00)
[1684289362.931542][3435807:3435807] CHIP:SPL: Passcode:            20202021
```
The Passcode is 20202021 in this QRCode. TH2 cannot commission the DUT with the wrong passcode. It is a TH issue. Could you please tell me the version of your TH and try with the latest TH version?

wqx6, 2023-05-17:
@esterdv Your error is different from the error of @stefano055415. In your log, the TH2 is using the right QRCode to pair the device, but it fails in Attestation Verification. The error code is 101(PaaNotFound). But TH1 can pair the DUT successfully so I guess that the TH2 doesn't use the paa_trusted_root_path that is used by TH1. It is still the TH issue. Can you please use the latest TH version and try again? If there is still the same issue, please file an issue in [chip-certification-tool](https://github.com/CHIP-Specifications/chip-certification-tool).

esterdv, 2023-05-17:
@wqx6 Thank you for looking into this and directing me to the chip-certification-tool page. It looks like the error is a known issue and we just need to apply a patch. Thank you again!

stefano055415, 2023-05-17:
@wqx6 

Version: v2.7-official-TE
Sha: 821e3f8

it is the version that is indicated on the csa documentation

wqx6, 2023-05-17:
I think there was still some issues for this test case on TH-v2.7. https://github.com/CHIP-Specifications/chip-certification-tool/issues/674, The latest TH should have fixed it.

stefano055415, 2023-05-17:
@wqx6 

the problem is that i don't have access for this repo

stefano055415, 2023-05-18:
@wqx6

Could you please link the latest TH image of the raspberry?

Thank you

stefano055415, 2023-05-19:
@wqx6 

it is possible that the test does not work because the default values ​​are fixed in the  [Test_TC_ACE_1_5.yaml](https://github.com/espressif/connectedhomeip/blob/82db97695c2c899cc130c2d6a0d72b3c92c144db/src/app/tests/suites/certification/Test_TC_ACE_1_5.yaml)

VaishaliAvhale, 2023-10-09:
@stefano055415 This is a semi-automated test case you have to execute some steps manually please check the semi-automated test case execution section in the below document and retest the TC.
[Matter_TH User Guide.pdf](https://github.com/espressif/esp-matter/files/12844664/Matter_TH.User.Guide.pdf)
- Use the latest TH version (i.e TH Fall2023)
- If still does not work please attach the device, TH, and TH chip-tool log here

VaishaliAvhale, 2024-01-31:
For release/v1.1 certification you have test this TC manually please refer release/v1.1 verification document and execute it manually.
Pls refer to the “TCSummary” sheet of Verification  doc for the list of TCs to execute for release/v1.1 certification. There you can find more details about the TC means how to execute it and which PICS needed for execution of particular TC

PhLuReh, 2024-02-09:
I really don't get to the point. I tried to figure out, how the things work, but did not succeed. After reading through the [TC_ACE_1-5](https://github.com/espressif/connectedhomeip/blob/v1.1.0.1/src/app/tests/suites/certification/Test_TC_ACE_1_5.yaml), I tried to manually execute the following steps:
```
./chip-tool pairing ble-wifi 1 SSID PASSWD 91436417 1862
./chip-tool operationalcredentials read CurrentFabricIndex 1
# reads CurrentFabricIndex: 1
./chip-tool administratorcommissioning open-commissioning-window 360 'hex:b96170aae803346884724fe9a3b287c30330c2a660375d17bb205a8cf1aecb350457f8ab79ee253ab6a8e46bb09e543ae422736de501e3db37d441fe344920d09548e4c18240630c4ff4913c53513839b7c07fcc0627a1b8573a149fcd1fa466cf' 3840 1000 "SPAKE2P Key SALT" 1 0 --timedInteractionTimeoutMs 30000
# Commissioning window opened on the DUT
```

Even the DUT is advertising, the next command I does not succeed (like the TH did not). Here's a excerpt from the DUT log:
```
I (5789122) chip[DIS]: Advertise operational node 49B67B9CEE0369C6-0000000000000001                                                                                      I (5789122) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 49B67B9CEE0369C6-0000000000000001.
I (5789142) chip[DIS]: mDNS service published: _matter._tcp
I (5789142) app_main: advertise operational done
I (5789332) app_task: window 0 identification remaining time: -1 (segments: 0x0)
```

Starting with the one from the TC `./chip-tool pairing code-paseonly 2 "MT:-24J0AFN00KA0648G00" --commissioner-name beta` failing in a discovery problem:
```
[1707483477.997825][573319:573319] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1707483478.003409][573319:573319] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1707483478.003665][573319:573319] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1707483478.003771][573319:573319] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1707483478.004125][573319:573319] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Ndw09E)
[1707483478.004743][573319:573319] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1707483478.004800][573319:573319] CHIP:DL: NVS set: chip-counters/reboot-count = 9039 (0x234F)
[1707483478.005651][573319:573319] CHIP:DL: Got Ethernet interface: eth0
[1707483478.006206][573319:573319] CHIP:DL: Found the primary Ethernet interface:eth0
[1707483478.006793][573319:573319] CHIP:DL: Got WiFi interface: wlan0
[1707483478.006858][573319:573319] CHIP:DL: Failed to reset WiFi statistic counts
[1707483478.006910][573319:573319] CHIP:IN: UDP::Init bind&listen port=0
[1707483478.007019][573319:573319] CHIP:IN: UDP::Init bound to port=41258
[1707483478.007042][573319:573319] CHIP:IN: BLEBase::Init - setting/overriding transport
[1707483478.007059][573319:573319] CHIP:IN: TransportMgr initialized
[1707483478.007091][573319:573319] CHIP:FP: Initializing FabricTable from persistent storage
[1707483478.007265][573319:573319] CHIP:TS: Last Known Good Time: 2023-04-11T06:55:26
[1707483478.008794][573319:573319] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x49B67B9CEE0369C6, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1707483478.009259][573319:573319] CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0xE2B270F8EA256EE6, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
[1707483478.011764][573319:573319] CHIP:ZCL: Using ZAP configuration...
[1707483478.015596][573319:573319] CHIP:DL: Avahi client registered
[1707483478.017025][573319:573319] CHIP:CTL: System State Initialized...
[1707483478.017153][573319:573319] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1707483478.017207][573319:573319] CHIP:CTL: Setting attestation nonce to random value
[1707483478.017243][573319:573319] CHIP:CTL: Setting CSR nonce to random value
[1707483478.017548][573319:573321] CHIP:DL: CHIP task running
[1707483478.017818][573319:573321] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1707483478.018148][573319:573321] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1707483478.018193][573319:573321] CHIP:CTL: Setting attestation nonce to random value
[1707483478.018347][573319:573321] CHIP:CTL: Setting CSR nonce to random value
[1707483478.019247][573319:573321] CHIP:CTL: Generating NOC
[1707483478.020234][573319:573321] CHIP:FP: Validating NOC chain
[1707483478.021963][573319:573321] CHIP:FP: NOC chain validation successful
[1707483478.022117][573319:573321] CHIP:FP: Updated fabric at index: 0x2, Node ID: 0x000000000001B669
[1707483478.022147][573319:573321] CHIP:TS: Last Known Good Time: 2023-04-11T06:55:26
[1707483478.022166][573319:573321] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1707483478.022184][573319:573321] CHIP:TS: Retaining current Last Known Good Time
[1707483478.024838][573319:573321] CHIP:FP: Metadata for Fabric 0x2 persisted to storage.
[1707483478.027696][573319:573321] CHIP:TS: Committing Last Known Good Time to storage: 2023-04-11T06:55:26
[1707483478.030211][573319:573321] CHIP:CTL: Joined the fabric at index 2. Fabric ID is 0x0000000000000002 (Compressed Fabric ID: E2B270F8EA256EE6)
[1707483478.047479][573319:573321] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[... cut ...]
I (6050592) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 9226558D0001A7B7.                                                    0601,2404(SEGW),01:00000001
I (6050612) chip[DIS]: mDNS service published: _matterc._udp
I (6050612) chip[EM]: <<< [E:50682r S:0 M:109696317 (Ack:139786415)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FD61:3599:E3A1::1C2%st1]:40090] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (6050622) app_main: Commissioning window opened
```

VaishaliAvhale, 2024-02-21:
Please attach the complete log files. It would be helpful for debugging. Also, please find the attached ACE-1.5 test case steps. Re-execute the test case and share the log files if you face any issues.
[TC-ACE-1_5_Steps.ods](https://github.com/espressif/esp-matter/files/14359223/TC-ACE-1_5_Steps.ods)

PhLuReh, 2024-02-22:
I ran the steps from you specification, which resulted in the following logs:
[2024-02-22-TC-ACE-1_5_chip-tool.log](https://github.com/espressif/esp-matter/files/14370524/2024-02-22-TC-ACE-1_5_chip-tool.log)
[2024-02-22-TC-ACE-1_5_DUT.log](https://github.com/espressif/esp-matter/files/14370525/2024-02-22-TC-ACE-1_5_DUT.log)
```
./chip-tool pairing open-commissioning-window 1 1 400 2000 3841 > 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool pairing code 2 35163245584 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool operationalcredentials read current-fabric-index 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": 31, "endpoint": 0, "deviceType": null }]}, {"fabricIndex": 1, "privilege": 1, "authMode": 2, "subjects": null, "targets": [{ "cluster": 29, "endpoint": 0, "deviceType": null }]}]' 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool descriptor read device-type-list 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool basicinformation read vendor-id 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool descriptor read device-type-list 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool basicinformation read vendor-id 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}]' 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool descriptor read device-type-list 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
./chip-tool operationalcredentials remove-fabric 2 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
```

The `read device-type-list` step fails as it can successfully read the device types, but should fail. Don't know why, are there any ACLs wrongly configured? Or is there something not inited correctly?

VaishaliAvhale, 2024-02-22:
Could you please attach the chip-tool logs containing the chip-tool command you executed? It would be helpful for me to debug. Additionally, please provide the commit IDs for the esp-matter release/v1.1 branch, esp-idf, and connectedhomeip.

On my end, I tested with:

esp-matter: release/v1.1 branch, commit ID: 5abc3095da0da9342dcd1250a01d2c47f840fed0
esp-idf: v5.1.1, commit ID: e088c3766ba440e72268b458a68f27b6e7d63986
connectedhomeip: release/v1.1 branch
All steps executed as expected. Please take a look at the attached logs.
[TC-ACE-1-5-chip-tool.log](https://github.com/espressif/esp-matter/files/14374136/TC-ACE-1-5-chip-tool.log)
[TC-ACE-1.5-DUT.log](https://github.com/espressif/esp-matter/files/14374138/TC-ACE-1.5-DUT.log)

PhLuReh, 2024-02-22:
I knew you would request for this... No problem, here's the combined file
[2024-02-22-TC-ACE-1_5_chip-tool.log](https://github.com/espressif/esp-matter/files/14375042/2024-02-22-TC-ACE-1_5_chip-tool.log)
[2024-02-22-TC-ACE-1_5_DUT.log](https://github.com/espressif/esp-matter/files/14375045/2024-02-22-TC-ACE-1_5_DUT.log)

ESP-Matter commit b2a6701ac6f41a2fdd1549b37f4b58aeab6e0b11
ESP-IDF commit e088c3766ba440e72268b458a68f27b6e7d63986
connectedhomeip commit 181b0cb14ff007ec912f2ba6627e05dfb066c008
Test runs were executed on Version: v2.8.1-official Sha: 1446f09

PhLuReh, 2024-02-22:
Just saw, that I was not developing on the release/v1.1 as I expected. Something with my docker container went wrong. Still waiting for the v1.1 docker image https://github.com/espressif/esp-matter/issues/756#issuecomment-1938392188

VaishaliAvhale, 2024-02-23:
1] Commission TH1
 2] ./chip-tool pairing open-commissioning-window 1 1 400 2000 3841 > 2024-02-22-TC-ACE-1_5_chip-tool.log
 3] ./chip-tool pairing code 2 35163245584 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
 4] ./chip-tool operationalcredentials read current-fabric-index 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
 5] ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": 31, "endpoint": 0, "deviceType": null }]}, {"fabricIndex": 1, "privilege": 1, "authMode": 2, "subjects": null, "targets": [{ "cluster": 29, "endpoint": 0, "deviceType": null }]}]' 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
6]  ./chip-tool accesscontrol write acl '[{"fabricIndex": 2, "privilege": 5, "authMode": 2, "subjects": [223344], "targets": [{ "cluster": 31, "endpoint": 0, "deviceType": null }]}, {"fabricIndex": 2, "privilege": 1, "authMode": 2, "subjects": null, "targets": [{ "cluster": 40, "endpoint": 0, "deviceType": null }]}]' 2 0 --commissioner-name beta
7] ./chip-tool descriptor read device-type-list 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
8] ./chip-tool basicinformation read vendor-id 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
9] ./chip-tool descriptor read device-type-list 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
10] ./chip-tool basicinformation read vendor-id 2 0 --commissioner-name beta >> 2024-02-22-TC-ACE-1_5_chip-tool.log
11] ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}]' 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log
12] ./chip-tool operationalcredentials remove-fabric 2 1 0 >> 2024-02-22-TC-ACE-1_5_chip-tool.log

Hi @PhLuReh, I looked into the shared logs and found that you missed executing Step 6, 'access control write,' for endpoint 2.

VaishaliAvhale, 2024-03-01:
Yes this TestCase Passed Successfully!
Checked logs sent by you and each steps response is according to the spec.

PhLuReh, 2024-03-01:
Thank you @VaishaliAvhale !!

VaishaliAvhale, 2024-03-01:
Hi @PhLuReh, as TC-ACE-1.5 has been executed successfully on your end, could you please close this issue?

PhLuReh, 2024-03-01:
@stefano055415 could you also succeed with these manual steps? so you could close it (I'm not allowed to do so)

dhrishi, 2024-04-24:
Closing this now

closed 2024-04-24 by dhrishi