espressif/esp-matter #384: Returning Status codes

stage-01, 2023-05-12:
Is it possible to add manufacturer specific errors and return them with a failure?

For example when creating a device with a light bulb, would it be possible to return a failure with an error message attached that says: "LIGHT_NOT_FUNCTIONING_PROPERLY"

If possible, how would it be implemented in my code?

dhrishi, 2023-05-15:
@stage-01 What return codes are you talking about: Those for the esp-matter data model APIs provided? Or specification defined return codes (for interaction model or the cluster specific ones)?

stage-01, 2023-05-15:
Im talking about the status codes for the interaction model. Page 445 of the Matter specification

jonsmirl, 2023-05-15:
You are probably looking for the GeneralDiagnostics cluster.  Page 643.

stage-01, 2023-05-15:
thanks for your reply! Yes I've seen that as well, but I cant really understand how it would be implemented. And for the use case that I gave there doesn't really seem to be a good category since there isn't just a general hardware issue. 
It's also unclear how this would be implemented on google/apple's side, I have never seen an error message with a clear indication as to what the problem is in my google home app for example.

jonsmirl, 2023-05-15:
Google and Apple are not going to display your custom errors. You need to make a vendor commissioning app. Then you device can implement custom clusters and return anything you want to your vendor app. Of course Google/Apple will know nothing about this custom cluster so they will just ignore it.

dhrishi, 2023-05-24:
@stage-01 I don't think it should be possible to return custom status codes in the interaction model. It could either be a global interaction model status code or a cluster specific status code

closed 2023-05-24 by dhrishi