espressif/esp-matter #386: Color light example not working

stage-01, 2023-05-15:
When I use the color light example, or when I implement the extended color light myself, it shows up in google home as a color temperature light. I've tried both color mode 0 and 1 both give the same result.

Im using an esp32-c6.

stage-01, 2023-05-15:
Using chip-tool I am able to control the color control cluster, are there more people having trouble using extended color light 
with google home?

dhrishi, 2023-05-15:
> Using chip-tool I am able to control the color control cluster, are there more people having trouble using extended color light
with google home?

Are you able to control the clusters with chip-tool as expected for the device type you are working with? If yes, I would request you to instead raise an issue in upstream connectedhomeip for Google folks to take a look

stage-01, 2023-05-15:
yes chip-tool works. I'll raise an issue in the connectedhomeip repo, thanks for the reply!

closed 2023-05-15 by stage-01