espressif/esp-matter #387: ESP-Matter only for Wifi Comissioning

Tim4497, 2023-05-15:
I want to ask if it's possible to create a Matter device with an "empty" endpoint, so the only purpose of the matter device is to connect the ESP with wifi network trough google/apple home?

dhrishi, 2023-05-15:
> I found endpoint::aggregator::create(node, ENDPOINT_FLAG_NONE, NULL); to create a bridge type (which would be a good endpoint, but Apple Home is saying This accessory is not responsing, and when I remove the bridge, the matter device is not going back into commissioning mode, what am I missing?

I didn't quite understand what are you trying to achieve over here? Is it the same requirement about using it only for getting connected to the IP network? Or which bridge are you trying to implement?

jonsmirl, 2023-05-15:
Check out the BLE Mesh bridge example. https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge
That should do what you want, just don't add any devices to the bridge.

Tim4497, 2023-05-15:
> It isn't possible. A matter node requires the endpoint 0 (root node).

Ah yes of course I have still the root node, but no device type endpoint. I tried it and it will commission the device, but Apple will tell the user its not properly set up, thats why asked for another empty endpoint solution.

>  Is it the same requirement about using it only for getting connected to the IP network?

Yes, I thought it might be better to use the bridge for that requirement if there is no solution via an endpoint? @dhrishi 

> That should do what you want, just don't add any devices to the bridge. 

Thanks, I tried to do it and I got the initial commissioning, but after that the bridge is not responding. I still have the case if I "delete" the bridge from my phone, that I can re-commission it in another home/ wifi.

jonsmirl, 2023-05-15:
There are unresolved issues around devices not responding after first commissioning.  If you wait for a while they will resolve, otherwise just reboot the device.

jonsmirl, 2023-05-15:
If you delete the last commissioned fabric on the device, the device should revert back to Bluetooth commissioning mode.

Tim4497, 2023-05-15:
My ESP doesn’t react when I delete the bridge in Apple Home, any suggestions? Or could you elaborate further? Maybe there is something I missing!

jonsmirl, 2023-05-15:
Watch the logs on the ESP device. When you delete the bridge there should be a bunch of traffic back to the device doing the deletion. At the end of that it should say the last fabric was deleted and then revert to BLE advertising for commissioning.

Tim4497, 2023-05-16:
> Watch the logs on the ESP device. When you delete the bridge there should be a bunch of traffic back to the device doing the deletion. At the end of that it should say the last fabric was deleted and then revert to BLE advertising for commissioning.

Negative, I get no traffic when I delete the bridge, but I get traffic when I reconnect the Bridge after deleting it. Any advice?

But thanks for the help so far, commissioning is stable!

jonsmirl, 2023-05-16:
The only way there would be no traffic on the Apple side is if Apple thinks the bridge is dead when you delete it. If they think it is dead, they just abandon it and don't really delete it.

What should be going on - You delete bridge in IOS. IOS then removes the commissioning from the device. If that was last commissioning on the device, then the device reverts back to BLE advertising for commissioning.

Tim4497, 2023-05-24:
@jonsmirl Seems like it's on the Apple side, as I can’t get it to work. Is there another way to delete the wifi credentials or de-commission the device another way? maybe when I remove the device from power?

jonsmirl, 2023-05-24:
Do a factory reset? 

Apple put those credentials there, when you delete the device Apple is supposed to remove them.  File a bug on Apple. I suspect this is not well tested in Apple because I don't think anyone has shipped a Matter bridge product yet.

shubhamdp, 2023-07-26:
@aggaddam Device shall use BLE commissioning only if it does not have an IP connection. Depending on the use case you can open [enhanced](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.h#L90C16-L90C47) or [basic](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.h#L78)  commissioning window. Enhanced will always advertise on mdns. For basic, there's an option `advertisementMode` for configuring the advertising [mode](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/CommissioningWindowManager.h#L34).

aggaddam, 2023-07-26:
@shubhamdp Is ESP-Matter SDK maintaining the status of the commissioning process or do I need to implement it in the application? Are there APIs"s to know the device is commissioned or not or any other state it is ?

shubhamdp, 2023-07-26:
One way to know if device is commissioned or not is to check the fabric count, if its greater than zero then device is commissioned. ([ref impl](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L78C74-L78C74))

aggaddam, 2023-07-26:
Can I open the BLE commission window instead of kDnssdOnly if the fabric count is 0?

CHIP_ERROR err = commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds,
                                                    chip::CommissioningWindowAdvertisement::kAllSupported);

shubhamdp, 2023-07-26:
[`esp_matter::start()`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L997) is not just about putting device into commissioning mode, it does a lot of initialization, and one of its part is to put the device into commissioning mode.

Decision to put the in commissioning mode is in connectedhomeip repo's Server.cpp, please see these [bits](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L997).

shubhamdp, 2023-08-11:
@Tim4497 If your question is answered can you please close this issue?

closed 2023-08-21 by Tim4497