espressif/esp-matter #388: Device reboots

stefano055415, 2023-05-15:
@dhrishi 

after commit ad126a98dfda9c2988d24efc78b381e76294458a the device reboots

dhrishi, 2023-05-16:
@stefano055415 Which example are you trying out, and on which branch? ESP-IDF version? Have you changed any configuration options? Can you provide these details.

I would recommend you to clean the example build and give it a try again.
```
$ cd /path/to/examples/your_example
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py erase-flash build flash monitor
```
 Please upload (and not paste) the entire device side logs if you still see the issue.

stefano055415, 2023-05-16:
@dhrishi 

ok i found the problem, i had to add

CONFIG_FACTORY_COMMISSIONABLE_DATA_PROVIDER=y
CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_FACTORY_DEVICE_INFO_PROVIDER=y

ivishaltejwani, 2023-05-18:
> CONFIG_FACTORY_DEVICE_INFO_PROVIDER=y

where did you add these?