espressif/esp-matter #389: Serial number field in Basic Information

jonsmirl, 2023-05-15:
Would it be possible to make a CONFIG option to fill the optional Serial-Number field of Basic Information cluster with the serial number inside the ESP32 chip? Don't store it in flash where it will be wiped out, get it from the chip.

Why? Because if you factory reset a device and then recommission, I want to purge the old entry out of my device database. And the only way I can identify those old entries is via the chip serial number.

jonsmirl, 2023-05-15:
Another option would be to put the ESP serial number in to the CSR the device generates during commissioning. Then when I am signing the certificate I can purge old certificates with the same serial number.

dhrishi, 2023-05-16:
@jonsmirl There isn't a serial number. Instead, each ESP chip is programmed with a unique MAC address that can be used. You can use [esp_efuse_mac_get_default function](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv425esp_efuse_mac_get_defaultP7uint8_t) to read it.

jonsmirl, 2023-05-16:
How do you get it into Basic Info / Serial-Number since you are in control of those attributes? Or the CSR?

shubhamdp, 2023-05-23:
One option is to override [`GetSerialNumber()`](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/DeviceInstanceInfoProvider.h#L116) method and set the custom provider using [`set_custom_device_instance_info_provider()`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_providers.h#L48)

dhrishi, 2023-06-24:
Closing this now. Please reopen in case of any issues

closed 2023-06-24 by dhrishi