espressif/esp-matter #390: Test Harness TC-CGEN-2.2

stefano055415, 2023-05-15:
During the test you are asked to add a new certificate

./chip-tool operatingcredentials add-trusted-root-certificate hex:...

But I get this error status: FAILSAFE_REQUIRED

ubuntu@ubuntu:~/apps$ ./chip-tool operationalcredentials add-trusted-root-certificate  hex:153001087348A52373B626912402013703271401000000CACACACA182604EF171B2726056EB5B94C3706271401000000CACACACA1824070124080130094104A72D2E5ADBEEF493DD9EB03BD82E992F69770AE18300A9B3B7A8B737002F439A29F01B783B27FE25878DCF79C60A522CBC2A3FCD44720AEB2ADA059F4DF21D4A370A3501290118240260300414D4372741B1443BB18E7EB52868EF1520BB33D469300514D4372741B1443BB18E7EB52868EF1520BB33D46918300B4053E77C21954E9F0DFD0B408DC03FB87D11EF47D34509FC7D13E809B4726A39FEE68B573FA67CBA2D1D71B83C3B2B092038FE9D5CF1BA8180F5950106E396045C18 1 0
[1684164411.555003][76628:76628] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1684164411.561560][76628:76628] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1684164411.561869][76628:76628] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1684164411.561998][76628:76628] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1684164411.562404][76628:76628] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-NvxhHL)
[1684164411.563085][76628:76628] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1684164411.563147][76628:76628] CHIP:DL: NVS set: chip-counters/reboot-count = 27 (0x1B)
[1684164411.564080][76628:76628] CHIP:DL: Got Ethernet interface: eth0
[1684164411.564723][76628:76628] CHIP:DL: Found the primary Ethernet interface:eth0
[1684164411.565466][76628:76628] CHIP:DL: Got WiFi interface: wlan0
[1684164411.565546][76628:76628] CHIP:DL: Failed to reset WiFi statistic counts
[1684164411.565602][76628:76628] CHIP:IN: UDP::Init bind&listen port=0
[1684164411.565730][76628:76628] CHIP:IN: UDP::Init bound to port=47160
[1684164411.565765][76628:76628] CHIP:IN: BLEBase::Init - setting/overriding transport
[1684164411.565786][76628:76628] CHIP:IN: TransportMgr initialized
[1684164411.565832][76628:76628] CHIP:FP: Initializing FabricTable from persistent storage
[1684164411.566000][76628:76628] CHIP:TS: Last Known Good Time: 2023-01-28T06:29:17
[1684164411.567706][76628:76628] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xCFF336499D3FCDA6, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1684164411.571515][76628:76628] CHIP:ZCL: Using ZAP configuration...
[1684164411.576239][76628:76628] CHIP:DL: Avahi client registered
[1684164411.577732][76628:76628] CHIP:CTL: System State Initialized...
[1684164411.577866][76628:76628] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1684164411.577928][76628:76628] CHIP:CTL: Setting attestation nonce to random value
[1684164411.577989][76628:76628] CHIP:CTL: Setting CSR nonce to random value
[1684164411.578343][76628:76630] CHIP:DL: CHIP task running
[1684164411.578587][76628:76630] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1684164411.578920][76628:76630] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1684164411.578970][76628:76630] CHIP:CTL: Setting attestation nonce to random value
[1684164411.579158][76628:76630] CHIP:CTL: Setting CSR nonce to random value
[1684164411.580144][76628:76630] CHIP:CTL: Generating NOC
[1684164411.581213][76628:76630] CHIP:FP: Validating NOC chain
[1684164411.583334][76628:76630] CHIP:FP: NOC chain validation successful
[1684164411.583519][76628:76630] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1684164411.583555][76628:76630] CHIP:TS: Last Known Good Time: 2023-01-28T06:29:17
[1684164411.583581][76628:76630] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1684164411.583605][76628:76630] CHIP:TS: Retaining current Last Known Good Time
[1684164411.595866][76628:76630] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1684164411.600617][76628:76630] CHIP:TS: Committing Last Known Good Time to storage: 2023-01-28T06:29:17
[1684164411.605514][76628:76630] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: CFF336499D3FCDA6)
[1684164411.631284][76628:76630] CHIP:TOO: Sending command to node 0x1
[1684164411.631819][76628:76630] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000001]
[1684164411.631858][76628:76630] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1684164411.631899][76628:76630] CHIP:CTL: OperationalSessionSetup[1:0000000000000001]: State change 1 --> 2
[1684164411.631936][76628:76630] CHIP:DIS: Resolving CFF336499D3FCDA6:0000000000000001 ...
[1684164411.634094][76628:76630] CHIP:DL: Avahi resolve found
[1684164411.634174][76628:76630] CHIP:DIS: Node ID resolved for CFF336499D3FCDA6:0000000000000001
[1684164411.634201][76628:76630] CHIP:DIS: 	Hostname: C8F09E43BC40
[1684164411.634233][76628:76630] CHIP:DIS: 	IP Address #1: fe80::caf0:9eff:fe43:bc40
[1684164411.634260][76628:76630] CHIP:DIS: 	Port: 5540
[1684164411.634286][76628:76630] CHIP:DIS: 	Mrp Interval idle: not present
[1684164411.634310][76628:76630] CHIP:DIS: 	Mrp Interval active: not present
[1684164411.634335][76628:76630] CHIP:DIS: 	TCP Supported: 1
[1684164411.634472][76628:76630] CHIP:DIS: UDP:[fe80::caf0:9eff:fe43:bc40%eth0]:5540: new best score: 3
[1684164411.634504][76628:76630] CHIP:DIS: Checking node lookup status after 3 ms
[1684164411.634528][76628:76630] CHIP:DIS: Keeping DNSSD lookup active
[1684164411.832270][76628:76630] CHIP:DIS: Checking node lookup status after 200 ms
[1684164411.832601][76628:76630] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:[fe80::caf0:9eff:fe43:bc40%eth0]:5540 while in state 2
[1684164411.832666][76628:76630] CHIP:CTL: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
[1684164411.832870][76628:76630] CHIP:IN: SecureSession[0xffff8400b2e0]: Allocated Type:2 LSID:54602
[1684164411.832963][76628:76630] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000001
[1684164411.834845][76628:76630] CHIP:EM: <<< [E:20517i M:76803348] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1684164411.835173][76628:76630] CHIP:IN: (U) Sending msg 76803348 to IP address 'UDP:[fe80::caf0:9eff:fe43:bc40%eth0]:5540'
[1684164411.835491][76628:76630] CHIP:SC: Sent Sigma1 msg
[... 49 lines cut ...]
[1684164414.643656][76628:76630] CHIP:DMG: 		{
[1684164414.643718][76628:76630] CHIP:DMG: 			CommandStatusIB =
[1684164414.643798][76628:76630] CHIP:DMG: 			{
[1684164414.643868][76628:76630] CHIP:DMG: 				CommandPathIB =
[1684164414.643947][76628:76630] CHIP:DMG: 				{
[1684164414.644022][76628:76630] CHIP:DMG: 					EndpointId = 0x0,
[1684164414.644107][76628:76630] CHIP:DMG: 					ClusterId = 0x3e,
[1684164414.644186][76628:76630] CHIP:DMG: 					CommandId = 0xb,
[1684164414.644262][76628:76630] CHIP:DMG: 				},
[1684164414.644349][76628:76630] CHIP:DMG: 				
[1684164414.644418][76628:76630] CHIP:DMG: 				StatusIB =
[1684164414.644548][76628:76630] CHIP:DMG: 				{
[1684164414.644702][76628:76630] CHIP:DMG: 					status = 0xca (FAILSAFE_REQUIRED),
[1684164414.644807][76628:76630] CHIP:DMG: 				},
[1684164414.644896][76628:76630] CHIP:DMG: 				
[1684164414.644964][76628:76630] CHIP:DMG: 			},
[1684164414.645042][76628:76630] CHIP:DMG: 			
[1684164414.645104][76628:76630] CHIP:DMG: 		},
[1684164414.645176][76628:76630] CHIP:DMG: 		
[1684164414.645232][76628:76630] CHIP:DMG: 	],
[1684164414.645303][76628:76630] CHIP:DMG: 	
[1684164414.645360][76628:76630] CHIP:DMG: 	InteractionModelRevision = 1
[1684164414.645417][76628:76630] CHIP:DMG: },
[1684164414.645552][76628:76630] CHIP:DMG: Received Command Response Status for Endpoint=0 Cluster=0x0000_003E Command=0x0000_000B Status=0xca
Segmentation fault (core dumped)

wqx6, 2023-05-16:
`add-trusted-root-certificate` command is fail-safe-required. So when you send this command, please ensure that the device is fail-safe state. You can send `arm-fail-safe` command with a longer timeout(eg. 900 seconds) or resend the `arm-fail-safe` command before you send `add-trusted-root-certificate` command.

stefano055415, 2023-05-16:
@wqx6 

ok it works thanks

closed 2023-05-16 by stefano055415