espressif/esp-matter #391: commissioning error on Esp32 all-cluster-app

aravindone, 2023-05-16:
Dear devs .

Im facing a issue on the commissioning part of esp 32 with client has chip-tool and app on esp 32 is all-clusters-app
While i pass
$ ./chip-tool pairing ble-wifi <node_id> <pin_code>

im facing opcreds based error.

Here are the log files.

https://raw.githubusercontent.com/aravindone/aravindone/main/ble-wifilog

dhrishi, 2023-05-16:
@aravindone The log shows that the Wi-Fi add network command response is received as invalid on the chip-tool, which is very strange.

Can you please provide the Git head commit and the example you are using? The comment says all-clusters-app but the logs look like that of the lighting app. Do you have any changes on top of the example? I would suggest to recompile and re-flash the device and give it a try again:

```
cd /path/to/example
rm -rf sdkconfig sdkconfig.old build
idf.py erase-flash build flash monitor
```

Also, please check if the chip-tool is compiled on the same commit, and is not very old (for consistency)

dhrishi, 2023-05-24:
@aravindone Please close this if your issue is resolved.

aravindone, 2023-05-24:
the issue is resolved after using a direct ubuntu laptop

closed 2023-05-24 by aravindone