espressif/esp-matter #394: Ethernet headers.

ronny-antoon, 2023-05-17:
iam trying to control matter accessories over ethernet, not wifi. 
uncheck "Enable WiFi Telemetry"
check "Enable Ethernet Telemetry"

when i build the project i got error :-
`
../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_Ethernet.cpp: In member function 'CHIP_ERROR chip::DeviceLayer::ConnectivityManagerImpl::InitEthernet()':
../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_Ethernet.cpp:60:5: error: 'eth_mac_config_t' was not declared in this scope
   60 |     eth_mac_config_t mac_config  = ETH_MAC_DEFAULT_CONFIG();
`

PSONALl, 2023-05-19:
@ronny-antoon I am not able to reproduce this issue,
Can you please check if you have updated your submodules
Also, can you share the IDF version you are using, the esp-matter and connectedhome ip commit ids, and complete device logs?

dhrishi, 2023-05-24:
@ronny-antoon Please provide all the necessary details about the esp-idf and esp-matter versions/GIT commits, the example you are trying out and your sdkconfig. If your issue is resolved, then please close this

ronny-antoon, 2023-05-24:
resolved by force include some headers, I am beginner so I don't know how to get this details, versions and commits, sorry.
thank you anyway.

closed 2023-05-24 by ronny-antoon