espressif/esp-matter #396: Not working without internet

ivishaltejwani, 2023-05-18:
Problem : matter accessory works when internet connection is available 

when I removed ethernet wire from the router basically simulating no internet connection ,
 **matter device on Apple home and google home showed no response** ( note on apple home it worked for few minutes after no internet )

Processor : ESP32D

tested this issue extensively and reproduced it few times same fault

Even commissioning doesn't work on google home without internet , On Apple home I am able to commission and add device but it says no response Homepod is on and connected to wifi )

wqx6, 2023-05-19:
Internet is required for both the Apple fabric and the Google Fabric, The phone and the smart speaker should be connected to internet so that the command can be sent from the phone to the controller(nesthub or homepod mini) via the cloud.

ivishaltejwani, 2023-05-19:
> Internet is required for both the Apple fabric and the Google Fabric, The phone and the smart speaker should be connected to internet so that the command can be sent from the phone to the controller(nesthub or homepod mini) via the cloud.

even when theyre in the same network? I am talking about when the hubs , devices and phone are all on local and same network

ivishaltejwani, 2023-05-19:
> even when theyre in the same network? I am talking about when the hubs , devices and phone are all on local and same network

https://csa-iot.org/all-solutions/matter/matter-faq/

in the faqs it is mentioned that matter is a local protocol and doesnt require internet to function locally only when youre out side of the network

jonsmirl, 2023-05-19:
The problems you are seeing have nothing to do with Espressif, the design Google, Amazon and Apple Home are using works like this...

Your phone works anywhere inside house or in another country. That is because your phone is not talking directly to the Matter devices, instead it is talking to their cloud servers.  Those cloud servers provide the connectivity your phone needs in order to work outside the home. Inside your home you have a hub by GAA. Those hubs maintain a connection to the GAA cloud.  So when phone sends a command it goes to the GAA cloud, and then the GAA cloud relays it o the GAA hub in your house. And then that hub sends the Matter commands.

Why does it work this way? Two main reasons. 1) This allows you to control the devices from outside of the home 2) ALL of GAA's voice processing happens in the cloud. The only locally processed voice is the hot word detection.

It is possible for GAA to directly control the devices inside the house, but GAA is not putting much effort into doing that. Instead all of their effort is focused on the cloud. If you want a system focused on local control get Home Assistant.

ivishaltejwani, 2023-05-19:
@jonsmirl 

If the communication is happening over the cloud server , then whats the point of hub anyway.

example Apple's homekit - it works completely locally without the need of hub , only if you have to use it outside of your network the hub is necessary 

and isnt matter protocols main selling point is local control? For home assistant you need to setup a server and same goes for zigbee too 

so correct me if I am wrong ... matter on Apple home , google home or alexa doesnt work without internet and this is not a bug?

@dhrishi @shubhamdp

jonsmirl, 2023-05-19:
The hub is there so that the home does not need to have an open server on the internet. The hub establishes an outgoing connection to the cloud, and then the cloud hosts the server open to the Internet.  This protects the hub from the continuous attacks Internet exposed servers receive.

Matter is able to work inside the home without use of a hub. GAA just haven't gotten around to implementing that feature yet. But that is a problem with GAA's implementation so you should complain to them. The Espressif nodes have no say in this. They will work fine in either mode.

HA needs a hub because they bolted Matter onto the side of their existing hub based system. Again HA could eliminate the hub if they chose to, but that would entail a lot of code changes.

ivishaltejwani, 2023-05-19:
understood @jonsmirl , Thanks

closed 2023-05-22 by dhrishi