espressif/esp-matter #399: Radio regulatory region

jonsmirl, 2023-05-19:
If I write a country code into the Location field on the Basic Info cluster will the Espressif radio code pick that up and set the correct regulatory region?  Boris just told me that's what that field is supposed to do.

wqx6, 2023-05-22:
The `ConfigurationManagerImpl` in ESP32 platform does not override the functions `GenericConfigurationManagerImpl<ConfigClass>::GetCountryCode()` and `GenericConfigurationManagerImpl<ConfigClass>::StoreCountryCode()` in generic implement, so our platform should read and store the location in the flash.

jonsmirl, 2023-06-20:
https://issuetracker.google.com/issues/283470659

Google says the Google Home app is setting this, so Espressif code needs to respect it.

They are setting it from Google Home, not the GPS commissioning library. I'm trying to get them to fix that.

shubhamdp, 2023-06-21:
@jonsmirl Raised the PR here: https://github.com/project-chip/connectedhomeip/pull/27365

jonsmirl, 2023-06-23:
There is also problem in Matter core. You should be getting this from the SetRegulatoryConfig command, not the attribute.

https://github.com/project-chip/connectedhomeip/issues/27445

shubhamdp, 2023-06-25:
Okay, I fixed the function on `platform/ESP32` which sets the country code, `StoreCountryCode()`.
Now, this functions is called in two flows: 1. Write on `Location` attribute and 2. Invoke on `SetRegulatoryConfig` command.

Where is the problem here, I do not see one, if you have more details please share.

jonsmirl, 2023-06-25:
It is a problem with the CHIP spec. There should not have been two ways to set this. What happened was Google and I picked each picked a different way to do it. Google picked SetRegulatoryConfig and I picked Location.   We each did not know about the other way to set it so we did not account for the other method.

The correct fix is for Matter core to make that Location attribute ReadOnly and add a note saying it is set via the SetRegulatoryConfig command.

shubhamdp, 2023-08-11:
@jonsmirl This is a problem with Specification right? And Implementation is in place for what is there in specification right now, so, can we close this issue which fixes setting the country in PHY layer?

closed 2023-08-11 by jonsmirl

shubhamdp, 2023-08-14:
I'll put up an issue in specification repo