espressif/esp-matter #401: The thermostat is having problems with homekit

maenkai, 2023-05-20:
1. Switching the automatic mode causes homekit APP to crash
2. The temperature cannot be adjusted in the cooling and heating modes

code 

```c
   node::config_t node_config;
   node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

   thermostat::config_t thermostat_config;
   endpoint_t *endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);
   cluster_t *cluster = cluster::get(endpoint,Thermostat::Id);
   
   //heat
   cluster::thermostat::feature::heating::config_t heat_config;
   cluster::thermostat::feature::heating::add(cluster,&heat_config);

    //cool 
   cluster::thermostat::feature::cooling::config_t cool_config;
   cluster::thermostat::feature::cooling::add(cluster,&cool_config);    

    //auto
   cluster::thermostat::feature::auto_mode::config_t auto_config;
   cluster::thermostat::feature::auto_mode::add(cluster,&auto_config);

maenkai, 2023-05-20:
esp-matter: commit-id = 056c0b1e90f52fc6dce8217cf4faad87bc21f66d
esp-idf: v4.4.3

maenkai, 2023-05-20:
error:
[image]

Jerry-ESP, 2023-05-22:
> 

@maenkai I can't find this commit id : 056c0b1e90f52fc6dce8217cf4faad87bc21f66d in esp-matter, please make sure the commit id is correct; and I tested this example in latest esp-matter, it can work well, maybe you can have a try, we fixed some issues about sensors before.
And maybe the IOS, Homekit and Homepod version should be update to latest.
IOS: 16.5beta, Homepod: 16.5
My esp-matter commit: b0af236c1b411c8cb813228efc5d644acb1e3d99
idf: v5.0.1

maenkai, 2023-05-23:
> > 
> 
> @maenkai我在 esp-matter 中找不到此提交 ID：056c0b1e90f52fc6dce8217cf4faad87bc21f66d，请确保提交 ID 正确；我在最新的esp-matter中测试了这个例子，它可以很好地工作，也许你可以试试，我们之前修复了一些关于传感器的问题。 也许 IOS、Homekit 和 Homepod 版本应该更新到最新版本。 IOS：16.5beta，Homepod：16.5 我的 esp-matter 提交：[b0af236](https://github.com/espressif/esp-matter/commit/b0af236c1b411c8cb813228efc5d644acb1e3d99) idf：v5.0.1

Thanks, I updated the ios version to fix the problem

closed 2023-05-23 by maenkai