espressif/esp-matter #404: OTA Failed

SYESHASWINI, 2023-05-22:
While the ota process is happening, after downloading few packets it throws the error as:
>E (375507) chip[EM]: Failed to Send CHIP MessageCounter:121761502 on exchange 10803i sendCount: 4 max retries: 4

Jerry-ESP, 2023-05-23:
> While the ota process is happening, after downloading few packets it throws the error as:
> 
> > E (375507) chip[EM]: Failed to Send CHIP MessageCounter:121761502 on exchange 10803i sendCount: 4 max retries: 4

The error log means that a message send failed.  Did this happen only once or every OTA process? 
And which chip do you use ? Please upload all logs for the OTA process.

SYESHASWINI, 2023-05-26:
@Jerry-ESP sorry for the delay. Attaching the error log.
[ota_error.txt](https://github.com/espressif/esp-matter/files/11574318/ota_error.txt)

Jerry-ESP, 2023-05-26:
> @Jerry-ESP sorry for the delay. Attaching the error log. [ota_error.txt](https://github.com/espressif/esp-matter/files/11574318/ota_error.txt)

Please tell me your idf version, esp-matter version, develop board version and ota test platform

SYESHASWINI, 2023-05-30:
idf version: ESP-IDF v4.4.2-dirty
esp-matter version: v1.0
dev board: esp32c3
ota test platform: Chip-tool

dhrishi, 2023-05-30:
@SYESHASWINI Please use ESP-IDF v5.0.1 or v4.4.4 if you are using the main branch.

shubhamdp, 2023-06-13:
@SYESHASWINI As we are already tracking OTA issues in https://github.com/project-chip/connectedhomeip/issues/26346 can you please close this one

closed 2023-06-13 by SYESHASWINI