espressif/esp-matter #405: mbedTLS error: CCM - Authenticated decryption failed

stage-01, 2023-05-22:
[device log](https://github.com/espressif/esp-matter/files/11531740/device.log)
after installing the latest version of both esp-idf and esp-matter I run into the following two issues during commisioning:

E (22076) chip[CR]: mbedTLS error: CCM - Authenticated decryption failed
E (22086) chip[IN]: Secure transport received message, but failed to decode/authenticate it, discarding

After throwing this error the commisioning stops.

shubhamdp, 2023-05-23:
What do you mean by latest? Can you provide more details about your setup like esp-idf and esp-matter commits or branch.

And if you have the commissioner side logs, then that would help as well.

marciogranzotto, 2023-05-29:
> By adding : CONFIG_MBEDTLS_HKDF_C=y
> on line 1543 to the sdkconfig the problem seems to be solved.

Thanks @stage-01 ! it also fixed my problem

dhrishi, 2023-05-31:
@stage-01 Thank you for the issue and the fix :-) It has been added on Github in https://github.com/espressif/esp-matter/commit/d0b81170f3a68eace4d689e9db4aac3a53ef151a

closed 2023-05-31 by dhrishi

moculll, 2023-06-02:
> 

you are my **ddy!!!fu***g hero!!!