espressif/esp-matter #408: What this code does

maenkai, 2023-05-24:
esp-matter/examples/generic_switch/app_main.cpp
What are these values used for, and where in the code are they useful
```c
    nvs_handle_t handle;
    nvs_open_from_partition(CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL, "chip-factory", NVS_READWRITE, &handle);

    int32_t out_value = 0;
    if (nvs_get_i32(handle, "fl-sz/1", &out_value) == ESP_ERR_NVS_NOT_FOUND)
    {
       nvs_set_i32(handle, "fl-sz/1", 2);
       nvs_set_str(handle, "fl-k/1/0", "myEP1LBL1");
       nvs_set_str(handle, "fl-v/1/0", "valEP1LBL1");
       nvs_set_str(handle, "fl-k/1/1", "myEP1LBL2");
       nvs_set_str(handle, "fl-v/1/1", "valEP1LBL2");
    }

    nvs_commit(handle);
    nvs_close(handle);

shripad621git, 2023-05-24:
As mentioned in the README, the generic_switch example demonstrates the use of fixed-labels. Right now,there are two ways of using the fixed-labels . 1. using nvs and 2. using using Example implementation of [device info provider](https://github.com/espressif/connectedhomeip/tree/4088a77f557e8571a39338fad51a1d8eb0131d79/examples/providers).The code snippet above adds two entries for fixed-labels using nvs on endpiont 1. For more detailed information on the API's used , you can refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#api-reference..

maenkai, 2023-05-24:
I would like to know the function of a value like "fl-sz/1", is there any documentation for that

maenkai, 2023-05-24:
> 如 README 中所描述，generic_switch 说明了固定标签的使用。目前，有两种使用固定标签的方法。1. 使用 nvs 和 2. 使用[设备信息提供程序](https://github.com/espressif/connectedhomeip/tree/4088a77f557e8571a39338fad51a1d8eb0131d79/examples/providers)的演示实例。上面的代码片段在结尾 1 上使用nvs添加了两个固定标签条目。有相关使用API​​的更多详细信息，您可以参考[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/ nvs_flash.html#api-参考](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#api-reference)..

I would like to know the function of a value like "fl-sz/1", is there any documentation for that

shripad621git, 2023-05-24:
You can get the reference for this here[https://github.com/project-chip/connectedhomeip/blob/e022057aea58fee7f855d8994c7f971c5ac2c875/src/platform/ESP32/ESP32Config.h#L168](https://github.com/project-chip/connectedhomeip/blob/e022057aea58fee7f855d8994c7f971c5ac2c875/src/platform/ESP32/ESP32Config.h#L168).

dhrishi, 2023-05-24:
@maenkai So, basically:
```
fl - fixed label
sz - size
k - key
v -value
```

closed 2023-05-24 by maenkai