espressif/esp-matter #411: callback to read the attributes of a subscription

glennmullerar, 2023-05-24:
Hello, 

I'm trying to design a controller that can subscribe to an attribute and process the value of the freshly modified attribute. 

My question is the following, is it possible to use a callback when to be able to process the attribute to which the controller has subscribed? 

Or is there an easier way? 

After a lot of research the only similar function I found is this one: 

``client::set_command_callback``

Thanks in advance for your feedback 

PS: the desired treatment is simply the insertion of these attributes in a database.

wqx6, 2023-05-31:
We are planning to add these report callbacks for [controller example](https://github.com/espressif/esp-matter/tree/main/examples/controller). The attributes could be decoded and printed in [OnAttributeData](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/esp_matter_controller_subscribe_command.cpp#L102). A callback will be added here so that you can create a command object with the callbacks

glennmullerar, 2023-05-31:
Finally, I created a clone of the ```subscribe_command``` class, which also inherits from the ```ReadClient::Callback``` class. 

And I modified the ```OnAttributeData``` function. 

I don't know if this is the best way to do it, but it works for me.

glennmullerar, 2023-06-02:
@wqx6 Do you know if there is a way to retrieve the node-id of the node which the controller has subscribed to, in the OnAttributeData function?

closed 2023-06-12 by espressif-bot