espressif/esp-matter #413: BLE Broadcasting Problem in Light App

SeherMavi, 2023-05-25:
I am trying to commission the Light app. Initially, ChipTool-Android was seeing the device but was giving an error in the "pairing" step. Now it doesn't see the device at all. I also use ChipTool in Linux terminal, it doesn't see the device either. I'm pretty sure the device isn't turning on BLE broadcasting. I deleted all packages and SDK and reinstalled, I got a rebuild but it doesn't fix. How can I open BLE broadcast? The logs are as follows.

Device Log;

`entry 0x40080694
I (26) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (26) boot: compile time 14:31:09
I (27) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (74) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (81) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (89) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (96) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (104) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (111) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (119) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (126) boot: End of partition table
I (131) boot: No factory image, trying OTA 0
I (136) boot_comm: chip revision: 1, min. application chip revision: 0
I (143) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=2dfa0h (188320) map
I (219) esp_image: segment 1: paddr=0004dfc8 vaddr=3ffbdb60 size=02050h (  8272) load
I (223) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=ee7e4h (976868) map
I (579) esp_image: segment 3: paddr=0013e80c vaddr=3ffbfbb0 size=036b0h ( 14000) load
I (584) esp_image: segment 4: paddr=00141ec4 vaddr=40080000 size=1cd1ch (118044) load
I (633) esp_image: segment 5: paddr=0015ebe8 vaddr=50000000 size=00010h (    16) load
I (648) boot: Loaded app from partition at offset 0x20000
I (675) boot: Set actual ota_seq=1 in otadata[0]
I (675) boot: Disabling RNG early entropy source...
I (686) cpu_start: Pro cpu up.
I (686) cpu_start: Starting app cpu, entry point is 0x400813e4
0x400813e4: call_start_cpu1 at /home/seher/ESP-NEW/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (702) cpu_start: Pro cpu start user code
I (702) cpu_start: cpu freq: 160000000
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     light
I (711) cpu_start: App version:      v1.0
I (716) cpu_start: Compile time:     May 25 2023 08:31:17
I (722) cpu_start: ELF file SHA256:  f68ab6aebe290dbc...
I (728) cpu_start: ESP-IDF:          v4.4.2
I (733) heap_init: Initializing. RAM available for dynamic allocation:
I (740) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (746) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (752) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (759) heap_init: At 3FFD4BA0 len 0000B460 (45 KiB): DRAM
I (765) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (771) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (777) heap_init: At 4009CD1C len 000032E4 (12 KiB): IRAM
I (785) spi_flash: detected chip: gd
I (788) spi_flash: flash io: dio
W (792) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (808) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (971) led_driver_gpio: Initializing light driver
I (971) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (1001) esp_matter_core: Command 0x00000047 on cluster 0x00000300 already exists. Not creating again.
I (1001) app_main: Light created with endpoint_id 1
I (1011) wifi:wifi driver task: 3ffdcd54, prio:23, stack:6656, core=0
I (1011) system_api: Base MAC address is not set
I (1011) system_api: read default base MAC address from EFUSE
I (1031) wifi:wifi firmware version: eeaa27d
I (1031) wifi:wifi certification version: v7.0
[... 218 lines cut ...]
[1684997384.858004][13811:13813] CHIP:IN: UDP::Init bound to port=5550
[1684997384.858006][13811:13813] CHIP:IN: UDP::Init bind&listen port=5550
[1684997384.858014][13811:13813] CHIP:IN: UDP::Init bound to port=5550
[1684997384.858017][13811:13813] CHIP:IN: TransportMgr initialized
[1684997384.861248][13811:13813] CHIP:CTL: Setting wifi credentials from parameters
[1684997384.861255][13811:13813] CHIP:CTL: Setting attestation nonce to random value
[1684997384.861260][13811:13813] CHIP:CTL: Setting CSR nonce to random value
[1684997384.861266][13811:13813] CHIP:CTL: Commission called for node ID 0x0000000000007283
[1684997384.862193][13811:13812] CHIP:DL: TRACE: Bus acquired for name C-35f3
[1684997384.863632][13811:13813] CHIP:DL: PlatformBlueZInit init success
[1684997384.864538][13811:13812] CHIP:BLE: BLE removing known devices.
[1684997384.864762][13811:13812] CHIP:BLE: BLE initiating scan.
[1684997384.988436][13811:13813] CHIP:DL: Long dispatch time: 127 ms, for event type 2
[1684997385.007681][13811:13812] CHIP:BLE: Device 31:CA:94:59:DB:0A does not look like a CHIP device.
[1684997385.032577][13811:13812] CHIP:BLE: Device 07:C8:5B:B7:E3:BF does not look like a CHIP device.
[1684997385.039295][13811:13812] CHIP:BLE: Device 11:AE:DE:B8:8C:86 does not look like a CHIP device.
.
.
.
[1684997394.907789][13811:13812] CHIP:BLE: Device 31:CA:94:59:DB:0A does not look like a CHIP device.
[1684997394.968018][13811:13812] CHIP:BLE: Device 53:47:74:D3:F4:8C does not look like a CHIP device.
[1684997394.977929][13811:13812] CHIP:BLE: Device 6B:16:2A:95:DA:30 does not look like a CHIP device.
[1684997394.989239][13811:13813] CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:154: CHIP Error 0x00000032: Timeout
[1684997394.996405][13811:13812] CHIP:BLE: Scan complete. No matching device found.
`

dhrishi, 2023-05-25:
@SeherMavi Your device logs show that the device is already commissioned in the fabric:

> I (1431) chip[SVR]: Fabric already commissioned. Disabling BLE advertisement

After that, by default, BLE is disabled as it is not required. Please try to erase-flash and re-flash the device and retry commissioning if you want to try commissioning over BLE

Also, please upload (and not paste) the logs from next time.

closed 2023-05-25 by dhrishi

SeherMavi, 2023-05-25:
@dhrishi Thank you very much for your quick and useful answer. It works! 

I missed that part in the logs. Apart from that, there was no information such as "commissioning success" in the application on Android, or a new page was not opened where I could control the device. The last message of the application is "pairing", then the screen remains blank. I was expecting a button to turn the light on/off. Do you have any knowledge about this?

[LightApp.txt](https://github.com/espressif/esp-matter/files/11562156/LightApp.txt)

dhrishi, 2023-05-25:
@SeherMavi 

>Apart from that, there was no information such as "commissioning success" in the application on Android, or a new page was not opened where I could control the device.

It is like a sample testing tool and hence does not have all the UI based features. You need to explicitly navigate to a different tab for controlling the device.

>The last message of the application is "pairing", then the screen remains blank. I was expecting a button to turn the light on/off. Do you have any knowledge about this?

I do not have details about the current state of the app. You can instead use the ecosystem Home apps to try out commissioning. You would need a hub like a Nest Hub or HomePod for that

SeherMavi, 2023-05-25:
Dear @dhrishi ,  thank you a lot. I will try what you said :)