espressif/esp-matter #414: esp-matter as lib

maxkondor, 2023-05-25:
If I use esp-idf as library (main project used pure cmake and my setup described in esp_idf/examples/esp_idf_as_lib) can I use esp-matter wrapper with connnectedhomeip as library too?
I am trying to add that components to esp-idf, like` idf_build_component(path_to_esp_matter_and_their_dependencies)`. Then I pass that modules to `idf_build_process(...)` func and link it to my code. After fix of missing path components (like zap-cli) I have success cmake generation, but when I`m trying to build that, all builds in static libraries, trying to link and I have a linker error: `cannot find -lchip`. And in linker args I don`t see -L "path/to/libchip". 

So, tell me right way to do that. What I want: cmake library with esp-matter/matter functionality, but I don`t want to use esp-idf in all project as main skeleton.

shubhamdp, 2023-06-01:
Have you also ported/moved `connectedhomeip/connectedhomeip/config/esp32/components/chip` component?

shubhamdp, 2023-06-13:
@maxkondor are you still seeing this issue?

maxkondor, 2023-06-14:
HI. I fixed it. The problem was in cmake targets.

closed 2023-06-14 by maxkondor

marciogranzotto, 2023-10-20:
@maxkondor can you share the CMakeLists.txt part where you use it as a lib? I'm trying to do the same