espressif/esp-matter #415: The production information and serial number of the device that is bridged on homekit are not displayed properly

maenkai, 2023-05-26:
esp-matter: commit-id = 99bd741a09b03f70a70d9cead637840a1aa17d30

[image]

code
```c

  // product name
  char       name[]  = "传感器";
  cluster_t *cluster = cluster::get(device->endpoint, BridgedDeviceBasicInformation::Id);
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::NodeLabel::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(name, strlen(name)));
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::ProductName::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(name, strlen(name)));

  // label
  char pidlabel[] = "20X1";
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::ProductLabel::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(pidlabel, strlen(pidlabel)));

  // serial
  char serial[] = "1122233344";
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::SerialNumber::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(serial, strlen(serial)));

  // sofe version
  char Version[] = "1.0.0";
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::SoftwareVersionString::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(Version, strlen(Version)));

  // vid 
  char lipro[] = "lipro";
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::VendorName::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(lipro, strlen(lipro)));

  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::VendorID::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_uint16(0x1398));

  // manual
  char data[] = "hello lipro";
  attribute::create(cluster, BridgedDeviceBasicInformation::Attributes::ManufacturingDate::Id, (ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE),
                    esp_matter_char_str(data, strlen(data)));

Jerry-ESP, 2023-05-26:
I think you can't create the attribus twice, they had been created when you call ***node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);*** 
Can you paste all your app_main.cpp file that in order for us to analyze

maenkai, 2023-05-26:
1. The source code here is added in set_device_type in esp-matter-bridge.cpp
2. This is the bridge endpoint, so there are no duplications. These properties can be read normally using "chip-tool"
3. I wonder if the red box in the picture says "unknow" is a problem with IOS and homepod, or because I'm missing some attributes

maenkai, 2023-05-26:
> * The source code here is added in set_device_type in esp-matter-bridge.cpp
> * This is the bridge endpoint, so there are no duplications. These properties can be read normally using "chip-tool"
> * I wonder if the red box in the picture says "unknow" is a problem with IOS and homepod, or because I'm missing some attributes

1.The source code here is added in set_device_type in esp-matter-bridge.cpp
2.This is the bridge endpoint, so there are no duplications. These properties can be read normally using "chip-tool"
3.I wonder if the red box in the picture says "unknow" is a problem with IOS and homepod, or because I'm missing some attributes

maenkai, 2023-05-26:
> I think you can't create the attribus twice, they had been created when you call ***node_t _node = node::create(&node_config, app_attribute_update_cb, NULL);_** Can you paste all your app_main.cpp file that in order for us to analyze

1.The source code here is added in set_device_type in esp-matter-bridge.cpp
2.This is the bridge endpoint, so there are no duplications. These properties can be read normally using "chip-tool"
3.I wonder if the red box in the picture says "unknow" is a problem with IOS and homepod, or because I'm missing some attributes

YOGYUI, 2023-05-26:
I also have a same problem.
Before iOS 16.5, below attributes of 'BridgedDeviceBasicInformation' cluster are well displayed on Home app.
- BridgedDeviceBasicInformation::Attributes::VendorName::Id
- BridgedDeviceBasicInformation::Attributes::SerialNumber::Id
- ...
But after updating iOS 16.5, vendorName, SerialNumber and Model are displayed as 'Unknown' like your figure.
I guess this issue is related to Home app architecture, but I'm not sure.

jadhavrohit924, 2023-07-12:
This looks like an issue of apple. Have you guys tried same with Google?

shubhamdp, 2023-08-11:
@maenkai @YOGYUI any update here?

shripad621git, 2023-08-14:
I have tried the Brigde example on the latest main with your provided patch and it works fine on the home app.I am able to see the serial number and production information correctly. The  ipad os version used is ios 16.6.Can you please give it a try on the latest main and the specified ios version.

[image]

maenkai, 2023-08-15:
The latest version has indeed been fixed, but now the remaining bridge names are not displayed properly, as shown below

[image]
[image]

shripad621git, 2023-08-18:
@maenkai you can read the node label set using chip-tool.But it does not get reflected as the bridge name in homekit.
If you want to have a name for the bridge, the matter accessory label above is editable in the UI and you can set the name from there itself for identification purposes.

shripad621git, 2023-08-25:
@maenkai any updates on the issue faced . Please close this if the issue is solved.

maenkai, 2023-08-28:
> @maenkai you can read the node label set using chip-tool.But it does not get reflected as the bridge name in homekit. If you want to have a name for the bridge, the matter accessory label above is editable in the UI and you can set the name from there itself for identification purposes.

I used homeAPP to pair the gateway on the market, the default name is not matter accessory, how do I do that

maenkai, 2023-09-01:
This name is from DCL 
problem view https://github.com/project-chip/connectedhomeip/issues/26110

closed 2023-09-01 by maenkai