espressif/esp-matter #417: How do I trigger a ReachableCahnged Event

maenkai, 2023-05-26:
1. I want the display on homekit to refresh in real time after the bridge device dynamically creates an endpoint
2. I want the device to report back to homekit when it is offline, such as triggering a ReachableChangeEvent

jonsmirl, 2023-05-26:
The new endpoint creation should be modifying the Partslist in the Descriptor cluster of the Aggregate Endpoint to include the new endpoint. Apple will have a subscription watching that cluster and should see the change.  The esp-matter bridge code should be doing this for you.

maenkai, 2023-05-29:
> The new endpoint creation should be modifying the Partslist in the Descriptor cluster of the Aggregate Endpoint to include the new endpoint. Apple will have a subscription watching that cluster and should see the change. The esp-matter bridge code should be doing this for you.

During the test,ios APP always fails to refresh in time. Is there any way to make firmware actively push information

jonsmirl, 2023-05-29:
The refresh period is under control of IOS, you can't impact it.

maenkai, 2023-05-29:
I would like to know how to proactively report the offline status to the ios APP

maenkai, 2023-08-15:
Is there any way esp can cause homekit to trigger a refresh mechanism

dhrishi, 2023-09-18:
>Is there any way esp can cause homekit to trigger a refresh mechanism.

That is not possible as per the current specification

closed 2023-09-18 by dhrishi