espressif/esp-matter #418: I need to put multiple clusters in one endpoint

shun-dgut, 2023-05-26:
```
namespace generic_switch {
uint32_t get_device_type_id()
{
    return ESP_MATTER_GENERIC_SWITCH_DEVICE_TYPE_ID;
}

uint8_t get_device_type_version()
{
    return ESP_MATTER_GENERIC_SWITCH_DEVICE_TYPE_VERSION;
}

endpoint_t *create(node_t *node, config_t *config, uint8_t flags, void *priv_data)
{
    endpoint_t *endpoint = endpoint::create(node, flags, priv_data);
    return add(endpoint, config);
}

endpoint_t *add(endpoint_t *endpoint, config_t *config)
{
    if (!endpoint) {
        ESP_LOGE(TAG, "Endpoint cannot be NULL");
        return NULL;
    }
    add_device_type(endpoint, get_device_type_id(), get_device_type_version());

    descriptor::create(endpoint, CLUSTER_FLAG_SERVER);
    identify::create(endpoint, &(config->identify), CLUSTER_FLAG_SERVER);
    switch_cluster::create(endpoint, &(config->switch_cluster), CLUSTER_FLAG_SERVER);

    humidity_measurement::create(endpoint, &(config->humidity_measurement), CLUSTER_FLAG_SERVER);
    temperature_measurement::create(endpoint, &(config->temperature_measurement), CLUSTER_FLAG_SERVER);
    return endpoint;
}
} /* generic_switch */
```
This seems to be the wrong code, how should I modify

Jerry-ESP, 2023-05-26:
> ```
> namespace generic_switch {
> uint32_t get_device_type_id()
> {
>     return ESP_MATTER_GENERIC_SWITCH_DEVICE_TYPE_ID;
> }
>
> uint8_t get_device_type_version()
> {
>     return [...]

namespace generic_switch {
typedef struct config {
    cluster::identify::config_t identify;
    cluster::switch_cluster::config_t switch_cluster;
} config_t;

The generic switch struct have no member humidity_measurement and temperature_measurement

Generally, to add additional attributes in esp-matter, you would do the following, you can refer to the light_switch example:
[image]

jonsmirl, 2023-05-26:
You have to make three endpoints here or none of Google/Apple/Amazon are going to understand your device. Three endpoints with these three device types.
1) Generic Switch, 0x000F
2) Temp sensor, 0x0302
3) Humidity Sensor, 0x0307

Now a thermostat will be able to bind to these sensors.

shun-dgut, 2023-05-29:
> You have to make three endpoints here or none of Google/Apple/Amazon are going to understand your device. Three endpoints with these three device types.
> 
> 1. Generic Switch, 0x000F
> 2. Temp sensor, 0x0302
> 3. Humidity Sensor, 0x0307
> 
> Now a thermostat will be able to bind to these sensors.

Can I achieve my goal like this

```
temperature_sensor::config_t temperature_sensor_conf;
bridged_device->endpoint = temperature_sensor::add(bridged_device->endpoint, &temperature_sensor_conf);
humidity_sensor::config_t humidity_sensor_conf;
bridged_device->endpoint = humidity_sensor::add(bridged_device->endpoint, &humidity_sensor_conf);
```

shun-dgut, 2023-05-29:
Only one data can be displayed on google home, such as humidity, but temperature and humidity can be displayed on apple home.

dhrishi, 2023-05-30:
@shun-dgut Multiple clusters on one endpoint are possible through the APIs. How to interpret/understand them is on the commissioner phone app that is used. I will close this issue now

closed 2023-05-30 by dhrishi