espressif/esp-matter #420: light_sensor shows no response on apple home

shun-dgut, 2023-05-29:
I tried to increase the light sensor, but the apple home display is not responding.
Using chiptool to read the data, the data cannot be parsed. The log is as follows
```
illuminancemeasurement read  measured-value 0x7283 0x2
[1685330495.529337][31915:31920] CHIP:TOO: Sending command to node 0x7283
[1685330495.529493][31915:31920] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000007283]
[1685330495.529508][31915:31920] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1685330495.529528][31915:31920] CHIP:CTL: Found an existing secure session to [1:0000000000007283]!
[1685330495.529536][31915:31920] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1685330495.529570][31915:31920] CHIP:TOO:      cluster 0x0000_0400, attribute: 0x0000_0000, endpoint 2
[1685330495.529582][31915:31920] CHIP:TOO: Sending ReadAttribute to:
[1685330495.529595][31915:31920] CHIP:DMG: SendReadRequest ReadClient[0x7f05e4010540]: Sending Read Request
[1685330495.529718][31915:31920] CHIP:EM: <<< [E:29968i M:71041587] (S) Msg TX to 1:0000000000007283 [BEA2] --- Type 0001:02 (IM:ReadRequest)
[1685330495.529739][31915:31920] CHIP:IN: (S) Sending msg 71041587 on secure session with LSID: 11259
[1685330495.529857][31915:31920] CHIP:DMG: MoveToState ReadClient[0x7f05e4010540]: Moving to [AwaitingIn]
[1685330495.735078][31915:31920] CHIP:EM: >>> [E:29968i M:128230097 (Ack:71041587)] (S) Msg RX from 1:0000000000007283 [BEA2] --- Type 0001:05 (IM:ReportData)
[1685330495.735118][31915:31920] CHIP:EM: Found matching exchange: 29968i, Delegate: 0x7f05e4010550
[1685330495.735137][31915:31920] CHIP:EM: Rxd Ack; Removing MessageCounter:71041587 from Retrans Table on exchange 29968i
[1685330495.735172][31915:31920] CHIP:DMG: ReportDataMessage =
[1685330495.735183][31915:31920] CHIP:DMG: {
[1685330495.735190][31915:31920] CHIP:DMG:      AttributeReportIBs =
[1685330495.735203][31915:31920] CHIP:DMG:      [
[1685330495.735210][31915:31920] CHIP:DMG:              AttributeReportIB =
[1685330495.735222][31915:31920] CHIP:DMG:              {
[1685330495.735229][31915:31920] CHIP:DMG:                      AttributeDataIB =
[1685330495.735239][31915:31920] CHIP:DMG:                      {
[1685330495.735248][31915:31920] CHIP:DMG:                              DataVersion = 0xdd4a96ba,
[1685330495.735262][31915:31920] CHIP:DMG:                              AttributePathIB =
[1685330495.735272][31915:31920] CHIP:DMG:                              {
[1685330495.735281][31915:31920] CHIP:DMG:                                      Endpoint = 0x2,
[1685330495.735289][31915:31920] CHIP:DMG:                                      Cluster = 0x400,
[1685330495.735298][31915:31920] CHIP:DMG:                                      Attribute = 0x0000_0000,
[1685330495.735306][31915:31920] CHIP:DMG:                              }
[1685330495.735319][31915:31920] CHIP:DMG:
[1685330495.735329][31915:31920] CHIP:DMG:                              Data = 0,
[1685330495.735335][31915:31920] CHIP:DMG:                      },
[1685330495.735345][31915:31920] CHIP:DMG:
[1685330495.735352][31915:31920] CHIP:DMG:              },
[1685330495.735361][31915:31920] CHIP:DMG:
[1685330495.735371][31915:31920] CHIP:DMG:      ],
[1685330495.735385][31915:31920] CHIP:DMG:
[1685330495.735392][31915:31920] CHIP:DMG:      SuppressResponse = true,
[1685330495.735398][31915:31920] CHIP:DMG:      InteractionModelRevision = 1
[1685330495.735404][31915:31920] CHIP:DMG: }
[1685330495.735465][31915:31920] CHIP:TOO: Endpoint: 2 Cluster: 0x0000_0400 Attribute 0x0000_0000 DataVersion: 3712652986
[1685330495.735483][31915:31920] CHIP:TOO: Response Failure: Can not decode Data
[1685330495.735566][31915:31920] CHIP:EM: <<< [E:29968i M:71041588 (Ack:128230097)] (S) Msg TX to 1:0000000000007283 [BEA2] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1685330495.735580][31915:31920] CHIP:IN: (S) Sending msg 71041588 on secure session with LSID: 11259
[1685330495.735673][31915:31920] CHIP:EM: Flushed pending ack for MessageCounter:128230097 on exchange 29968i
[1685330495.735684][31915:31915] CHIP:TOO: Run command failure: ../../src/lib/core/CHIPTLVReader.cpp:239: CHIP Error 0x00000026: Wrong TLV type
```

shun-dgut, 2023-05-31:
[image]

shun-dgut, 2023-05-31:
[image]

wqx6, 2023-05-31:
Could you tell me the commit of esp-matter for this issue? I didn't see the error for current commit on main branch.

shun-dgut, 2023-05-31:
I added a save to flash flag to the code below, and the chiptool read the correct data, but there was still no response from apple home

```
attribute_t *create_illuminance_measured_value(cluster_t *cluster, nullable<uint16_t> value, nullable<uint16_t> min, nullable<uint16_t> max)
{
    attribute_t *attribute = esp_matter::attribute::create(cluster, IlluminanceMeasurement::Attributes::MeasuredValue::Id, ATTRIBUTE_FLAG_NULLABLE|ATTRIBUTE_FLAG_NONVOLATILE, esp_matter_nullable_uint16(value));
    if (!attribute) {
	ESP_LOGE(TAG, "Could not create attribute");
	return NULL;
    }
    esp_matter::attribute::add_bounds(attribute, esp_matter_nullable_uint16(min), esp_matter_nullable_uint16(max));
    return attribute;
}
```

shun-dgut, 2023-06-01:
> Could you tell me the commit of esp-matter for this issue? I didn't see the error for current commit on main branch.

commit id :   [8ae4eec257b01c11349a68429705](https://github.com/espressif/esp-matter/commit/273efe2b7c4b8ae4eec257b01c11349a68429705)

dhrishi, 2023-06-23:
@shun-dgut Do you still see this issue with the latest main branch? As @wqx6 mentioned, we didn't encounter the issue. Please close this if it works

dhrishi, 2023-07-11:
Closing this now. Feel free to re-open if you face the issue

closed 2023-07-11 by dhrishi