espressif/esp-matter #421: Duplicate attribute override registration failed

Wvirgil123, 2023-05-29:
环境: 
esp-matter 版本: 2f0929e5f8fd58b1a4cbaa2bd8b171ce50ad2a75
设备: esp32s3
demo: light

问题：运行过程提示 Duplicate attribute override registration failed

```
I (1818) chip[ZCL]: Using ZAP configuration...
I (1818) esp_matter_cluster: Cluster plugin init common callback
I (1828) chip[DMG]: AccessControlCluster: initializing
I (1828) chip[ZCL]: 0x3c0f9a0cep 0 clus 0x0000_0030 attr 0x0000_0000 not supported
I (1838) chip[ZCL]: Initiating Admin Commissioning cluster.
E (1848) chip[ZCL]: Duplicate attribute override registration failed
I (1858) chip[DIS]: Updating services using commissioning mode 1
I (1858) chip[DIS]: CHIP minimal mDNS started advertising.
I (1868) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (1878) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: DC7867F430170236.
I (1888) chip[DIS]: mDNS service published: _matterc._udp
I (1888) chip[IN]: CASE Server enabling CASE session setups
I (1898) chip[SVR]: Joining Multicast groups
I (1908) chip[SVR]: Server Listening...
```

rfortunatov, 2023-05-29:
Same thing here. Doesn't actually create any issues for me except errors in logs.

dhrishi, 2023-05-29:
@Wvirgil123 Does the error log cause any issue in the functionality?

shubhamdp, 2023-06-01:
@Wvirgil123 please see here https://github.com/project-chip/connectedhomeip/blob/68621d05753e277b38fd066b3d830cfbc758e4cf/src/app/util/attribute-storage.cpp#L1426

This is just a print and not an actual issue, If someone tries to registers the same attribute accessor interface then this error shows up.

dhrishi, 2023-06-01:
Closing this as it does not affect the functionality

closed 2023-06-01 by dhrishi