espressif/esp-matter #422: ESP Reboots after flashing custom device model built using ZAP

va1labh, 2023-05-30:
I have used the all-clusters-app example in the connectedhomeip/examples folder to create a 2channel switch using all-clusters-app.zap file. I was able to build it successfully, but as soon as I flash it on a ESP32 dev-kit, it reboots on its own. Below is the error log : 

E (2392) chip[ZCL]: Failed to read DoorLock attribute: attribute=0x1c, status=0xc3
E (2402) chip[ZCL]: Unable to get number of credentials supported per user when initializing lock endpoint, defaulting to 5 [endpointId=1]
E (2422) chip[ZCL]: Failed to read DoorLock attribute: attribute=0x11, status=0xc3
E (2422) chip[ZCL]: Unable to get number of supported users when initializing lock endpoint, defaulting to 10 [endpointId=1]
E (2432) chip[ZCL]: Failed to read DoorLock attribute: attribute=0x14, status=0xc3
E (2442) chip[ZCL]: Unable to get number of supported weekday schedules when initializing lock endpoint, defaulting to 10 [endpointId=1]
E (2462) chip[ZCL]: Failed to read DoorLock attribute: attribute=0x15, status=0xc3
E (2462) chip[ZCL]: Unable to get number of supported yearday schedules when initializing lock endpoint, defaulting to 10 [endpointId=1]
E (2472) chip[ZCL]: Failed to read DoorLock attribute: attribute=0x16, status=0xc3
E (2482) chip[ZCL]: Unable to get number of supported holiday schedules when initializing lock endpoint, defaulting to 10 [endpointId=1]

abort() was called at PC 0x400ec35d on core 0
0x400ec35d: BoltLockManager::Init(chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>, ESP32DoorLock::LockInitParams::LockParam) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chip::Optional<chip::app::Clusters::DoorLock::DlLockState>::Value() & at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h:164
 (inlined by) BoltLockManager::Init(chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>, ESP32DoorLock::LockInitParams::LockParam) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/platform/esp32/lock/BoltLockManager.cpp:96
 (inlined by) BoltLockManager::Init(chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>, ESP32DoorLock::LockInitParams::LockParam) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/platform/esp32/lock/BoltLockManager.cpp:34

Backtrace:0x40081dc6:0x3ffe67200x40090f95:0x3ffe6740 0x40097706:0x3ffe6760 0x400ec35d:0x3ffe67d0 0x400eca70:0x3ffe6800 0x400e7dcd:0x3ffe6830 0x400d6950:0x3ffe6850 0x4012b33d:0x3ffe6870 0x4012b62a:0x3ffe6890 0x4012b64d:0x3ffe6900 0x40094231:0x3ffe6920 
0x40081dc6: panic_abort at /home/vallabh/MatterPrj/esp-idf/components/esp_system/panic.c:402

0x40090f95: esp_system_abort at /home/vallabh/MatterPrj/esp-idf/components/esp_system/esp_system.c:128

0x40097706: abort at /home/vallabh/MatterPrj/esp-idf/components/newlib/abort.c:46

0x400ec35d: BoltLockManager::Init(chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>, ESP32DoorLock::LockInitParams::LockParam) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chip::Optional<chip::app::Clusters::DoorLock::DlLockState>::Value() & at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/Optional.h:164
 (inlined by) BoltLockManager::Init(chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>, ESP32DoorLock::LockInitParams::LockParam) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/platform/esp32/lock/BoltLockManager.cpp:96
 (inlined by) BoltLockManager::Init(chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>, ESP32DoorLock::LockInitParams::LockParam) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/platform/esp32/lock/BoltLockManager.cpp:34

0x400eca70: BoltLockManager::InitLockState() at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/platform/esp32/lock/BoltLockManager.cpp:823

0x400e7dcd: AppTask::LockInit() at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/main/AppTask.cpp:175

0x400d6950: InitServer(int) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/main/main.cpp:117

0x4012b33d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:290

0x4012b62a: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:505
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257

0x4012b64d: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:405
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/vallabh/MatterPrj/esp-matter/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280

0x40094231: vPortTaskWrapper at /home/vallabh/MatterPrj/esp-idf/components/freertos/port/xtensa/port.c:131

dhrishi, 2023-05-31:
@va1labh Please provide your changes on top of the existing example to help us recreate this issue. Also, please provide details about the connectedhomeip and ESP-IDF versions and/or GIT head commit used

dhrishi, 2023-06-06:
Please re-open once you add the relevant details

closed 2023-06-06 by dhrishi