espressif/esp-matter #425: [Fan Control] will esp-matter support attributes ID:0x0004 to ID:0x000a?

mmt-10, 2023-05-31:
As my understanding, esp-matter supports only below 4 attributes.
|ID|Name|Type|Constraint|Quality|Default|Access|Conformance|
|---|---|---|---|---|---|---|---|
|0x0000|FanMode|enum8|0 to 6|N|0|RW VO|M
|0x0001|FanModeSequence|enum8|0 to 5|N|MS|R[W] VO|M|
|0x0002|PercentSetting|uint8|0 to 100|X|0|RW VO|M|
|0x0003|PercentCurrent|uint8|0 to 100|desc| |R V|M|

Will esp-matter support attributes ID:0x0004 to ID:0x000a as below?

|ID|Name|Type|Constraint|Quality|Default|Access|Conformance|
|---|---|---|---|---|---|---|---|
|0x0004|SpeedMax|uint8|1 to 100|F|MS|R V|SPD|
|0x0005|SpeedSetting|uint8|0 to SpeedMax|X|0|RW VO|SPD|
|0x0006|SpeedCurrent|uint8|0 to SpeedMax|P|desc|R V|SPD|
|0x0007|RockSupport|map8|desc|F|0|R V|RCK|
|0x0008|RockSetting|map8|desc|P|0|RW VO|RCK|
|0x0009|WindSupport|map8|desc|F|0|R V|WND|
|0x000a|WindSetting|map8|desc|P|0|RW VO|WND|

mmt-10, 2023-05-31:
`FanModeSequence` and `PercentSetting` does not follow correct constraint range.
it looks like below issue.
[https://github.com/espressif/esp-matter/issues/354#issue-1687901350](https://github.com/espressif/esp-matter/issues/354#issue-1687901350)

```bash
./chip-tool fancontrol write fan-mode-sequence 10 ${NODE_ID} ${ENDPOINT}
./chip-tool fancontrol read fan-mode-sequence  ${NODE_ID} ${ENDPOINT}
```
 it returned `6`.

```bash
./chip-tool fancontrol write percent-setting 200 ${NODE_ID} ${ENDPOINT}
./chip-tool fancontrol read percent-setting  ${NODE_ID} ${ENDPOINT}
```
it returned `200`.

I cannot checked all clusters and all attributes.
but, constraint that attributes on other clusters may not follow correct range.

dhrishi, 2023-05-31:
@mmt-10 The Fan Control cluter was provisional for the previous Matter releases. It has some recent changes (xml as well as the implementation) and is planned for the Fall release, this year. We will soon add the changes to this repo. Are you executing TE1 and need it for this?

mmt-10, 2023-05-31:
@dhrishi 
Thank you for your reply.
I understand fan control on Matter is planning now.
And, I would like to ask your plan for fan control cluster. 

I can wait it for Fall release. 

I use fan control on my proof of concept product.
I will modify esp_matter_attribute.h, cpp and etc for fan control at now.

dhrishi, 2023-05-31:
@mmt-10 Updated fan control cluster should be added soon

closed 2023-05-31 by mmt-10