espressif/esp-matter #426: CHIP Error 0x000000AC: Internal error

Focalcrest-Madi, 2023-05-31:
“CHIP Error 0x000000AC: Internal error” occurs when using chip-tool to pair devices
Chip-tool log
[chip-tool.log](https://github.com/espressif/esp-matter/files/11608080/chip-tool.log)
Matter device log
[device.log](https://github.com/espressif/esp-matter/files/11608083/device.log)

dhrishi, 2023-05-31:
@Focalcrest-Madi Please provide details about the versions (IDF and esp-matter) and any changes on top of the existing applications. Without that, it is difficult to debug. The log here shows that the attestation verification has failed?

Are you using a different PAA than the test one? Or a different DAC is flashed in the factory partition?

Focalcrest-Madi, 2023-05-31:
@dhrishi 
ESP-IDF version is v4.4.3; ESP-Matter version is master.
I didn't flashed DAC certificate, ESP matter device can be added to Apple Home APP.
I have enabled the OTA function on the ESP matter device, and modified the Vendor ID and product ID.
chip-tool added custom cluster function.

dhrishi, 2023-05-31:
@Focalcrest-Madi 

>I didn't flashed DAC certificate

>I have enabled the OTA function on the ESP matter device, and modified the Vendor ID and product ID.

So this is expected to fail.

How did you do that? Once that is changed, you need to also change the attestation certificates and even the certification declaration (CD) to reflect the VID PID.

Please use the [mfg_tool](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool) to flash the factory partition with the correct artifacts https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html#matter-factory-partition-binary . Feel free to raise a new issue with the appropriate details if you face any issue with that.

closed 2023-05-31 by dhrishi