espressif/esp-matter #428: Unable to commission ESP32C6 with Light example

TimmyRB, 2023-05-31:
ESP-IDF: `v5.1-dev-4528-g420ebd208a`
ESP-Matter: [`2f0929e`](https://github.com/espressif/esp-matter/commit/2f0929e5f8fd58b1a4cbaa2bd8b171ce50ad2a75)
Board: `ESP32-C6-DevKitC-1`

Hey, I'm pretty new to this so it's entirely possible I'm missing some steps but I am having trouble commissioning my ESP32C6 that's flashed with the light example. 

Steps:
1. Run `idf.py erase-flash`
2. Build, Flash and Monitor the light example in VSCode
3. Run `matter device factoryreset` in device's console
4. Run `matter onboardingcodes ble qrcodeurl`
5. Scan resulting QR code in Apple Home app
6. Home app warns about uncertified device, press Add anyways
7. Says connecting to device for awhile until eventually it says it could not pair with the accessory

I've attached logs from the device's boot and from after commissioning has started
[boot.log](https://github.com/espressif/esp-matter/files/11613561/boot.log)
[commission.log](https://github.com/espressif/esp-matter/files/11613594/commission.log)

The commissioning log repeats itself seemingly indefinitely even after the Home app says it was Unable to add the Accessory. The Home I'm trying to add it to has a HomePod mini on the same wireless network I'm trying to add it to

[image]

Any help in figuring this out would be greatly appreciated!

shubhamdp, 2023-06-01:
@TimmyRB Looks like device is trying to connect to the WiFi network but it is not finding the wifi network. Are you using a 5GHz WiFi router? If yes, then ESP32 only support 2.4GHz bands.

Also, I see a crash in the logs, can you exactly tell us the steps that you carried out so that we can try to reproduce the crash and debug further.

Also, rather than retrieving qrcode from the device can you please use qrcode from here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning

TimmyRB, 2023-06-01:
I disabled the 5GHz band on my router and ran exactly the following
`idf.py erase-flash build flash monitor`
then I scanned the QR code from the link you mentioned and got the follow output which looks mostly identical to what I have uploaded above
[latest.log](https://github.com/espressif/esp-matter/files/11625259/latest.log)

Then after about a minute after the commissioning had already failed, the device started endlessly and quickly throwing this into the console
[repeat.log](https://github.com/espressif/esp-matter/files/11625265/repeat.log)

TimmyRB, 2023-06-01:
After disabling Wifi 6 on my router and running `idf.py erase-flash build flash monitor` then `matter device factoryreset` then attempting commission again through the Home app, I get the following errors from trying to join the Wifi Network
[wifi.log](https://github.com/espressif/esp-matter/files/11626151/wifi.log)

shubhamdp, 2023-06-02:
- From latest.log, it looks like device is having a hard time connecting to the network ([error details](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#c.ESP_ERR_WIFI_NOT_CONNECT))
> E (6571) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F

- repeat.log
Device is crashing in loop.

- wifi.log
> W (62911) wifi:Haven't to connect to a suitable AP now!

This logs basically says there is no AP to connect to.

Can you try using [commit ea5e0ff](https://github.com/espressif/esp-idf/tree/ea5e0ff) rather than some untested esp-idf commit(v5.1-dev-4528-g420ebd208a) with esp-matter.

shubhamdp, 2023-06-13:
@TimmyRB any updates on this issue?

dhrishi, 2023-06-23:
I am closing this issue now. Please re-open in case of any updates to share

closed 2023-06-23 by dhrishi

mukeshbauskar, 2024-06-27:
@dhrishi I am seeing this issue as well. Please let me know what information you want.

PeterKjaerFisker, 2024-08-29:
@dhrishi I have the same/similar issue.
I have tried the wifi_matter_light example from the launchpad with my ESP-C6 and Google Home, and it fails during commision. 
 
I have also tried building and flashing the light example from esp-matter locally using 
ESP-IDF: a322e6b
ESP-Matter: 183fe9b
connectedhomeip: 0460715

and I get same issues with connecting to my AP

mukeshbauskar, 2024-08-29:
@PeterKjaerFisker Make sure following 

1. The esp32 and your device which you are suing to add are on same network
2. If you are using google home then use it on android device.

PeterKjaerFisker, 2024-08-29:
@mukeshbauskar My problem seems to be that I can't get the ESP32 to connect to the network at all. 😅

I am using an android device, my Pixel 7 if it helps any.

mukeshbauskar, 2024-08-29:
@PeterKjaerFisker Ohhh. I would recommend create a separate issue. If you decide to do so, can you please tag me as well...