espressif/esp-matter #429: Build failed (mbedtls_hkdf)

YOGYUI, 2023-05-31:
I pulled esp-matter repository to the latest commit (9be53266b1e934797b9a1cc08961551bc4661677) and found a problem that building project can't be done because of 'mbedtls_hkdf' reference issue.
```
/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:330: undefined reference to `mbedtls_hkdf'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I solved this problem by adding below config to sdkconfig.
```
CONFIG_MBEDTLS_HKDF_C=y
```
Should I always enable HKDF like above?
(I already found HKDF related commits in d0b81170f3a68eace4d689e9db4aac3a53ef151a)

shubhamdp, 2023-06-01:
@YOGYUI yes, you will have to always enable `CONFIG_MBEDTLS_HKDF_C=y`. You can also add this to your project's `sdkconfig.defaults`.

closed 2023-06-01 by dhrishi

YOGYUI, 2023-06-01:
@shubhamdp I got it, Thanks for your reply.