espressif/esp-matter #431: Apple Home creates an additional fabric

JonathanWitthoeft, 2023-05-31:
Hello I am commissioning with Apple Home and noticed 2 fabrics are created. Then when the accessory is removed from Apple Home, only the first fabric that was created is removed. The second, stale fabric (index 0x2) remains. Logs attached.

esp-idf 5.0.1:
https://github.com/espressif/esp-idf/tree/a4afa44435ef4488d018399e1de50ad2ee964be8
[stale_fabric_dbg.txt](https://github.com/espressif/esp-matter/files/11618462/stale_fabric_dbg.txt)

esp-matter:
https://github.com/espressif/esp-matter/tree/273efe2b7c4b8ae4eec257b01c11349a68429705

jadhavrohit924, 2023-06-01:
Hi @JonathanWitthoeft,
You are right. We are observing same behaviour.

JonathanWitthoeft, 2023-06-01:
@jadhavrohit924 Looking at my logs it appears the first fabric is created before the device is commissioned to my Wi-Fi network. It is created over the BLE link. Then the second fabric is created over the network after BLE is disabled. Is there a wayt to tell what interface fabric is created on? If I can search for fabrics on bluetooth, I can delete them.

jonsmirl, 2023-06-04:
This is due to Apple/Google not wanting to allow third party apps to see the wifi credentials. So Apple/Google are first using BLE to commission a fabric which only use is to set in the wifi credentials. Then after they create that first fabric their libraries use IP based OpenComissioningWindow to add more fabrics for Home or vendors apps to use for everything else. That second fabric and subsequent fabrics can't see the wifi credentials from the first fabric.  So if you use Home and a vendor app there will be three fabrics.

So this is expected behavior from Apple/Google.

dhrishi, 2023-06-06:
@JonathanWitthoeft Adding an extra fabric is something that is the implementation by the ecosystems. But, IMO, when you remove the device, the fabrics should be removed. We will confirm this.

jonsmirl, 2023-06-06:
This procedure has some issues. Maybe you can get Apple/Google to pay attention to you, they are just ignoring me.

Commission a device and have their first commission work and then your second commission fail for some reason.  Hmm.... that leaves the device on the wifi network with BLE turned off.   Now do what every consumer will do... try to commission your device again.   Now you are stuck. The QRCode telling it to commission via BLE, the Google/Apple has already commissioned it onto wifi. They can't find the device.

Only way out is a factory reset.

dhrishi, 2023-06-06:
@jonsmirl 

>The QRCode telling it to commission via BLE, the Google/Apple has already commissioned it onto wifi.

Ideally the commissioner should also try to find the device over the IP network even if the QR code says BLE. From v1.0 spec -

```
Commissioners SHALL always support discovering a device using DNS-based Service Discovery
(DNS-SD) for commissioning, irrespective of the Discovery Capabilities Bitmask specified in the
Section 5.1.1, “Onboarding Payload Contents”.
```

jonsmirl, 2023-06-06:
That does not apply.  The device is not in commissioning mode, the first commissioning succeeded. 

So when you recommission from the vendor app, Google/Apple need to recognize that the first commissioning succeeded and then instruct the device using the Matter protocol to OpenCommissioningWindow.  That's the part they are missing.

shubhamdp, 2023-06-13:
@JonathanWitthoeft Please close the issue, this is an ecosystem issue rather than the SDK one.

closed 2023-06-13 by JonathanWitthoeft

JonathanWitthoeft, 2025-05-29:
> An old post but for anyone stuck on this issue recently as I was, the matter device must also be removed from your phone via `settings`->`general`->`matter accessories`. Users will definitely get stuck on this.

This issue was created before apple started creating an apple key-chain pairing followed by an apple Home pairing. 
The new technique allows apple key-chain to add pairings for 3rd party apps like Google Home.

What I believe you are doing is removing the device from apple key-chain, which removes it's original pairing.