espressif/esp-matter #432: Matter SDK Install error

gabrielfonte, 2023-05-31:
I have tried to install the matter SDK of ESP32 on my PC, I followed the tutorial available at https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html# in Linux session. Also tried cloning with the --recursive option and found the same problem.

The error message is the following when I try to run _install.sh_:

```
Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (43.0s)
  Setting up Python environment.....done (1m58.4s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20230531 17:04:35 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Building host tools

Done. Made 6568 targets from 324 files in 269ms
ninja: Entering directory `/home/gabriel/esp/esp-matter/connectedhomeip/connectedhomeip/out/host'
[674/2074] c++ obj/src/app/tests/suites/pics/libPICS.PICSBooleanExpressionParser.cpp.o
FAILED: obj/src/app/tests/suites/pics/libPICS.PICSBooleanExpressionParser.cpp.o 
g++ -MMD -MF obj/src/app/tests/suites/pics/libPICS.PICSBooleanExpressionParser.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -c ../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp -o obj/src/app/tests/suites/pics/libPICS.PICSBooleanExpressionParser.cpp.o
In file included from ../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp:19:
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:46:107: error: ‘uint8_t’ has not been declared
   46 |     static bool EvaluateExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS, uint8_t & index);
      |                                                                                                           ^~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:47:110: error: ‘uint8_t’ has not been declared
   47 |     static bool EvaluateSubExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS, uint8_t & index);
      |                                                                                                              ^~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp: In static member function ‘static bool PICSBooleanExpressionParser::Eval(std::string, std::map<std::__cxx11::basic_string<char>, bool>&)’:
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp:30:45: error: cannot bind non-const lvalue reference of type ‘int&’ to a value of type ‘uint8_t’ {aka ‘unsigned char’}
   30 |     return EvaluateExpression(tokens, PICS, index);
      |                                             ^~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:46:117: note:   initializing argument 3 of ‘static bool PICSBooleanExpressionParser::EvaluateExpression(std::vector<std::__cxx11::basic_string<char> >&, std::map<std::__cxx11::basic_string<char>, bool>&, int&)’
   46 |     static bool EvaluateExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS, uint8_t & index);
      |                                                                                                           ~~~~~~~~~~^~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp: At global scope:
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp:97:6: error: no declaration matches ‘bool PICSBooleanExpressionParser::EvaluateExpression(std::vector<std::__cxx11::basic_string<char> >&, std::map<std::__cxx11::basic_string<char>, bool>&, uint8_t&)’
   97 | bool PICSBooleanExpressionParser::EvaluateExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:46:17: note: candidate is: ‘static bool PICSBooleanExpressionParser::EvaluateExpression(std::vector<std::__cxx11::basic_string<char> >&, std::map<std::__cxx11::basic_string<char>, bool>&, int&)’
   46 |     static bool EvaluateExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS, uint8_t & index);
      |                 ^~~~~~~~~~~~~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:28:7: note: ‘class PICSBooleanExpressionParser’ defined here
   28 | class PICSBooleanExpressionParser
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.cpp:129:6: error: no declaration matches ‘bool PICSBooleanExpressionParser::EvaluateSubExpression(std::vector<std::__cxx11::basic_string<char> >&, std::map<std::__cxx11::basic_string<char>, bool>&, uint8_t&)’
  129 | bool PICSBooleanExpressionParser::EvaluateSubExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:47:17: note: candidate is: ‘static bool PICSBooleanExpressionParser::EvaluateSubExpression(std::vector<std::__cxx11::basic_string<char> >&, std::map<std::__cxx11::basic_string<char>, bool>&, int&)’
   47 |     static bool EvaluateSubExpression(std::vector<std::string> & tokens, std::map<std::string, bool> & PICS, uint8_t & index);
      |                 ^~~~~~~~~~~~~~~~~~~~~
../../src/app/tests/suites/pics/PICSBooleanExpressionParser.h:28:7: note: ‘class PICSBooleanExpressionParser’ defined here
   28 | class PICSBooleanExpressionParser
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
[681/2074] c++ obj/src/app/tests/suites/commands/interaction_model/libInteractionModel.InteractionModel.cpp.o
ninja: build stopped: subcommand failed.

```

Any clues on how to solve this problem?

**Branch**: main [9be53266b1e934797b9a1cc08961551bc4661677]
**OS:** Fedora 38

shubhamdp, 2023-06-01:
@gabrielfonte I hope you have installed all the required prerequisites for your platform.

From the logs, your environment is setup and building host tools step failed.
For retrying the last part, can you try the below command please.
```
cd connectedhomeip/connectedhomeip
gn gen out/host
ninja -C out/host
```

As we have not tested building Matter on Fedora, there could be a problem.

riwalker, 2023-06-01:
Related Question.  About to start development for the ESP32-C6 :
1) when will there be a combined Matter development stack ?  should we wait for that ? (vs the "ESP32-C6(preview) and Zigbee Bridge example, ESP-IDF branch at [commit ea5e0ff](https://github.com/espressif/esp-idf/tree/ea5e0ff) in branch release/v5.1 should be used")
2) What platform / OS / IDE is best to develop Matter on the C6 - to match the Espressif development team, so we do not see any incompatibilities here in builds ? or add any complications (i.e if Python scripts are required, more difficult in windows than Linux)

shubhamdp, 2023-06-02:
@riwalker 
1. We are planning to move to IDF release v5.1 as soon as it will be released (expecting by end of this month, not guaranteeing). But, we still need to make sure that SDK works with prior IDF version.
2. For OS, Ubuntu > 20.4 or MacOS > 10.15 is preferred, but we still support Window using WSL ([ref](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup)). For IDF, probably that your choice, but most of us here use (terminal + vim) or vscode.

closed 2023-06-03 by dhrishi

hermlon, 2023-06-17:
This issue appears when compiling with gcc 13, as explained [here](https://gcc.gnu.org/gcc-13/porting_to.html) you now have to include `<cstdint>` for `uint8_t` explicitly. 

I had to include it in `connectedhomeip/src/app/tests/suites/pics/PICSBooleanExpressionParser.h` and `connectedhomeip/ src/setup_payload/AdditionalDataPayload.h` to make it compile. It looks like this was already fixed in the connectedhomeip upstream: https://github.com/project-chip/connectedhomeip/pull/26731 last month. Is this already updated in the espressif/connectedhomeip fork?

kunjumonkk, 2023-08-06:
It got fixed for me after installing following pkg and able to install matter successfully:

sudo apt-get install git gcc g++ pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev