espressif/esp-matter #433: Matter Thread network

Gonta01, 2023-06-01:
How to configure device to be one of these device types in Thread network topology:
Router-Eligible End Device (REED);
Full End Device (FED);
Minimal End Device (MED);
Sleepy End Device (SED).

I saw that there are options in menuconfig in OpenThread configuration.
Config the Thread device type:
Full Thread Device;
Minimal Thread Device;
Radio Only Device.

By default Full Thread Device and it works. When Minimal Thread Device is chosen it fails to build the project. Are some additional configurations needed? What is Radio Only Device?

wqx6, 2023-06-02:
> How to configure device to be one of these device types in Thread network topology

Here are the roles and types for thread network https://openthread.io/guides/thread-primer/node-roles-and-types#device_types

> Sleepy End Device (SED)

To make a device SED you can use `otThreadSetLinkMode`.

> When Minimal Thread Device is chosen it fails to build the project.

Can you tell me the idf commit you are using?

> What is Radio Only Device?

It is a RCP device. https://openthread.io/platforms/co-processor#radio_co-processor_rcp

Gonta01, 2023-06-02:
@wqx6 thank you for reply!
> To make a device SED you can use otThreadSetLinkMode.

Are there any examples for ESP32C6 or ESP32H2? 

> Can you tell me the idf commit you are using?

commit 6ad6fb9755ac6433be4d22e30f1811f6605de085 (HEAD -> release/v5.1, origin/release/v5.1)

wqx6, 2023-06-02:
> Are there any examples for ESP32C6 or ESP32H2?

You can use [ot-cli](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli) example in esp-idf with the command 'mode -' to make a thread device SED. This command will call the `otThreadSetLinkMode`.

> commit 6ad6fb9755ac6433be4d22e30f1811f6605de085 (HEAD -> release/v5.1, origin/release/v5.1)

Please use the commit [ea5e0ff2](https://github.com/espressif/esp-idf/tree/ea5e0ff298e6257b31d8e0c81435e6d3937f04c7) to build the mtd example.

dhrishi, 2023-06-02:
Closing as the questions are answered

closed 2023-06-02 by dhrishi