espressif/esp-matter #434: PacketBuffer: pool EMPTY.

jonsmirl, 2023-06-03:
Any idea what is going on here? I am getting these a lot during testing. This node was just sitting idle when I tried to use it.  How can it be out of packet buffers while sitting idle?

~~~
I (44483984) ROUTE_HOOK: Received RIO
I (44483984) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (44483984) chip[DL]: IP_EVENT_GOT_IP6
I (44483984) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd02:832c:729a:16f1:86f7:03ff:fe66:ffd0
I (44483994) ROUTE_HOOK: Hook already installed on netif, skip...
I (44484004) ep: Interface IP Address Changed
I (44484004) chip[DIS]: Updating services using commissioning mode 0
I (44484014) chip[DIS]: CHIP minimal mDNS started advertising.
I (44484054) chip[DIS]: Advertise operational node 4284C2F5A80B5F7C-3856320C25731B0E
I (44484054) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 4284C2F5A80B5F7C-3856320C25731B0E.
I (44484064) chip[DIS]: mDNS service published: _matter._tcp
I (44484074) chip[DIS]: Advertise operational node C1EDEF2ECC2CF621-AB0DE00000000002
I (44484074) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: C1EDEF2ECC2CF621-AB0DE00000000002.
I (44484104) chip[DIS]: mDNS service published: _matter._tcp
I (44493184) ROUTE_HOOK: Received RIO
I (44493184) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (44493584) NimBLE: GAP procedure initiated: advertise; 
I (44493584) NimBLE: disc_mode=0
I (44493584) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (44493594) NimBLE: 

I (44493624) NimBLE: GAP procedure initiated: stop advertising.

I (44500894) app_main: Free internal 39820 total 39820
I (44503014) ROUTE_HOOK: Received RIO
I (44503014) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (44503594) NimBLE: GAP procedure initiated: advertise; 
I (44503594) NimBLE: disc_mode=0
I (44503594) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (44503604) NimBLE: 

I (44503634) NimBLE: GAP procedure initiated: stop advertising.

I (44510284) chip[EM]: >>> [E:58885r S:0 M:218073044] (U) Msg RX from 0:70FB937B7AE37B4D [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (44510284) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9cf30
I (44510294) chip[IN]: CASE Server disabling CASE session setups
I (44510304) chip[EM]: <<< [E:58885r S:0 M:265404581 (Ack:218073044)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (44510314) chip[IN]: (U) Sending msg 265404581 to IP address 'UDP:192.168.1.190%st1:46180'
I (44510324) chip[SC]: Received Sigma1 msg
I (44510334) chip[SC]: CASE matched destination ID: fabricIndex 2, NodeID 0xAB0DE00000000002
I (44510604) chip[EM]: <<< [E:58885r S:0 M:265404582 (Ack:218073044)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (44510604) chip[IN]: (U) Sending msg 265404582 to IP address 'UDP:192.168.1.190%st1:46180'
E (44510614) chip[CSL]: PacketBuffer: pool EMPTY.
I (44510624) chip[EM]: <<< [E:58885r S:0 M:265404583 (Ack:218073044)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (44510634) chip[IN]: (U) Sending msg 265404583 to IP address 'UDP:192.168.1.190%st1:46180'
E (44510644) chip[CSL]: PacketBuffer: pool EMPTY.
E (44510644) chip[SC]: Failed to send status report message: b
E (44510654) chip[IN]: CASE Session establishment failed: b
I (44510664) chip[IN]: CASE Server enabling CASE session setups
E (44510664) chip[EM]: OnMessageReceived failed, err = b
I (44513614) NimBLE: GAP procedure initiated: advertise; 
I (44513614) NimBLE: disc_mode=0
I (44513614) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (44513614) NimBLE: 
~~~

jonsmirl, 2023-06-04:
In this trace I hit the error and then it clears itself a few seconds latter.
~~~
I (491632) NimBLE: GAP procedure initiated: advertise; 
I (491632) NimBLE: disc_mode=0
I (491632) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (491642) NimBLE: 

I (491672) NimBLE: GAP procedure initiated: stop advertising.

I (492442) chip[EM]: >>> [E:23313r S:0 M:8168059] (U) Msg RX from 0:D3128C7EEC4120F8 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (492442) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9cf30
I (492452) chip[IN]: CASE Server disabling CASE session setups
I (492462) chip[EM]: <<< [E:23313r S:0 M:30504474 (Ack:8168059)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (492472) chip[IN]: (U) Sending msg 30504474 to IP address 'UDP:192.168.1.190%st1:44857'
I (492482) chip[SC]: Received Sigma1 msg
I (492492) chip[SC]: CASE matched destination ID: fabricIndex 2, NodeID 0xAB0DE00000000005
I (492752) chip[EM]: <<< [E:23313r S:0 M:30504475 (Ack:8168059)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (492762) chip[IN]: (U) Sending msg 30504475 to IP address 'UDP:192.168.1.190%st1:44857'
I (492772) chip[SC]: Sent Sigma2 msg
I (492782) chip[EM]: >>> [E:23313r S:0 M:8168060 (Ack:30504475)] (U) Msg RX from 0:D3128C7EEC4120F8 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (492942) chip[EM]: >>> [E:23313r S:0 M:8168061 (Ack:30504475)] (U) Msg RX from 0:D3128C7EEC4120F8 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (492952) chip[EM]: <<< [E:23313r S:0 M:30504476 (Ack:8168061)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (492962) chip[IN]: (U) Sending msg 30504476 to IP address 'UDP:192.168.1.190%st1:44857'
I (492972) chip[SC]: Received Sigma3 msg
E (492982) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (493612) chip[EM]: <<< [E:23313r S:0 M:30504477 (Ack:8168061)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (493612) chip[IN]: (U) Sending msg 30504477 to IP address 'UDP:192.168.1.190%st1:44857'
E (493622) chip[CSL]: PacketBuffer: pool EMPTY.
E (493632) chip[SC]: Failed to send status report message: b
I (493632) chip[SC]: SecureSession[0x3fc9b730]: Moving from state 'kEstablishing' --> 'kActive'
I (493642) chip[IN]: CASE Session established to peer: <AB0DE00000000000, 2>
I (493652) chip[IN]: CASE Server enabling CASE session setups
I (500772) app_main: Free internal 48324 biggest block 21504 total 48324
I (501642) NimBLE: GAP procedure initiated: advertise; 
I (501642) NimBLE: disc_mode=0
I (501642) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (501652) NimBLE: 

I (501682) NimBLE: GAP procedure initiated: stop advertising.

I (506382) ROUTE_HOOK: Received RIO
I (506382) ROUTE_HOOK: prefix FD89:EBF4:10D6:: lifetime 1800
I (506382) ROUTE_HOOK: Received RIO
I (506382) ROUTE_HOOK: prefix FD89:EBF4:10D6:: lifetime 1800
I (511662) NimBLE: GAP procedure initiated: advertise; 
I (511662) NimBLE: disc_mode=0
I (511662) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (511662) NimBLE: 

I (511702) NimBLE: GAP procedure initiated: stop advertising.

I (514062) ROUTE_HOOK: Received RIO
I (514062) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (520772) app_main: Free internal 48324 biggest block 21504 total 48324
I (521672) NimBLE: GAP procedure initiated: advertise; 
I (521672) NimBLE: disc_mode=0
I (521672) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (521682) NimBLE: 

I (521712) NimBLE: GAP procedure initiated: stop advertising.

I (531252) ROUTE_HOOK: Received RIO
I (531252) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (531252) chip[DL]: IP_EVENT_GOT_IP6
I (531252) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd02:832c:729a:16f1:86f7:03ff:fe68:1340
I (531262) ROUTE_HOOK: Hook already installed on netif, skip...
I (531272) chip[EM]: >>> [E:23318r S:0 M:8168070] (U) Msg RX from 0:86A8527FA03C0CEC [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (531282) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9cf80
I (531292) chip[IN]: CASE Server disabling CASE session setups
I (531302) chip[EM]: <<< [E:23318r S:0 M:30504478 (Ack:8168070)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[... 41 lines cut ...]
I (533172) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x0000FFFD is 1 **********
I (533182) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x00000028's Attribute 0x00000005 is Office/Rocker2 **********
E (533192) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = b
I (533212) chip[EM]: <<< [E:23319r S:57093 M:136743085 (Ack:178024671)] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0001:05 (IM:ReportData)
I (533212) chip[IN]: (S) Sending msg 136743085 on secure session with LSID: 57093
I (533402) chip[EM]: >>> [E:23319r S:57093 M:178024672 (Ack:136743085)] (S) Msg RX from 2:AB0DE00000000000 [F621] --- Type 0001:01 (IM:StatusResponse)
I (533412) chip[IM]: Received status response, status is 0x00
[... cut ...]
I (533662) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003C's Attribute 0x0000FFFC is 0 **********
I (533672) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003C's Attribute 0x0000FFFD is 1 **********
I (533682) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000003E's Attribute 0x0000FFFC is 0 **********
E (533702) chip[DMG]: Error retrieving data from clusterId: 0x0000_003E, err = b
I (533712) chip[EM]: <<< [E:23319r S:57093 M:136743087 (Ack:178024673)] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0001:05 (IM:ReportData)
~~~

jonsmirl, 2023-06-04:
Where are these coming from?
CHIP_SYSTEM_CONFIG_USE_LWIP
CHIP_SYSTEM_PACKETBUFFER_FROM_CHIP_POOL
CHIP_SYSTEM_PACKETBUFFER_FROM_CHIP_HEAP

Can I change the packet allocator on ESP32? Is there some kind of heap debug which can be dumped when this fails? My monitoring of memory says we are not out of memory, yet this allocator is failing.

jonsmirl, 2023-06-05:
This scenario triggers the failure about 50% of the time.  This first sequence is because I killed the Android app with an active subscription.
~~~
I (524222) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
I (525262) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
E (526702) chip[EM]: Failed to Send CHIP MessageCounter:51027795 on exchange 45679i sendCount: 4 max retries: 4
~~~
Then I restarted the app here...   So it looks like to me when I killed the app the first time things are not get properly cleaned up.
~~~
I (566142) chip[EM]: >>> [E:41330r S:0 M:24509645] (U) Msg RX from 0:A2A0DB7688BFB46A [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (566152) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9d760
I (566162) chip[IN]: CASE Server disabling CASE session setups
I (566162) chip[EM]: <<< [E:41330r S:0 M:135017071 (Ack:24509645)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (566182) chip[IN]: (U) Sending msg 135017071 to IP address 'UDP:192.168.1.190%st1:48125'
I (566192) chip[SC]: Received Sigma1 msg
I (566192) chip[SC]: CASE matched destination ID: fabricIndex 2, NodeID 0xAB0DE00000000005
E (566412) chip[CSL]: PacketBuffer: pool EMPTY.
I (566412) chip[EM]: <<< [E:41330r S:0 M:135017072 (Ack:24509645)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (566422) chip[IN]: (U) Sending msg 135017072 to IP address 'UDP:192.168.1.190%st1:48125'
E (566432) chip[CSL]: PacketBuffer: pool EMPTY.
~~~

Full sequence....
~~~
I (521792) NimBLE: GAP procedure initiated: stop advertising.

I (522842) chip[DMG]: Refresh subscribe timer sync after 9 seconds
I (522842) chip[EM]: <<< [E:45679i S:62415 M:51027795] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0001:05 (IM:ReportData)
I (522842) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
I (522852) chip[DMG]: Refresh Subscribe Sync Timer with min 1 seconds and max 10 seconds
I (522962) wifi:<ba-add>idx:1 (ifx:0, b8:f8:53:90:83:c2), tid:6, ssn:0, winSize:64
I (523262) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
I (523612) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
I (524222) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
I (525262) chip[IN]: (S) Sending msg 51027795 on secure session with LSID: 62415
E (526702) chip[EM]: Failed to Send CHIP MessageCounter:51027795 on exchange 45679i sendCount: 4 max retries: 4
I (528152) ROUTE_HOOK: Received RIO
I (528152) ROUTE_HOOK: prefix FD89:EBF4:10D6:: lifetime 1800
I (528162) ROUTE_HOOK: Received RIO
I (528162) ROUTE_HOOK: prefix FD89:EBF4:10D6:: lifetime 1800
I (529062) chip[SC]: SecureSession[0x3fc9bea8]: Moving from state 'kActive' --> 'kDefunct'
E (529062) chip[DMG]: Time out! failed to receive status response from Exchange: 45679i
I (531762) NimBLE: GAP procedure initiated: advertise; 
I (531762) NimBLE: disc_mode=0
I (531762) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (531772) NimBLE: 

I (531802) NimBLE: GAP procedure initiated: stop advertising.

I (533992) ROUTE_HOOK: Received RIO
I (533992) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (533992) ROUTE_HOOK: Received RIO
I (533992) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (536762) ROUTE_HOOK: Received RIO
I (536762) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (540802) app_main: Free internal 46172 biggest block 20480 total 46172
I (541782) NimBLE: GAP procedure initiated: advertise; 
I (541782) NimBLE: disc_mode=0
I (541782) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (541782) NimBLE: 

I (541822) NimBLE: GAP procedure initiated: stop advertising.

I (551792) NimBLE: GAP procedure initiated: advertise; 
I (551792) NimBLE: disc_mode=0
I (551792) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (551802) NimBLE: 

I (551832) NimBLE: GAP procedure initiated: stop advertising.

I (560802) app_main: Free internal 46172 biggest block 20480 total 46172
[... 6 lines cut ...]

I (566142) chip[EM]: >>> [E:41330r S:0 M:24509645] (U) Msg RX from 0:A2A0DB7688BFB46A [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (566152) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9d760
I (566162) chip[IN]: CASE Server disabling CASE session setups
I (566162) chip[EM]: <<< [E:41330r S:0 M:135017071 (Ack:24509645)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (566182) chip[IN]: (U) Sending msg 135017071 to IP address 'UDP:192.168.1.190%st1:48125'
I (566192) chip[SC]: Received Sigma1 msg
I (566192) chip[SC]: CASE matched destination ID: fabricIndex 2, NodeID 0xAB0DE00000000005
E (566412) chip[CSL]: PacketBuffer: pool EMPTY.
I (566412) chip[EM]: <<< [E:41330r S:0 M:135017072 (Ack:24509645)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (566422) chip[IN]: (U) Sending msg 135017072 to IP address 'UDP:192.168.1.190%st1:48125'
E (566432) chip[CSL]: PacketBuffer: pool EMPTY.
E (566432) chip[SC]: Failed to send status report message: b
E (566442) chip[IN]: CASE Session establishment failed: b
I (566442) chip[IN]: CASE Server enabling CASE session setups
E (566452) chip[EM]: OnMessageReceived failed, err = b
I (570272) ROUTE_HOOK: Received RIO
I (570272) ROUTE_HOOK: prefix 2600:4040:51BF:4000:: lifetime 7200
I (571822) NimBLE: GAP procedure initiated: advertise; 
I (571822) NimBLE: disc_mode=0
I (571822) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (571822) NimBLE: 

I (571862) NimBLE: GAP procedure initiated: stop advertising.
~~~

jonsmirl, 2023-06-05:
MDNS task is running at a default priority of 1. Maybe that should higher? What should I try setting it to?

jonsmirl, 2023-06-05:
I have been running for several hours now with MDNS priority set to 5 and have not encounter the previous problem.  This fix makes sense. Running MDNS at a higher priority is letting it filter the packets more quickly and quickly free the buffers from irrelevant messages.

jonsmirl, 2023-06-06:
I'm up to about 8 hours of running with priority set to 5. I've hit 'pool EMPTY" once with three nodes running. So P5 seems to mitigate my problem but it does not completely fix them. This failure is in a new place.

~~~
I (34803244) ROUTE_HOOK: prefix 2600:4040:51BF:4000:: lifetime 7200
I (34806454) chip[DMG]: Refresh subscribe timer sync after 9 seconds
I (34806454) chip[EM]: <<< [E:44466i S:26162 M:82824454] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0001:05 (IM:ReportData)
I (34806454) chip[IN]: (S) Sending msg 82824454 on secure session with LSID: 26162
I (34806464) chip[DMG]: Refresh Subscribe Sync Timer with min 1 seconds and max 10 seconds
I (34806474) chip[EM]: >>> [E:44466i S:26162 M:196567103 (Ack:82824454)] (S) Msg RX from 2:AB0DE00000000000 [F621] --- Type 0001:01 (IM:StatusResponse)
I (34806494) chip[IM]: Received status response, status is 0x00
I (34806494) chip[EM]: <<< [E:44466i S:26162 M:82824455 (Ack:196567103)] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (34806514) chip[IN]: (S) Sending msg 82824455 on secure session with LSID: 26162
I (34807144) ROUTE_HOOK: Received RIO
I (34807144) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (34807664) chip[DL]: IP_EVENT_GOT_IP6
I (34807664) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd02:832c:729a:16f1:86f7:03ff:fe66:ffd0
I (34807664) ROUTE_HOOK: Hook already installed on netif, skip...
I (34807674) ep: Interface IP Address Changed
I (34807674) chip[DIS]: Updating services using commissioning mode 0
I (34807684) chip[DIS]: CHIP minimal mDNS started advertising.
I (34807724) chip[DIS]: Advertise operational node 4284C2F5A80B5F7C-3856320C25731B0E
I (34807724) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 4284C2F5A80B5F7C-3856320C25731B0E.
I (34807734) chip[DIS]: mDNS service published: _matter._tcp
I (34807734) chip[DIS]: Advertise operational node C1EDEF2ECC2CF621-AB0DE00000000002
I (34807744) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: C1EDEF2ECC2CF621-AB0DE00000000002.
E (34807764) chip[CSL]: PacketBuffer: pool EMPTY.
I (34807774) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (34807774) chip[DIS]: Failed to advertise records: b
E (34807784) chip[CSL]: PacketBuffer: pool EMPTY.
I (34807784) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (34807794) chip[DIS]: Failed to advertise records: b
I (34807794) chip[DIS]: mDNS service published: _matter._tcp
I (34808554) NimBLE: GAP procedure initiated: advertise; 
I (34808554) NimBLE: disc_mode=0
I (34808554) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (34808564) NimBLE: 
~~~

jonsmirl, 2023-06-06:
Another variation on the mDNS error
~~~
I (18155764) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18155764) chip[DIS]: Failed to advertise records: 3000001
I (18155774) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18155774) chip[DIS]: Failed to advertise records: 3000001
I (18155784) chip[DIS]: mDNS service published: _matter._tcp
~~~

I am also very slowly leaking memory while sitting idle with a subscription going on.
~~~
I (160674) app_main: Free internal 46044 biggest block 20480 total 46044
I (43105014) app_main: Free internal 37696 biggest block 10240 total 37696
~~~
Over the course of 12hrs I have lost 10KB in leaks.

shubhamdp, 2023-06-06:
@jonsmirl We are looking into this

dhrishi, 2023-06-06:
@jonsmirl Can you please upload (and not paste) the entire log file since bootup. We have been doing some long duration testing and in that context added a couple of fixes (related to CASE errors) to upstream which are now in the v1.1-branch.

dhrishi, 2023-06-06:
The device was commissioned to 2 ecosystems and the chiptool

jonsmirl, 2023-06-06:
I have the BLE bridge running too. It is not bridging anything but BLE mesh is active.

jonsmirl, 2023-06-06:
I have new test running with capture to a file. It is set to dump the heap every 30 minutes. Is there something with more useful info than heap_caps_print_heap_info(MALLOC_CAP_INTERNAL);?

Can you try putting a taskYIELD() or whatever FreeRTOS wants right before CHIP tries to allocate a packet buffer? Doing that will let the higher priority mDNS task free up filtered packets.

jonsmirl, 2023-06-06:
Check out this log. I booted the device and then immediately (72926) went into 'pool EMPTY'.  There is plenty of free memory available.  This looks like something is not working right with mDNS.  First thing I would try is a taskYIELD() or similar right before allocating the packet buffer. Maybe I can hack one in.

[ACM0-save.txt](https://github.com/espressif/esp-matter/files/11667655/ACM0-save.txt)

jonsmirl, 2023-06-06:
I've added a taskYIELD() in the two places where 'pool EMPTY' can happen. I also have the MDNS task priority at 5. I've been running now for an hour on two nodes with zero MDNS errors.

I will have to leave a lot longer for leaks issues. Those take about 12hrs to become visible and then kill the device around 24hrs. These are C3 nodes so only internal RAM.  My suspicion is that the mDNS errors are sending CHIP onto untested error paths, and those error paths leak memory.

~~~
E (1826960) chip[EM]: OnMessageReceived failed, err = 70
~~~
I see those occasionally, but I believe those are corrupted wifi packets. Wifi does occasionally corrupt on-air packets from interference so they are not unexpected.

jonsmirl, 2023-06-06:
[ACM1-save.txt](https://github.com/espressif/esp-matter/files/11669665/ACM1-save.txt)

That is three hours of logs. First failure is at:
~~~
[0;31mE (6732790) chip[CSL]: PacketBuffer: pool EMPTY.[0m
~~~
Once it starts failing, it keeps on failing.

I just noticed "Use minimal mDNS from CHIP" is selected.  I can try again without that.

jonsmirl, 2023-09-18:
This is memory fragmentation on the C3 leading to an inability to allocate packet buffers. 
I got tired of fighting this and moved to S3 with 2MB of PSRAM and don't encounter this any more.

jonsmirl, 2023-10-31:
This is a symptom of the slow leak problem.  https://github.com/espressif/esp-matter/issues/674

closed 2023-10-31 by jonsmirl