espressif/esp-matter #436: W (18261185) wifi:m f null

jonsmirl, 2023-06-05:
I am letting nodes run overnight and in the morning all of them are dead.
So what is this failure?
~~~
 (18261085) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (18261085) chip[DL]: IP_EVENT_GOT_IP6
I (18261095) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd02:832c:729a:16f1:86f7:03ff:fe66:ffd0
I (18261105) ROUTE_HOOK: Hook already installed on netif, skip...
I (18261115) ep: Interface IP Address Changed
I (18261115) chip[DIS]: Updating services using commissioning mode 0
I (18261125) chip[DIS]: CHIP minimal mDNS started advertising.
I (18261155) chip[DIS]: Advertise operational node 4284C2F5A80B5F7C-3856320C25731B0E
I (18261155) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 4284C2F5A80B5F7C-3856320C25731B0E.
I (18261175) chip[DIS]: mDNS service published: _matter._tcp
I (18261175) chip[DIS]: Advertise operational node C1EDEF2ECC2CF621-AB0DE00000000002
W (18261185) wifi:m f null

I (18261185) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: C1EDEF2ECC2CF621-AB0DE00000000002.
I (18261205) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18261205) chip[DIS]: Failed to advertise records: 3000001
I (18261215) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18261215) chip[DIS]: Failed to advertise records: 3000001
I (18261225) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18261235) chip[DIS]: Failed to advertise records: 3000001
I (18261245) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18261245) chip[DIS]: Failed to advertise records: 3000001
I (18261255) chip[DIS]: mDNS service published: _matter._tcp
I (18262855) app_main: Free internal 34880 biggest block 11776 total 34880
I (18265765) chip[DMG]: Refresh subscribe timer sync after 9 seconds
I (18265765) chip[EM]: <<< [E:59311i S:60663 M:48583524] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0001:05 (IM:ReportData)
I (18265775) chip[IN]: (S) Sending msg 48583524 on secure session with LSID: 60663
E (18265785) chip[EM]: Ignoring transient send error: 3000001 on exchange 59311i
I (18265785) chip[DMG]: Refresh Subscribe Sync Timer with min 1 seconds and max 10 seconds
I (18266015) ROUTE_HOOK: Received RIO
I (18266015) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (18266055) ROUTE_HOOK: Received RIO
I (18266055) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (18266185) chip[IN]: (S) Sending msg 48583524 on secure session with LSID: 60663
E (18266185) chip[EM]: Ignoring transient send error: 3000001 on exchange 59311i
I (18266555) chip[IN]: (S) Sending msg 48583524 on secure session with LSID: 60663
E (18266555) chip[EM]: Ignoring transient send error: 3000001 on exchange 59311i
I (18266925) ROUTE_HOOK: Received RIO
I (18266925) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (18267125) chip[IN]: (S) Sending msg 48583524 on secure session with LSID: 60663
E (18267125) chip[EM]: Ignoring transient send error: 3000001 on exchange 59311i
I (18267245) ROUTE_HOOK: Received RIO
I (18267245) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (18267245) ROUTE_HOOK: Received RIO
I (18267245) ROUTE_HOOK: prefix FD6F:48B9:FE1A:1:: lifetime 1800
I (18268005) chip[IN]: (S) Sending msg 48583524 on secure session with LSID: 60663
E (18268005) chip[EM]: Ignoring transient send error: 3000001 on exchange 59311i
E (18269585) chip[EM]: Failed to Send CHIP MessageCounter:48583524 on exchange 59311i sendCount: 4 max retries: 4
I (18269795) NimBLE: GAP procedure initiated: advertise; 
I (18269795) NimBLE: disc_mode=0
I (18269795) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (18269805) NimBLE: 

I (18269835) NimBLE: GAP procedure initiated: stop advertising.

I (18271995) chip[SC]: SecureSession[0x3fc9b730]: Moving from state 'kActive' --> 'kDefunct'
E (18271995) chip[DMG]: Time out! failed to receive status response from Exchange: 59311i
I (18272165) ROUTE_HOOK: Received RIO
I (18272165) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (18279815) NimBLE: GAP procedure initiated: advertise; 
I (18279815) NimBLE: disc_mode=0
I (18279815) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (18279815) NimBLE: 
~~~

that continues for a while until the packet buffers run out
~~~
I (18401065) ROUTE_HOOK: Received RIO
I (18401075) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (18401595) chip[DL]: IP_EVENT_GOT_IP6
I (18401595) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd02:832c:729a:16f1:86f7:03ff:fe66:ffd0
I (18401595) ROUTE_HOOK: Hook already installed on netif, skip...
I (18401605) ep: Interface IP Address Changed
I (18401605) chip[DIS]: Updating services using commissioning mode 0
I (18401615) chip[DIS]: CHIP minimal mDNS started advertising.
I (18401625) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401625) chip[DIS]: Failed to advertise records: 3000001
I (18401645) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401645) chip[DIS]: Failed to advertise records: 3000001
I (18401655) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401655) chip[DIS]: Failed to advertise records: 3000001
I (18401665) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401675) chip[DIS]: Failed to advertise records: 3000001
I (18401685) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401685) chip[DIS]: Failed to advertise records: 3000001
I (18401695) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401705) chip[DIS]: Failed to advertise records: 3000001
I (18401715) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401715) chip[DIS]: Failed to advertise records: 3000001
I (18401725) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401725) chip[DIS]: Failed to advertise records: 3000001
I (18401735) chip[DIS]: Advertise operational node 4284C2F5A80B5F7C-3856320C25731B0E
I (18401745) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 4284C2F5A80B5F7C-3856320C25731B0E.
I (18401755) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401765) chip[DIS]: Failed to advertise records: 3000001
I (18401775) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401775) chip[DIS]: Failed to advertise records: 3000001
I (18401785) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401795) chip[DIS]: Failed to advertise records: 3000001
I (18401805) chip[DIS]: mDNS broadcast full failed in 1 separate send attempts.
E (18401805) chip[DIS]: Failed to advertise records: 3000001
I (18401815) chip[DIS]: mDNS service published: _matter._tcp
I (18401825) chip[DIS]: Advertise operational node C1EDEF2ECC2CF621-AB0DE00000000002
I (18401825) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: C1EDEF2ECC2CF621-AB0DE00000000002.
E (18401845) chip[CSL]: PacketBuffer: pool EMPTY.
E (18401845) chip[DIS]: Failed to advertise records: b
E (18401855) chip[CSL]: PacketBuffer: pool EMPTY.
E (18401855) chip[DIS]: Failed to advertise records: b
E (18401865) chip[CSL]: PacketBuffer: pool EMPTY.
E (18401865) chip[DIS]: Failed to advertise records: b
E (18401875) chip[CSL]: PacketBuffer: pool EMPTY.
E (18401875) chip[DIS]: Failed to advertise records: b
I (18401885) chip[DIS]: mDNS service published: _matter._tcp
I (18402855) app_main: Free internal 31660 biggest block 11264 total 31660
I (18404765) ROUTE_HOOK: Received RIO
I (18404765) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (18409995) NimBLE: GAP procedure initiated: advertise; 
I (18409995) NimBLE: disc_mode=0
I (18409995) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (18410005) NimBLE: 
~~~

jonsmirl, 2023-06-05:
Right before this failure I had 44388 free
~~~
I (18242855) app_main: Free internal 44388 biggest block 22528 total 44388
~~~
then after the wifi:m f null I lost 10KB of memory.
~~~
I (18262855) app_main: Free internal 34880 biggest block 11776 total 34880
~~~

jonsmirl, 2023-06-05:
esp idf is f00c78b020b129b6a5371997be9fd1ab4cf36f70
esp-matter is a6b958ba055e505226ddd8d49d9e157792369d01

dhrishi, 2023-06-06:
>W (18261185) wifi:m f null

I do not recollect seeing this in any of the long duration logs. Let me check with the Wi-Fi team about this

jonsmirl, 2023-06-07:
Hit it again. And it lost 10KB of memory again.
[ACM0-save.txt](https://github.com/espressif/esp-matter/files/11671547/ACM0-save.txt)

~~~
I (10259207) ROUTE_HOOK: prefix FD9D:2EFF:AAAA:1:: lifetime 1800
I (10260667) app_main: Free internal 25336 biggest block 9216 total 25336
W (10262667) wifi:m f null

I (10263317) NimBLE: GAP procedure initiated: advertise; 
I (10263317) NimBLE: disc_mode=0
I (10263317) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (10263317) NimBLE: 

I (10263347) NimBLE: GAP procedure initiated: stop advertising.

I (10263837) chip[DMG]: Refresh subscribe timer sync after 9 seconds
I (10263847) chip[EM]: <<< [E:50809i S:35556 M:60072233] (S) Msg TX to 2:AB0DE00000000000 [F621] --- Type 0001:05 (IM:ReportData)
I (10263847) chip[IN]: (S) Sending msg 60072233 on secure session with LSID: 35556
E (10263847) chip[EM]: Ignoring transient send error: 3000001 on exchange 50809i
I (10263847) chip[DMG]: Refresh Subscribe Sync Timer with min 1 seconds and max 10 seconds
I (10264257) chip[IN]: (S) Sending msg 60072233 on secure session with LSID: 35556
E (10264257) chip[EM]: Ignoring transient send error: 3000001 on exchange 50809i
I (10264587) chip[IN]: (S) Sending msg 60072233 on secure session with LSID: 35556
E (10264587) chip[EM]: Ignoring transient send error: 3000001 on exchange 50809i
I (10265197) chip[IN]: (S) Sending msg 60072233 on secure session with LSID: 35556
I (10266137) chip[IN]: (S) Sending msg 60072233 on secure session with LSID: 35556
E (10267707) chip[EM]: Failed to Send CHIP MessageCounter:60072233 on exchange 50809i sendCount: 4 max retries: 4
I (10270067) chip[SC]: SecureSession[0x3fc9bd40]: Moving from state 'kActive' --> 'kDefunct'
E (10270067) chip[DMG]: Time out! failed to receive status response from Exchange: 50809i
I (10273317) NimBLE: GAP procedure initiated: advertise; 
I (10273327) NimBLE: disc_mode=0
I (10273327) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (10273327) NimBLE: 

I (10273357) NimBLE: GAP procedure initiated: stop advertising.

I (10280667) app_main: Free internal 15420 biggest block 3328 total 15420
I (10283327) NimBLE: GAP procedure initiated: advertise; 
I (10283327) NimBLE: disc_mode=0
I (10283327) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (10283327) NimBLE: 

I (10283357) NimBLE: GAP procedure initiated: stop advertising.

I (10291717) wifi:bcn_timeout,ap_probe_send_start
I (10293327) NimBLE: GAP procedure initiated: advertise; 
I (10293327) NimBLE: disc_mode=0
~~~

jonsmirl, 2023-06-07:
This is from the file attached to the previous message. Look at the free memory over time.

1) there is a slow leak over time
2) What happened at 5300657, 8280667, 8620667, 10260667. Something is periodically allocating about 10KB of memory. 10KB is a huge alloc in a ESP32C3.

Most important thing is to locate this slow leak. I am losing about 2KB/hr or 32 bytes a minute. That smells like a small alloc which is not getting freed.

~~~
jonsmirl@ares:~/lowpan$ grep app_main ACM0.txt
I (20617) app_main: Free internal 44364 biggest block 23552 total 44364
I (40617) app_main: Free internal 44364 biggest block 23552 total 44364
I (60617) app_main: Free internal 44344 biggest block 23552 total 44344
I (80617) app_main: Free internal 44144 biggest block 23552 total 44144
I (100617) app_main: Free internal 44180 biggest block 23552 total 44180
I (120617) app_main: Free internal 44180 biggest block 23552 total 44180
I (140617) app_main: Free internal 44168 biggest block 23552 total 44168
I (160617) app_main: Free internal 44164 biggest block 23552 total 44164
I (180617) app_main: Free internal 44092 biggest block 23552 total 44092
I (200617) app_main: Free internal 44084 biggest block 23552 total 44084
I (220617) app_main: Free internal 44088 biggest block 23552 total 44088
I (240617) app_main: Free internal 44064 biggest block 23552 total 44064
I (260617) app_main: Free internal 44052 biggest block 23552 total 44052
I (280617) app_main: Free internal 44040 biggest block 23552 total 44040
I (300617) app_main: Free internal 44040 biggest block 23552 total 44040
I (320617) app_main: Free internal 44040 biggest block 23552 total 44040
I (340617) app_main: Free internal 43976 biggest block 23552 total 43976
I (360617) app_main: Free internal 43960 biggest block 23552 total 43960
I (380617) app_main: Free internal 43972 biggest block 23552 total 43972
I (400617) app_main: Free internal 43104 biggest block 23552 total 43104
I (420617) app_main: Free internal 43892 biggest block 23552 total 43892
I (440617) app_main: Free internal 43520 biggest block 23552 total 43520
I (460617) app_main: Free internal 43904 biggest block 23552 total 43904
I (480617) app_main: Free internal 43904 biggest block 23552 total 43904
I (500617) app_main: Free internal 43904 biggest block 23552 total 43904
I (520617) app_main: Free internal 43904 biggest block 23552 total 43904
I (540617) app_main: Free internal 43912 biggest block 23552 total 43912
I (560617) app_main: Free internal 43900 biggest block 23552 total 43900
I (580617) app_main: Free internal 43836 biggest block 23552 total 43836
I (600617) app_main: Free internal 42292 biggest block 21504 total 42292
I (620617) app_main: Free internal 42288 biggest block 21504 total 42288
I (640617) app_main: Free internal 43604 biggest block 23552 total 43604
I (660617) app_main: Free internal 42240 biggest block 23552 total 42240
I (680617) app_main: Free internal 42264 biggest block 23552 total 42264
I (700617) app_main: Free internal 42024 biggest block 21504 total 42024
I (720617) app_main: Free internal 42244 biggest block 23552 total 42244
I (740627) app_main: Free internal 41076 biggest block 21504 total 41076
I (760647) app_main: Free internal 41828 biggest block 21504 total 41956
I (780647) app_main: Free internal 43728 biggest block 23552 total 43728
I (800647) app_main: Free internal 43728 biggest block 23552 total 43728
I (820647) app_main: Free internal 43640 biggest block 23552 total 43640
I (840647) app_main: Free internal 42836 biggest block 23552 total 42836
I (860647) app_main: Free internal 43252 biggest block 23552 total 43252
I (880647) app_main: Free internal 43636 biggest block 23552 total 43636
I (900647) app_main: Free internal 43620 biggest block 23552 total 43620
I (920647) app_main: Free internal 43648 biggest block 23552 total 43648
I (940647) app_main: Free internal 43648 biggest block 23552 total 43648
I (960647) app_main: Free internal 43568 biggest block 23552 total 43568
I (980647) app_main: Free internal 43568 biggest block 23552 total 43568
I (1000647) app_main: Free internal 43568 biggest block 23552 total 43568
I (1020647) app_main: Free internal 43504 biggest block 23552 total 43504
I (1040647) app_main: Free internal 43516 biggest block 23552 total 43516
I (1060647) app_main: Free internal 43516 biggest block 23552 total 43516
I (1080647) app_main: Free internal 43508 biggest block 23552 total 43508
I (1100647) app_main: Free internal 43516 biggest block 23552 total 43516
I (1120647) app_main: Free internal 43508 biggest block 23552 total 43508
I (1140647) app_main: Free internal 43516 biggest block 23552 total 43516
I (1160647) app_main: Free internal 43512 biggest block 23552 total 43512
I (1180647) app_main: Free internal 43512 biggest block 23552 total 43512
I (1200647) app_main: Free internal 43516 biggest block 23552 total 43516
I (1220647) app_main: Free internal 43508 biggest block 23552 total 43508
[... 442 lines cut ...]
I (10080667) app_main: Free internal 37860 biggest block 17408 total 37860
I (10100667) app_main: Free internal 38276 biggest block 19456 total 38276
I (10120667) app_main: Free internal 38276 biggest block 19456 total 38276
I (10140667) app_main: Free internal 38112 biggest block 19456 total 38244
I (10160667) app_main: Free internal 37468 biggest block 17408 total 37468
I (10180667) app_main: Free internal 37468 biggest block 17408 total 37468
I (10200667) app_main: Free internal 38200 biggest block 19456 total 38200
[... cut ...]
I (10420667) app_main: Free internal 12332 biggest block 7680 total 12332
I (10440667) app_main: Free internal 15476 biggest block 1856 total 15476
I (10460667) app_main: Free internal 14188 biggest block 2816 total 14188
I (10480667) app_main: Free internal 15360 biggest block 2816 total 15360
I (10500667) app_main: Free internal 12400 biggest block 7680 total 12400
I (10520667) app_main: Free internal 12352 biggest block 1664 total 12352
I (10540667) app_main: Free internal 16064 biggest block 1664 total 16064
~~~

dhrishi, 2023-06-07:
> Let me check with the Wi-Fi team about this

It’s a failure to allocate null packets.
Probable causes are memory crunch or memory leak

So looks like there's some leak, either in the component code or the app code. We will also setup the light app for long duration and share the results.

jonsmirl, 2023-06-07:
The BLE Bridge app is also suspect. I have that code incorporated, but nothing has been bridged. It is just sitting there idle.

Can you put something into malloc() which looks for those alloc's over 4KB in size? Try and track down whoever is allocating the ~10KB chunck. I'm betting that is somewhere in the CHIP code.  Tell me how to add the check and I can look too.

You can also try running my code. It should work on any ESP32C3 board. Not having target hardware won't bother it, the test is just using the hardware in boot up mode with nothing happening. I can give specific instructions if you want to do this.

shubhamdp, 2023-06-07:
There are [Heap Allocation and Free Function Hooks](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/system/heap_debug.html#heap-allocation-and-free-function-hooks) which can be used, you can put up a check there. But these are in IDF release/v5.1

jonsmirl, 2023-06-07:
One place to look for a leak, why is BLE doing a malloc() while sitting idle? And is it getting freed?
~~~
E (55120827) BLE_INIT: Malloc failed
E (55120827) BLE_MESH: Advertising set failed: err 519
E (55120827) BLE_MESH: Start advertising failed: err 519
W (55120827) BLE_MESH: Failed to send adv packet
I (55125427) app_main: Free internal 4996 biggest block 3584 total 4996
E (55130827) BLE_INIT: Malloc failed
E (55130827) BLE_MESH: Advertising set failed: err 519
E (55130827) BLE_MESH: Start advertising failed: err 519
W (55130827) BLE_MESH: Failed to send adv packet
~~~

jonsmirl, 2023-06-07:
My top suspicion for the leak is the BLE code.

dhrishi, 2023-07-11:
@jonsmirl Our long duration tests (device active for 5+ days) do not seem to leak memory. Did you get a chance to check the BLE code? If it is not something in the SDK, then can you close this?

dhrishi, 2023-07-11:
>Did you try with the BLE Bridge example?

No. It's the light example. Will setup the BLE bridge and update.

jonsmirl, 2023-09-18:
this is a variation of https://github.com/espressif/esp-matter/issues/434

closed 2023-09-18 by jonsmirl