espressif/esp-matter #437: Large static DRAM/IRAM size leads to wifi RAM issue

law-ko, 2023-06-05:
Hello,

We have occurred an issue where the static DRAM and IRAM are too big which sometimes lead to wifi cannot be connected during the provisioning stage. We are testing on ESP32-PICO-MINI02 device.
```
Total sizes:
Used static DRAM:   94716 bytes (  29864 remain, 76.0% used)
      .data size:   21492 bytes
      .bss  size:   73224 bytes
Used static IRAM:  124502 bytes (   6570 remain, 95.0% used)
      .text size:  123475 bytes
   .vectors size:    1027 bytes
Used Flash size : 1219001 bytes
      .text     :  984403 bytes
      .rodata   :  234342 bytes
Total image size: 1364995 bytes (.bin may be padded larger)
```

Is there any way to reduce the DRAM/IRAM size?

Thank you

law-ko, 2023-06-05:
This is the error log when HEAP ran out:
```
I (72213) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (72953) wifi:bcn_timeout,ap_probe_send_start
I (73233) ROUTE_HOOK: Received RIO
I (73233) ROUTE_HOOK: prefix FDE5:69EF:C331:: lifetime 1800
I (73233) ROUTE_HOOK: Received RIO
I (73243) ROUTE_HOOK: prefix FDE5:69EF:C331:: lifetime 1800
I (73253) chip[DL]: IP_EVENT_GOT_IP6
I (73253) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd0e:8d0a:0624:4906:4e75:25ff:fee4:8750
I (73263) ROUTE_HOOK: Hook already installed on netif, skip...
I (73273) app_main: Interface IP Address changed
I (73273) chip[DIS]: Updating services using commissioning mode 0
I (73283) chip[DIS]: CHIP minimal mDNS started advertising.
I (73313) chip[DIS]: Advertise operational node 638C92F1B94D5355-0000000097A58C09
I (73313) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 638C92F1B94D5355-0000000097A58C09.
I (73333) chip[DIS]: mDNS service published: _matter._tcp
I (74673) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (77183) NimBLE: GATT procedure initiated: indicate; 
I (77183) NimBLE: att_handle=14

I (77263) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (79743) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (82253) NimBLE: GATT procedure initiated: indicate; 
I (82253) NimBLE: att_handle=14

I (82353) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (82403) wifi:bcn_timeout,ap_probe_send_start
```

jonsmirl, 2023-06-05:
ESP32-PICO-MINI02 has 2MB of PSRAM. You don't appear to have PSRAM support enabled. Enable it and then set memory allocation strategy in the various components to use it.

law-ko, 2023-06-05:
@jonsmirl Enabling PSRAM would cause an overflow of the IRAM size, is there any way to reduce the Matter components overhead size?

```
Total sizes:
Used static DRAM:   96204 bytes (  28376 remain, 77.2% used)
      .data size:   22724 bytes
      .bss  size:   73480 bytes
Used static IRAM:  143002 bytes ( -11930 remain, 109.1% used) Overflow detected! You can run idf.py size-files for more information.
      .text size:  141975 bytes
   .vectors size:    1027 bytes
Used Flash size : 1273077 bytes
      .text     : 1034007 bytes
      .rodata   :  238814 bytes
Total image size: 1438803 bytes (.bin may be padded larger)
```

jonsmirl, 2023-06-05:
It is your app causing the overflow. The Matter libraries fit without problem. You need to switch your app to use PSRAM.

law-ko, 2023-06-05:
@jonsmirl  We are using the native example `lightbulb` nothing was added but only the change of supporting PSRAM in menuconfig

law-ko, 2023-06-14:
Any idea on this from the Espressif team?

law-ko, 2023-06-15:
This is the log, the device will disconnect from the wifi overtime:
```
I (6431600) wifi:bcn_timeout,ap_probe_send_start
I (6440610) wifi:bcn_timeout,ap_probe_send_start
I (6449620) wifi:bcn_timeout,ap_probe_send_start
I (6458640) wifi:bcn_timeout,ap_probe_send_start
I (6458760) ROUTE_HOOK: Received RIO
I (6458760) ROUTE_HOOK: prefix FDE5:69EF:C331:: lifetime 1800
I (6458760) ROUTE_HOOK: Received RIO
I (6458770) ROUTE_HOOK: prefix FDE5:69EF:C331:: lifetime 1800
I (6467660) wifi:bcn_timeout,ap_probe_send_start
I (6476680) wifi:bcn_timeout,ap_probe_send_start
I (6491600) wifi:bcn_timeout,ap_probe_send_start
I (6500620) wifi:bcn_timeout,ap_probe_send_start
```

shubhamdp, 2023-08-11:
@law-ko [Minimizing RAM Usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html) guide has some options that you can try.

dhrishi, 2023-09-30:
Closing this as the link to minimizing the RAM usage has been provided

closed 2023-09-30 by dhrishi