espressif/esp-matter #439: ESP32 or ESP32C3  run Example generic_switch , Unlimited restart

zsr391912829, 2023-06-05:
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6956
ho 0 tail 12 room 4
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (29) boot: ESP-IDF v5.0.1-397-g3050ea656f 2nd stage bootloader
I (29) boot: compile time 17:39:27
I (29) boot: chip revision: v3.0
I (33) boot.esp32: SPI Speed      : 40MHz
I (38) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (71) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (86) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (93) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (101) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (108) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (116) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (124) boot: End of partition table
I (128) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=38498h (230552) map
I (220) esp_image: segment 1: paddr=000584c0 vaddr=3ffbdb60 size=053f4h ( 21492) load
I (228) esp_image: segment 2: paddr=0005d8bc vaddr=40080000 size=0275ch ( 10076) load
I (233) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=ed404h (971780) map
I (587) esp_image: segment 4: paddr=0014d42c vaddr=4008275c size=1be90h (114320) load
I (650) boot: Loaded app from partition at offset 0x20000
I (650) boot: Disabling RNG early entropy source...
I (661) cpu_start: Pro cpu up.
I (661) cpu_start: Starting app cpu, entry point is 0x4008140c
0x4008140c: call_start_cpu1 at /home/zsr/Desktop/myShare/esp32_IDF5.0/esp-idf/components/esp_system/port/cpu_start.c:141

I (647) cpu_start: App cpu up.
I (678) cpu_start: Pro cpu start user code
I (678) cpu_start: cpu freq: 160000000 Hz
I (678) cpu_start: Application information:
I (683) cpu_start: Project name:     generic_switch
I (688) cpu_start: App version:      v1.0
I (693) cpu_start: Compile time:     Jun  5 2023 17:39:21
I (699) cpu_start: ELF file SHA256:  d736e29c701ddd25...
I (705) cpu_start: ESP-IDF:          v5.0.1-397-g3050ea656f
I (711) cpu_start: Min chip rev:     v0.0
I (716) cpu_start: Max chip rev:     v3.99 
I (721) cpu_start: Chip rev:         v3.0
I (726) heap_init: Initializing. RAM available for dynamic allocation:
I (733) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (739) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (745) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (751) heap_init: At 3FFD3E18 len 0000C1E8 (48 KiB): DRAM
I (757) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (763) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (770) heap_init: At 4009E5EC len 00001A14 (6 KiB): IRAM
I (778) spi_flash: detected chip: generic
I (781) spi_flash: flash io: dio
I (788) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (838) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (858) app_main: Generic Switch created with endpoint_id 1
I (868) wifi:wifi driver task: 3ffd9f90, prio:23, stack:6656, core=0
[... 37 lines cut ...]
0x4009884a: abort at /home/zsr/Desktop/myShare/esp32_IDF5.0/esp-idf/components/newlib/abort.c:38

0x40121ddd: chip::DeviceLayer::GetCommissionableDataProvider() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chip::DeviceLayer::GetCommissionableDataProvider() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/CommissionableDataProvider.cpp:32

0x4012b26f: chip::DeviceLayer::Internal::BLEManagerImpl::ConfigureAdvertisingData() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:963

0x4012b879: chip::DeviceLayer::Internal::BLEManagerImpl::DriveBLEState() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:748

0x4012ba42: chip::DeviceLayer::Internal::BLEManagerImpl::DriveBLEState(int) at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1602

0x4012a781: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:290

0x4012aa8a: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:505
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257

0x4012aac9: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:405
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280

0x400955b1: vPortTaskWrapper at /home/zsr/Desktop/myShare/esp32_IDF5.0/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: d736e29c701ddd25

Rebooting...
ets Jul 29 2019 12:21:46

maenkai, 2023-06-06:
The latest code for esp-matter has solved this problem

shubhamdp, 2023-06-06:
@zsr391912829 Can you share the esp-matter and esp-idf commit used.

> (inlined by) chip::DeviceLayer::GetCommissionableDataProvider() at /home/zsr/Desktop/myShare/espMatter/esp-matter/examples/generic_switch/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/CommissionableDataProvider.cpp:32

This line from backtrace says that `CommissionableDataProvider` was not set.

zsr391912829, 2023-06-06:
The latest code for esp-matter has solved this problem 。 But when I use the google service bind my Esp32  , report some error. 
....
(23863) chip[ZCL]: OpCreds: Failed CertificateChainRequest: 201
....

zsr391912829, 2023-06-06:
entry 0x40080650
I (33) boot: ESP-IDF v5.2-dev-921-gdc016f5987 2nd stage bootloader
I (33) boot: compile time Jun  6 2023 16:28:58
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (80) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (88) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (95) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (102) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (110) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (118) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (125) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (133) boot: End of partition table
I (137) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=37890h (227472) map
I (228) esp_image: segment 1: paddr=000578b8 vaddr=3ffbdb60 size=056c4h ( 22212) load
I (237) esp_image: segment 2: paddr=0005cf84 vaddr=40080000 size=03094h ( 12436) load
I (242) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=f1990h (989584) map
I (601) esp_image: segment 4: paddr=001519b8 vaddr=40083094 size=1b894h (112788) load
I (664) boot: Loaded app from partition at offset 0x20000
I (664) boot: Disabling RNG early entropy source...
I (676) cpu_start: Multicore app
I (676) cpu_start: Pro cpu up.
I (676) cpu_start: Starting app cpu, entry point is 0x40081308
0x40081308: call_start_cpu1 at /home/zsr/Desktop/myShare/espMatter/esp-gitee-tools/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (696) cpu_start: Pro cpu start user code
I (696) cpu_start: cpu freq: 160000000 Hz
I (696) cpu_start: Application information:
I (701) cpu_start: Project name:     generic_switch
I (706) cpu_start: App version:      v1.0
I (711) cpu_start: Compile time:     Jun  6 2023 16:28:53
I (717) cpu_start: ELF file SHA256:  2dec51a97...
I (722) cpu_start: ESP-IDF:          v5.2-dev-921-gdc016f5987
I (729) cpu_start: Min chip rev:     v0.0
I (734) cpu_start: Max chip rev:     v3.99 
I (738) cpu_start: Chip rev:         v3.0
I (743) heap_init: Initializing. RAM available for dynamic allocation:
I (750) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (756) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (762) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (769) heap_init: At 3FFD43B0 len 0000BC50 (47 KiB): DRAM
I (775) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (781) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (788) heap_init: At 4009E928 len 000016D8 (5 KiB): IRAM
I (795) spi_flash: detected chip: generic
I (798) spi_flash: flash io: dio
I (805) coexist: coex firmware version: ebddf30
I (808) app_start: Starting scheduler on CPU0
I (813) app_start: Starting scheduler on CPU1
mI (813) main_task: Started on CPU0
I (823) main_task: Calling app_main()
I (903) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (913) app_main: Generic Switch created with endpoint_id 1
I (923) wifi:wifi driver task: 3ffda768, prio:23, stack:6656, core=0
I (953) wifi:wifi firmware version: b2f1f86
I (953) wifi:wifi certification version: v7.0
I (953) wifi:config NVS flash: enabled
I (953) wifi:config nano formating: disabled
I (953) wifi:Init data frame dynamic rx buffer num: 32
I (963) wifi:Init management frame dynamic rx buffer num: 32
I (963) wifi:Init management short buffer num: 32
I (973) wifi:Init dynamic tx buffer num: 32
[... 219 lines cut ...]
I (23593) esp_matter_command: Received command 0x00000002 for endpoint 0x0000's cluster 0x00000030
I (23613) chip[DL]: NVS set: chip-config/reg-location = 0 (0x0)
I (23613) chip[DL]: NVS set: chip-config/country-code = "CN"
I (23623) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 1 **********
I (23633) chip[EM]: <<< [E:50409r S:30443 M:52086028] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (23653) chip[IN]: (S) Sending msg 52086028 on secure session with LSID: 30443
I (23653) NimBLE: GATT procedure initiated: indicate; 
I (23663) NimBLE: att_handle=14

I (23813) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (23813) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (23833) chip[EM]: >>> [E:50410r S:30443 M:45383778] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (23843) esp_matter_command: Received command 0x00000002 for endpoint 0x0000's cluster 0x0000003E
I (23853) chip[ZCL]: OpCreds: Certificate Chain request received for DAC
E (23863) chip[ZCL]**: **OpCreds: Failed CertificateChainRequest: 201****
I (23873) chip[EM]: <<< [E:50410r S:30443 M:52086029] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (23873) chip[IN]: (S) Sending msg 52086029 on secure session with LSID: 30443
I (23883) NimBLE: GATT procedure initiated: indicate; 
I (23893) NimBLE: att_handle=14

I (23963) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (24983) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0
I (24983) chip[DL]: CHIPoBLE unsubscribe received
I (24993) chip[DL]: BLE GAP connection terminated (con 0 reason 0x213)
I (24983) chip[BLE]: Releasing end point's BLE connection back to application.

dhrishi, 2023-06-06:
@zsr391912829 Please file a new issue with all the required details and upload (not paste) the logs. I will close this as the original problem is solved

closed 2023-06-06 by dhrishi