espressif/esp-matter #440: switch binding with light, but switch can't  control light

Wvirgil123, 2023-06-05:
switch device :  runing  light_switch demo
light device:  runing  light  demo

1.  commission  switch device  , node id 0x0123 ( 291 in decimal )
2. commission light device , node id 0x0124 ( 292 in decimal ) 
3. Update the light's acl attribute: `accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [ 112233, 291 ], "targets": null}]' 0x0124 0x0`
4. binding: ` binding write binding '[{"node":292, "endpoint":1, "cluster":6}]' 0x0123 0x1`

switch log: **FSend command failure**
```
> I (2492453) app_driver: Toggle button pressed
I (2492453) chip[DIS]: Found an existing secure session to [1:0000000000000124]!
I (2492453) chip[EM]: <<< [E:54804i S:43767 M:52359527] (S) Msg TX to 1:0000000000000124 [A192] --- Type 0001:08 (IM:InvokeCommandRequest)
I (2492473) chip[IN]: (S) Sending msg 52359527 on secure session with LSID: 43767
I (2492673) chip[EM]: >>> [E:54804i S:43767 M:52329957 (Ack:52359527)] (S) Msg RX from 1:0000000000000124 [A192] --- Type 0001:09 (IM:InvokeCommandResponse)
I (2492683) chip[DMG]: Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0002 Status=0xc3
I (2492683) esp_matter_client: FSend command failure
I (2492703) chip[EM]: <<< [E:54804i S:43767 M:52359528 (Ack:52329957)] (S) Msg TX to 1:0000000000000124 [A192] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (2492713) chip[IN]: (S) Sending msg 52359528 on secure session with LSID: 43767

```

light log: 
```
> I (1226559) chip[EM]: >>> [E:54804r S:60336 M:52359527] (S) Msg RX from 1:0000000000000123 [A192] --- Type 0001:08 (IM:InvokeCommandRequest)
I (1226569) chip[EM]: <<< [E:54804r S:60336 M:52329957 (Ack:52359527)] (S) Msg TX to 1:0000000000000123 [A192] --- Type 0001:09 (IM:InvokeCommandResponse)
I (1226579) chip[IN]: (S) Sending msg 52329957 on secure session with LSID: 60336
I (1226769) chip[EM]: >>> [E:54804r S:60336 M:52359528 (Ack:52329957)] (S) Msg RX from 1:0000000000000123 [A192] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

shubhamdp, 2023-06-06:
@Wvirgil123 Can you please attach the complete logs from light and switch. Also, please share more detail like esp-matter and esp-idf commit's used.

Wvirgil123, 2023-06-07:
> 
esp-idf: v5.0 
esp-matter: 2f0929e5f8fd58b1a4cbaa2bd8b171ce50ad2a75
device: esp32s3

I'll try again later, and I'll be able to toggle the light via switch.  and  I have to restart the lights before they can establish a connection. 

switch logs

```
I (25) boot: ESP-IDF v5.0-322-g6181cc6868 2nd stage bootloader
I (25) boot: compile time 16:26:04
I (25) boot: chip revision: v0.1
I (28) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (73) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (80) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (88) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (95) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (103) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (110) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (118) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (125) boot: End of partition table
I (130) boot_comm: chip revision: 1, min. application chip revision: 0
I (137) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=37068h (225384) map
I (186) esp_image: segment 1: paddr=00057090 vaddr=3fc98f00 size=050b0h ( 20656) load
I (195) esp_image: segment 2: paddr=0005c148 vaddr=40374000 size=03ed0h ( 16080) load
I (199) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=e2eb4h (929460) map
I (368) esp_image: segment 4: paddr=00142edc vaddr=40377ed0 size=10f74h ( 69492) load
I (392) boot: Loaded app from partition at offset 0x20000
I (392) boot: Disabling RNG early entropy source...
I (404) cpu_start: Pro cpu up.
I (404) cpu_start: Starting app cpu, entry point is 0x40375504
0x40375504: call_start_cpu1 at /Users/virgil/seeed/SenseCAP_Terminal_ESP32/code/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (418) cpu_start: Pro cpu start user code
I (418) cpu_start: cpu freq: 160000000 Hz
I (419) cpu_start: Application information:
I (421) cpu_start: Project name:     light_switch
I (427) cpu_start: App version:      v1.0
I (432) cpu_start: Compile time:     Jun  5 2023 16:39:24
I (438) cpu_start: ELF file SHA256:  badfeb69e3fba67f...
I (444) cpu_start: ESP-IDF:          v5.0-322-g6181cc6868
I (450) heap_init: Initializing. RAM available for dynamic allocation:
I (457) heap_init: At 3FCAE428 len 0003B2E8 (236 KiB): D/IRAM
I (463) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (470) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (476) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (483) spi_flash: detected chip: winbond
I (487) spi_flash: flash io: dio
W (491) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (506) coexist: coexist rom version e7ae62f
I (510) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (630) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (630) app_main: Switch created with endpoint_id 1
I (640) pp: pp rom version: e7ae62f
I (640) net80211: net80211 rom version: e7ae62f
I (650) wifi:wifi driver task: 3fcf7630, prio:23, stack:6656, core=0
I (650) system_api: Base MAC address is not set
I (650) system_api: read default base MAC address from EFUSE
I (670) wifi:wifi firmware version: aa12444
I (670) wifi:wifi certification version: v7.0
[... 179 lines cut ...]
> 
> 
> 
> 
> I (106530) app_driver: Toggle button pressed
I (106530) chip[DIS]: Found an existing secure session to [2:0000000000000124]!
I (106530) chip[EM]: <<< [E:54280i S:10177 M:45334610] (S) Msg TX to 2:0000000000000124 [A192] --- Type 0001:08 (IM:InvokeCommandRequest)
I (106540) chip[IN]: (S) Sending msg 45334610 on secure session with LSID: 10177
I (106870) chip[EM]: >>> [E:54280i S:10177 M:166260950 (Ack:45334610)] (S) Msg RX from 2:0000000000000124 [A192] --- Type 0001:09 (IM:InvokeCommandResponse)
I (106880) chip[DMG]: Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0002 Status=0x0
I (106880) esp_matter_client: Send command success
I (106890) chip[EM]: <<< [E:54280i S:10177 M:45334611 (Ack:166260950)] (S) Msg TX to 2:0000000000000124 [A192] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (106910) chip[IN]: (S) Sending msg 45334611 on secure session with LSID: 10177

> 
> I (111110) app_driver: Toggle button pressed
I (111110) chip[DIS]: Found an existing secure session to [2:0000000000000124]!
I (111110) chip[EM]: <<< [E:54281i S:10177 M:45334612] (S) Msg TX to 2:0000000000000124 [A192] --- Type 0001:08 (IM:InvokeCommandRequest)
I (111120) chip[IN]: (S) Sending msg 45334612 on secure session with LSID: 10177
I (111370) chip[EM]: >>> [E:54281i S:10177 M:166260951 (Ack:45334612)] (S) Msg RX from 2:0000000000000124 [A192] --- Type 0001:09 (IM:InvokeCommandResponse)
I (111380) chip[DMG]: Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0002 Status=0x0
I (111390) esp_matter_client: Send command success
I (111400) chip[EM]: <<< [E:54281i S:10177 M:45334613 (Ack:166260951)] (S) Msg TX to 2:0000000000000124 [A192] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111410) chip[IN]: (S) Sending msg 45334613 on secure session with LSID: 10177
```

dhrishi, 2023-06-24:
@Wvirgil123 Closing as it is working fine as mentioned above. Please check your setup and commands, if you are still seeing an issue

closed 2023-06-24 by dhrishi

goloveychuk, 2023-10-12:
I had same error
```
E (51730) chip[DIS]: OperationalSessionSetup[2:0000000000000124]: operational discovery failed: 32
E (51750) chip[SVR]: Failed to establish connection to node 0x0000000000000124
```
My flow is:
1) build and flash light and switch
2) commission to apple home via Apple tv (thread)
3) enable pairing mode and commission via chip tool (`pairing code ID CODE`)
4) set ACL and binding
5) get error.
I've fixed it by restarting Apple TV. (after both devices are paired)
It seems like it has bug and server which should resolve node to ip (SRP server)  does not work correctly.