espressif/esp-matter #441: load partition table error

Focalcrest-Madi, 2023-06-06:
I am creating factory NVS partition images, after flashing the bin file to the device, the device fails to start. The device log is as follows:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cfc2e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16b4
load:0x403cc710,len:0x930
load:0x403ce710,len:0x2d28
entry 0x403cc710
I (35) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (35) boot: compile time 15:09:02
I (35) boot: chip revision: 4
I (37) boot_comm: chip revision: 4, min. bootloader chip revision: 3
I (44) boot.esp32c3: SPI Speed      : 80MHz
I (49) boot.esp32c3: SPI Mode       : DIO
I (53) boot.esp32c3: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
E (63) flash_parts: partition 0 invalid magic number 0xfffc
E (70) boot: Failed to verify partition table
E (75) boot: load partition table error!

The command to flash the bin file is as follows
esptool.py -p /dev/tty.usbserial-143210 -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0xc000 ~/esp/esp-matter/tools/mfg_tool/out/130c_1/f05695fa-85e8-45d4-b23e-162288d881fa/f05695fa-85e8-45d4-b23e-162288d881fa-partition.bin 0x17000 build/ota_data_initial.bin 0x20000 build/light.bin

I configured the device as follows:
1、Enable ESP32 Factory Data Provider
[image]
2、Enable ESP32 Device Instance Info Provider
[image]
3、Enable Attestation - Factory
[image]
4、Set chip-factory namespace partition label
[image]

Focalcrest-Madi, 2023-06-06:
The Matter certificate generation command I use is as follows：
1、Generate the Test CD file
./out/host/chip-cert gen-cd --format-version 1 --vendor-id 0x130C --product-id 0x1 --device-type-id 0x010c --certificate-id CSA00000SWC00000-01 --security-level 0 --security-info 0 --version-number 1 --certification-type 1 --key credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem --cert credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem                           --out ~/esp/matter-proj/credentials/test/test-CD.der

2、Generating PAA Certificate
./out/host/chip-cert gen-att-cert --type a --subject-cn "EVVR PAA CN" --subject-vid 0x130C --valid-from "2023-06-06 14:38:50" --lifetime 4294967295 --out-key ~/esp/matter-proj/credentials/test/test-PAA_key.pem --out ~/esp/matter-proj/credentials/test/test-PAA_cert.pem

3、Generating Factory Partition Binary Files
./mfg_tool.py -cn EVVR --paa -c ~/esp/matter-proj/credentials/test/test-PAA_cert.pem -k ~/esp/matter-proj/credentials/test/test-PAA_key.pem -cd ~/esp/matter-proj/credentials/test/test-CD.der -v 0x130C --vendor-name EVVR -p 0x01 --product-name Smart-Plug --hw-ver 1 --hw-ver-str v1.2

shubhamdp, 2023-06-06:
@Focalcrest-Madi you forgot to flash the partition table at `CONFIG_PARTITION_TABLE_OFFSET` (you can find this is sdkconfig or can be configured in menuconfig).

Also, curious why are you building a command, why not `idf.py flash`?

Focalcrest-Madi, 2023-06-06:
@shubhamdp  Are you saying that I forgot to flash the partition table bin file?

What is the flash address of the manufacturing binary?
[image]

closed 2023-06-06 by Focalcrest-Madi