espressif/esp-matter #443: Commissioning window Open fail

Focalcrest-Madi, 2023-06-06:
I flash the Factory Partition Binary to the ESP Matter device, but the Commissioning window was not opened when the device started.
The device log is as follows：
[device.log](https://github.com/espressif/esp-matter/files/11663403/device.log)
The command to flash the Factory Partition Binary is： esptool.py -p /dev/tty.usbserial-143210 write_flash 0x10000 ~/esp/esp-matter/tools/mfg_tool/out/130c_1/f05695fa-85e8-45d4-b23e-162288d881fa/f05695fa-85e8-45d4-b23e-162288d881fa-partition.bin
The device partition table information is as follows:
[image]

dhrishi, 2023-06-06:
@Focalcrest-Madi There seems to be an issue with the BLE advertisement:

```
E (1193) chip[DL]: Configure Adv Data failed: Error ESP32:0x0500110F
E (1193) chip[DL]: Disabling CHIPoBLE service due to error: Error ESP32:0x0500110F
```

Please provide the following:
1. ESP-IDF and esp-matter GIT head commits
2. Example you are using. The changes, if any, you have done on the top
3. All steps for generating the factory partition and also the menuconfig option changes
4. The sdkconfig file that is generated.

Please re-open this issue once you provide the required details.

closed 2023-06-06 by dhrishi

Focalcrest-Madi, 2023-06-06:
ESP-IDF version is v4.4.3
esp-matter version is release/v1.0

I configured the device as follows:
1、Enable ESP32 Factory Data Provider
[image]
2、Enable ESP32 Device Instance Info Provider
[image]
3、Enable Attestation - Factory
[image]
4、Set chip-factory namespace partition label
[image]

The Matter certificate generation command I use is as follows：
1、Generate the Test CD file
./out/host/chip-cert gen-cd --format-version 1 --vendor-id 0x130C --product-id 0x1 --device-type-id 0x010c --certificate-id CSA00000SWC00000-01 --security-level 0 --security-info 0 --version-number 1 --certification-type 1 --key credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem --cert credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem --out ~/esp/matter-proj/credentials/test/test-CD.der

2、Generating PAA Certificate
./out/host/chip-cert gen-att-cert --type a --subject-cn "EVVR PAA CN" --subject-vid 0x130C --valid-from "2023-06-06 14:38:50" --lifetime 4294967295 --out-key ~/esp/matter-proj/credentials/test/test-PAA_key.pem --out ~/esp/matter-proj/credentials/test/test-PAA_cert.pem

3、Generating Factory Partition Binary Files
./mfg_tool.py -cn EVVR --paa -c ~/esp/matter-proj/credentials/test/test-PAA_cert.pem -k ~/esp/matter-proj/credentials/test/test-PAA_key.pem -cd ~/esp/matter-proj/credentials/test/test-CD.der -v 0x130C --vendor-name EVVR -p 0x01 --product-name Smart-Plug --hw-ver 1 --hw-ver-str v1.2

shubhamdp, 2023-06-06:
@Focalcrest-Madi "chip-factory namespace partition label" config shall be one of the label from your partition table. I do not see "chip-factory" in your partition table. Do you want to use "fctry" instead?

Also, this generated partition shall be flashed at the address of the configure partition. In your case if you use "fctry" partition then it has to be flashed at address `0x3e0000`.

Focalcrest-Madi, 2023-06-06:
@shubhamdp  I added the "chip-factory namespace partition label" partition, but I flash the Factory Partition Binary file with the following command, the device commissioning window open failed to open startup.
[image]
`esptool.py -p /dev/tty.usbserial-14310 write_flash 0x3E6000 ~/esp/esp-matter/tools/mfg_tool/out/130c_1/f05695fa-85e8-45d4-b23e-162288d881fa/f05695fa-85e8-45d4-b23e-162288d881fa-partition.bin`

The device log is as follows：
[device.log](https://github.com/espressif/esp-matter/files/11665146/device.log)

shubhamdp, 2023-06-06:
> It is strongly recommended that you include an NVS partition of at least 0x3000 bytes in your project.

Quoting from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#subtype

This could be the problem.

Focalcrest-Madi, 2023-06-07:
@shubhamdp 
I configured the "chip-factory" partition to be 0x4000 bytes, but I still can't open the debug window, and the device log is the same as before

Focalcrest-Madi, 2023-06-07:
@shubhamdp 
I configured the "chip-factory namespace partition label" as the nvs partition. After burning the certificate, the device can open the commissioning window, but the pairing fails.
[image]
Flash certificate command： esptool.py -p /dev/tty.usbserial-143210 write_flash 0x10000 ~/esp/esp-matter/tools/mfg_tool/out/130c_1/f05695fa-85e8-45d4-b23e-162288d881fa/f05695fa-85e8-45d4-b23e-162288d881fa-partition.bin

The device and chip-tool log is as follows：
[chip-tool.log](https://github.com/espressif/esp-matter/files/11672043/chip-tool.log)
[device.log](https://github.com/espressif/esp-matter/files/11672044/device.log)