espressif/esp-matter #444: Commissioning window Open fail

Focalcrest-Madi, 2023-06-06:
I flash the Factory Partition Binary to the ESP Matter device, but the Commissioning window was not opened when the device started.

ESP-IDF version is v4.4.3
esp-matter version is release/v1.0

I configured the device as follows:
1、Enable ESP32 Factory Data Provider
[image]
2、Enable ESP32 Device Instance Info Provider
[image]
3、Enable Attestation - Factory
[image]
4、Set chip-factory namespace partition label
[image]

ESP-IDF version is v4.4.3
esp-matter version is release/v1.0

I configured the device as follows:
1、Enable ESP32 Factory Data Provider
image
2、Enable ESP32 Device Instance Info Provider
image
3、Enable Attestation - Factory
image
4、Set chip-factory namespace partition label
image

The Matter certificate generation command I use is as follows：
1、Generate the Test CD file
./out/host/chip-cert gen-cd --format-version 1 --vendor-id 0x130C --product-id 0x1 --device-type-id 0x010c --certificate-id CSA00000SWC00000-01 --security-level 0 --security-info 0 --version-number 1 --certification-type 1 --key credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem --cert credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem --out ~/esp/matter-proj/credentials/test/test-CD.der

2、Generating PAA Certificate
./out/host/chip-cert gen-att-cert --type a --subject-cn "EVVR PAA CN" --subject-vid 0x130C --valid-from "2023-06-06 14:38:50" --lifetime 4294967295 --out-key ~/esp/matter-proj/credentials/test/test-PAA_key.pem --out ~/esp/matter-proj/credentials/test/test-PAA_cert.pem

3、Generating Factory Partition Binary Files
./mfg_tool.py -cn EVVR --paa -c ~/esp/matter-proj/credentials/test/test-PAA_cert.pem -k ~/esp/matter-proj/credentials/test/test-PAA_key.pem -cd ~/esp/matter-proj/credentials/test/test-CD.der -v 0x130C --vendor-name EVVR -p 0x01 --product-name Smart-Plug --hw-ver 1 --hw-ver-str v1.2

The device log is as follows：
[device.log](https://github.com/espressif/esp-matter/files/11663403/device.log)

The command to flash the Factory Partition Binary is： esptool.py -p /dev/tty.usbserial-143210 write_flash 0x10000 ~/esp/esp-matter/tools/mfg_tool/out/130c_1/f05695fa-85e8-45d4-b23e-162288d881fa/f05695fa-85e8-45d4-b23e-162288d881fa-partition.bin
The device partition table information is as follows:
[image]

shubhamdp, 2023-06-06:
@Focalcrest-Madi please take a look at https://github.com/espressif/esp-matter/issues/443#issuecomment-1578555455, I have replied there.

dhrishi, 2023-06-06:
@Focalcrest-Madi Ideally that should fix your issue. Please give it a try and close this

closed 2023-06-06 by Focalcrest-Madi