espressif/esp-matter #445: IDF: undefined reference to `ETH_EVENT'

jonsmirl, 2023-06-06:
I turned off "use CHIP MDNS" so that it would use the Espressif one. That worked on the ESP32C3, but it doesn't compile on the ESP32S3. I get this link error.
~~~
-idf/components/esp_phy/lib/esp32s3/libphy.a  /home/jonsmirl/esp/esp-idf/components/esp_phy/lib/esp32s3/libbtbb.a  esp-idf/esp_event/libesp_event.a  esp-idf/hal/libhal.a  esp-idf/esp_system/libesp_system.a  esp-idf/soc/libsoc.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/vfs/libvfs.a  esp-idf/driver/libdriver.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  esp-idf/mbedtls/libmbedtls.a  -Wl,--end-group && :
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/espressif__mdns/libespressif__mdns.a(mdns.c.obj):(.literal.unregister_predefined_handlers+0xc): undefined reference to `ETH_EVENT'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jonsmirl/lowpan/build/log/idf_py_stderr_output_666744 and /home/jonsmirl/lowpan/build/log/idf_py_stdout_output_666744
~~~

I am on IDF:
commit ca8bf608df6caf9f83df2e2f98ff71815ad5c186 (HEAD, origin/release/v5.0)

jonsmirl, 2023-06-07:
This fails on IDF 5.1 too. 

This is something broken in the IDF MDNS support for the ESP32S3.

shubhamdp, 2023-06-08:
@jonsmirl this will be fixed with https://github.com/project-chip/connectedhomeip/pull/27140

jonsmirl, 2023-06-08:
From that patch I can see there is a hidden ETH_ENABLED config variable for the S3.  USE_SPI_ETHERNET is defaulted on. That SPI option turns on ETH_ENABLED. I'm not using Ethernet, so the fix for me is to turn off USE_SPI_ETHERNET.

I just built without error after turning USE_SPI_ETHERNET off.

closed 2023-06-08 by jonsmirl