espressif/esp-matter #446: In the Provision chip device with wifi's input page, click Commission cause crash

BonyYeah, 2023-06-07:
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.google.chip.chiptool, PID: 8438
    java.lang.NoSuchMethodError: no non-static method "Lchip/devicecontroller/ChipDeviceControllerException;.<init>(ILjava/lang/String;)V"
        at chip.devicecontroller.ChipDeviceController.pairDeviceWithAddress(Native Method)
        at chip.devicecontroller.ChipDeviceController.pairDeviceWithAddress(ChipDeviceController.java:174)
        at com.google.chip.chiptool.provisioning.DeviceProvisioningFragment.pairDeviceWithAddress(DeviceProvisioningFragment.kt:156)
        at com.google.chip.chiptool.provisioning.DeviceProvisioningFragment.onCreateView(DeviceProvisioningFragment.kt:81)
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2950)
        at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:518)
        at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:277)
        at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2177)
        at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2088)
        at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1990)
        at androidx.fragment.app.FragmentManager$5.run(FragmentManager.java:524)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at android.app.ActivityThread.main(ActivityThread.java:7870)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

jonsmirl, 2023-06-07:
No one is maintaining that code, use this instead.
https://github.com/google-home/sample-app-for-matter-android

dhrishi, 2023-06-07:
@BonyYeah This looks like to be a crash in the CHIPTool Android app code. Please file an issue [here](https://github.com/project-chip/connectedhomeip/issues/)

closed 2023-06-07 by dhrishi